ros-industrial/motoman #301: [ERROR] [1571685314.425238047]: Action client not connected: gp8_position_controller/follow_joint_trajectory

Bdelspi, 2019-10-21:
Hi I'm trying to move our Yaskawa GP8 in moveit and keep getting the following error

[ INFO] [1571685301.003525675]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
[ WARN] [1571685302.424294442]: Waiting for gp8_position_controller/follow_joint_trajectory to come up
[ INFO] [1571685306.015098238]: Failed to call service get_planning_scene, have you launched move_group? at /tmp/binarydeb/ros-kinetic-moveit-ros-planning-0.9.17/planning_scene_monitor/src/planning_scene_monitor.cpp:491
[ INFO] [1571685306.225053330]: Constructing new MoveGroup connection for group 'gp8_arm' in namespace ''
[ WARN] [1571685308.424709191]: Waiting for gp8_position_controller/follow_joint_trajectory to come up
[ERROR] [1571685314.425238047]: Action client not connected: gp8_position_controller/follow_joint_trajectory

**My process**
I first launch a terminal and start roscore

Then open a new terminal and set rosparam

In a new terminal, I roslaunch the robot_interface_streaming_yrc1000.launch file

Then in a new terminal call the rosservice /robot_enable

in a new terminal, I do as done [here](https://groups.google.com/forum/#!topic/swri-ros-pkg-dev/hXic8PporvY), and roslaunch motoman_gp8_moveit_config moveit_planning_execution.launch robot_ip:=<address> controller:=yrc1000 sim:=false

However, I continue to get the same error message stated above.

**My controllers.yaml file looks like this:**

controller_list:
  - name: gp8_position_controller
    action_ns: joint_trajectory_action
    default: True
    type: FollowJointTrajectory
    joints: ['joint_1_s', 'joint_2_l', 'joint_3_u', 'joint_4_r', 'joint_5_b', 'joint_6_t']

**joint_names.yaml file like this:**
controller_joint_names: ['joint_1_s', 'joint_2_l', 'joint_3_u', 'joint_4_r', 'joint_5_b', 'joint_6_t']

**the controller spawner looks like this:**
bdelspi@bdelspi-Latitude-E5550:~/catkin_ws2$ roslaunch motoman_gp8_moveit_config ros_controllers.launch
... logging to /home/bdelspi/.ros/log/f311e750-f433-11e9-932c-60571895a164/roslaunch-bdelspi-Latitude-E5550-18464.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://bdelspi-Latitude-E5550:35523/

SUMMARY
========

PARAMETERS
 * /controller_list: [{'default': True...
 * /generic_hw_control_loop/cycle_time_error_threshold: 0.01
 * /generic_hw_control_loop/loop_hz: 300
 * /hardware_interface/joints: ['joint_1_s', 'jo...
 * /hardware_interface/sim_control_mode: 1
 * /joint_state_controller/publish_rate: 50
 * /joint_state_controller/type: joint_state_contr...
 * /moveit_sim_hw_interface/joint_model_group: controllers_initi...
 * /moveit_sim_hw_interface/joint_model_group_pose: controllers_initi...
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    controller_spawner (controller_manager/spawner)

ROS_MASTER_URI=http://localhost:11311

process[controller_spawner-1]: started with pid [18481]
usage: spawner [-h] [--stopped] [--wait-for topic] [--namespace ns]
               [--timeout T] [--shutdown-timeout SHUTDOWN_TIMEOUT]
               controller [controller ...]
spawner: error: too few arguments
[controller_spawner-1] process has died [pid 18481, exit code 2, cmd /home/bdelspi/catkin_ws2/src/ros_control/controller_manager/scripts/spawner __name:=controller_spawner __log:=/home/bdelspi/.ros/log/f311e750-f433-11e9-932c-60571895a164/controller_spawner-1.log].
log file: /home/bdelspi/.ros/log/f311e750-f433-11e9-932c-60571895a164/controller_spawner-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

**rosopic list outputs the following:**
bdelspi@bdelspi-Latitude-E5550:~/catkin_ws2$ rostopic list
/attached_collision_object
/collision_object
/execute_trajectory/cancel
/execute_trajectory/feedback
/execute_trajectory/goal
/execute_trajectory/result
/execute_trajectory/status
/feedback_states
/head_mount_kinect/depth_registered/points
/joint_path_command
/joint_states
/joint_trajectory_action/cancel
/joint_trajectory_action/feedback
/joint_trajectory_action/goal
/joint_trajectory_action/result
/joint_trajectory_action/status
/move_group/cancel
/move_group/display_contacts
/move_group/display_planned_path
/move_group/feedback
/move_group/filtered_cloud
/move_group/goal
/move_group/monitored_planning_scene
/move_group/ompl/parameter_descriptions
/move_group/ompl/parameter_updates
/move_group/plan_execution/parameter_descriptions
/move_group/plan_execution/parameter_updates
/move_group/planning_scene_monitor/parameter_descriptions
/move_group/planning_scene_monitor/parameter_updates
/move_group/result
/move_group/sense_for_plan/parameter_descriptions
/move_group/sense_for_plan/parameter_updates
/move_group/status
/move_group/trajectory_execution/parameter_descriptions
/move_group/trajectory_execution/parameter_updates
/pickup/cancel
/pickup/feedback
/pickup/goal
/pickup/result
/pickup/status
/place/cancel
/place/feedback
/place/goal
/place/result
/place/status
/planning_scene
/planning_scene_world
/recognized_object_array
/robot_status
/rosout
/rosout_agg
/rviz_bdelspi_Latitude_E5550_19428_4148281096224171070/motionplanning_planning_scene_monitor/parameter_descriptions
/rviz_bdelspi_Latitude_E5550_19428_4148281096224171070/motionplanning_planning_scene_monitor/parameter_updates
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/feedback
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update_full
/tf
/tf_static
/trajectory_execution_event

Rviz comes up fine and allows me to move the robot and plan. However, when I go to execute I get the following error

[ERROR] [1571688540.109444548]: Unable to identify any set of controllers that can actuate the specified joints: [ joint_1_s joint_2_l joint_3_u joint_4_r joint_5_b joint_6_t ]
[ERROR] [1571688540.109490577]: Known controllers and their joints:

can you please help?

marip8, 2019-10-21:
This error generally means that the MoveIt controller plugin didn't get loaded from your `controllers.yaml` correctly. The first place I would check is in `<your_package_name>_moveit_controller_manager.launch.xml` file. This file should load your `controllers.yaml` file as a ROS parameter like this:

```xml
<launch>
  <arg name="moveit_controller_manager"
       default="moveit_simple_controller_manager/MoveItSimpleControllerManager"/>
  <param name="moveit_controller_manager"
         value="$(arg moveit_controller_manager)"/>

  <rosparam file="PATH/TO/CUSTOM/controllers.yaml"/>
</launch>
```

The MoveIt setup assistant auto-populates this file now but loads the controller configuration from a file called **`ros_controllers.yaml`** which has a different structure and values. My guess is that the current launch file might just be pointing to the wrong controller YAML file

Bdelspi, 2019-10-21:
@marip8 thank you for all your help with this!

I made the change to my motoman_gp8_moveit_controller_manager.launch.xml file. However I am still recieving the same error message... Any thoughts?

I still get the same error:
[ INFO] [1571696874.248334435]: Using planning interface 'OMPL'
[ INFO] [1571696874.252249806]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1571696874.252959638]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1571696874.253671530]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1571696874.254302283]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1571696874.254855329]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1571696874.255440638]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1571696874.255528411]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1571696874.255559241]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1571696874.255584825]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1571696874.255620291]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1571696874.255638746]: Using planning request adapter 'Fix Start State Path Constraints'
[ INFO] [1571696874.365997781]: Stereo is NOT SUPPORTED
[ INFO] [1571696874.366091690]: OpenGl version: 4.6 (GLSL 4.6).
[ INFO] [1571696877.787740034]: Loading robot model 'motoman_gp8'...
[ INFO] [1571696877.844364433]: Loading robot model 'motoman_gp8'...
[ INFO] [1571696877.967054048]: Starting scene monitor
[ INFO] [1571696877.973142368]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1571696877.975109385]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
[ WARN] [1571696879.279774143]: Waiting for gp8_position_controller/joint_trajectory_action to come up
[ INFO] [1571696882.993169589]: Failed to call service get_planning_scene, have you launched move_group? at /tmp/binarydeb/ros-kinetic-moveit-ros-planning-0.9.17/planning_scene_monitor/src/planning_scene_monitor.cpp:491
[ INFO] [1571696883.210564857]: Constructing new MoveGroup connection for group 'gp8_arm' in namespace ''
[ WARN] [1571696885.280101282]: Waiting for gp8_position_controller/joint_trajectory_action to come up
[ERROR] [1571696891.280352281]: Action client not connected: gp8_position_controller/joint_trajectory_action
[ INFO] [1571696891.295412806]: Returned 0 controllers in list
[ INFO] [1571696891.313412248]: Trajectory execution is managing controllers

**motoman_gp8_moveit_controller_manager.launch.xml file:**
<launch>

  
  <arg name="moveit_controller_manager" default="moveit_simple_controller_manager/MoveItSimpleControllerManager" />
  <param name="moveit_controller_manager" value="$(arg moveit_controller_manager)"/>

  
  <rosparam file="$(find motoman_gp8_moveit_config)/config/controllers.yaml"/>
</launch>

**output of rostopic list:**
rostopic list
/attached_collision_object
/collision_object
/execute_trajectory/cancel
/execute_trajectory/feedback
/execute_trajectory/goal
/execute_trajectory/result
/execute_trajectory/status
/feedback_states
/head_mount_kinect/depth_registered/points
/joint_path_command
/joint_states
/joint_trajectory_action/cancel
/joint_trajectory_action/feedback
/joint_trajectory_action/goal
/joint_trajectory_action/result
/joint_trajectory_action/status
/move_group/cancel
/move_group/display_contacts
/move_group/display_planned_path
/move_group/feedback
/move_group/filtered_cloud
/move_group/goal
/move_group/monitored_planning_scene
/move_group/ompl/parameter_descriptions
/move_group/ompl/parameter_updates
/move_group/plan_execution/parameter_descriptions
/move_group/plan_execution/parameter_updates
/move_group/planning_scene_monitor/parameter_descriptions
/move_group/planning_scene_monitor/parameter_updates
/move_group/result
/move_group/sense_for_plan/parameter_descriptions
/move_group/sense_for_plan/parameter_updates
/move_group/status
/move_group/trajectory_execution/parameter_descriptions
/move_group/trajectory_execution/parameter_updates
/pickup/cancel
/pickup/feedback
/pickup/goal
/pickup/result
/pickup/status
/place/cancel
/place/feedback
/place/goal
/place/result
/place/status
/planning_scene
/planning_scene_world
/recognized_object_array
/robot_status
/rosout
/rosout_agg
/rviz_bdelspi_Latitude_E5550_26985_8863903909504226619/motionplanning_planning_scene_monitor/parameter_descriptions
/rviz_bdelspi_Latitude_E5550_26985_8863903909504226619/motionplanning_planning_scene_monitor/parameter_updates
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/feedback
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update
/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update_full
/tf
/tf_static
/trajectory_execution_event

Any thoughts?

> This error generally means that the MoveIt controller plugin didn't get loaded from your `controllers.yaml` correctly. The first place I would check is in `<your_package_name>_moveit_controller_manager.launch.xml` file in your robot's MoveIt config package. This file should load your `controllers.yaml` file as a ROS parameter like this:
> 
> ```
> <launch>
>   <arg name="moveit_controller_manager"
>        default="moveit_simple_controller_manager/MoveItSimpleControllerManager"/>
>   <param name="moveit_controller_manager"
>          value="$(arg moveit_controller_manager)"/>
> 
>   <rosparam file="PATH/TO/CUSTOM/controllers.yaml"/>
> </launch>
> ```
> 
> The MoveIt setup assistant auto-populates this file now but loads the controller configuration from a file called **`ros_controllers.yaml`** which has a different structure and values. My guess is that the current launch file might just be pointing to the wrong controller YAML file

marip8, 2019-10-21:
It looks like the `joint_trajectory_action` topics are in the "global" namespace per the output of `rostopic list`, but the `move_group` node reports that it is looking for those topics in the namespace `gp8_position_controller`

> [ WARN] [1571696879.279774143]: Waiting for gp8_position_controller/joint_trajectory_action to come up

Try using an empty string for the name of the controller in your `controllers.yaml` file, rather than `gp8_position_controller`. This should tell MoveIt to look in the "global" namespace for the `joint_trajectory_action` topics.

Alternatively, you could try to remap those topics (which are provided by the robot controller driver, I believe) into that specific namespace from within the launch file that brings up the driver

```xml
<remap from="joint_trajectory_action" to="gp8_position_controller/joint_trajectory_action"/>
<include file="$(find motoman_driver)../../..">
  ...
</include>
```

gavanderhoorn, 2019-10-22:
@Bdelspi: your initial post confuses me: you mention a `controller_spawner`, but that is only used with `ros_control` compatible drivers (or `gazebo_ros_control`). `motoman_driver` is stand-alone, it does not use `ros_control`.

The file you're trying to use is auto-generated by the MoveIt Setup Assistant and should not be used.

Unfortunately recent changes to the MSA have made this whole process rather confusing -- if you're not already comfortable with setting something like this up, that is.

I'll see if I can create an example MoveIt configuration for the GP8 later today. There are only a few steps and changes needed, but they are across different files, so it's inefficient to have to write it down here.

Bdelspi, 2019-10-22:
@marip8 thank you so much for your help and suggestions with this I will try these now and get back with the results!

@gavanderhoorn understandable, I apologize for the confusion. I am new to this. In researching others that were having similar problems, they included the controller_spawner output, my goal was to provide as much information as possible for us to trouble shoot this together. I will go ahead and try marip8's resolution while waiting for the example MoveIt configuration for the GP8 you discussed. 

Thank you both again for all your help @marip8 & @gavanderhoorn!

Bdelspi, 2019-10-22:
@marip8 & @gavanderhoorn you two are God sends and absolutely incredible! We are operational!

@marip8 switching the name of the controller to an empty string along along with @gavanderhoorn suggestion of not using the robot_interface_streaming_yrc1000.launch did the trick! YOU TWO ARE AWESOME! Thank you, thank you, thank you both!

marip8, 2019-11-04:
@Bdelspi not a problem; glad you're robot is up and running. Feel free to reopen this issue or post a new one if you run into problems in the future

closed 2019-11-04 by marip8