ros-industrial/motoman #302: yrc1000 MotoRos not started (5008)

emielke12, 2019-10-30:
I am attempting to follow the tutorials [here](http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer) and [here](http://wiki.ros.org/motoman_driver/Tutorials/Usage), but am running into this issue. When I attempt to call the `/robot_enable` service, I am met with this message:

`Failed to set TrajectoryMode: Not Ready (5) : MotoRos not started (5008)`

According to the tutorial, this means that the Inform job wasn't loaded. I followed the instructions on the tutorial numerous times. Is there any trick to getting this loaded?

The green `servo on` light lights up on the pendant when I run the service call, and the brakes disengage, but it gives me that error code.

EricMarcil, 2019-10-30:
Did you do Step 3: Installing INFORM Code?
When you list the robot job on the pendant, you should see the job: INIT_ROS in the list.

emielke12, 2019-10-30:
Yes, I did step 3. I can see INIT_ROS in the job list

EricMarcil, 2019-10-30:
Then it must be something else.  I don't have access to the code right now.  I would suggest that you look at the MotoRos drive source code and look for the error code 5008.  You can then see what condition might be causing it's generation.

EricMarcil, 2019-10-30:
I just looked at some note.  Basically, your controller is probably in a state where you can't run job.
Try running the job directly from the pendant.  You probably won't be able to start the Job and you'll have a message at the bottom of the pendant telling you why.  Last time I helped someone with a similar problem, they had a message to check second home position.  They had just received the robot and it had probably moved a bit during shipping.

emielke12, 2019-10-31:
That was it! Thanks for the help and quick responses!

closed 2019-10-31 by emielke12