ros-industrial/motoman #303: joint naming

emielke12, 2019-10-31:
I am trying to get moveit running on a motoman gp8 (yrc1000 controller). I am able to get everything pretty much up and running except for a naming issue. It appears that the controller is expecting the joint names to be in the `[joint_s, joint_l, ..., joint_t]` format, whereas in the `.urdf`, these joints are labeled as `[joint_1_s, joint_2_l, ..., joint_6_t]`.

Should the `.urdf` joint names be changed to match what the controller is expecting? I tried running the `rosrun motoman_driver move_to_joint.py` while using both naming formats and got the error `Start and End position joint_names mismatch` when trying to use the `joint_#_x` naming format, but was able to run with no problems using `joint_x` naming convention.

The names were placed in a `.yaml` file and loaded with the motion streaming interface launch file. Is there another way around this? The ROS wiki tutorial wasn't super clear on this step...

gavanderhoorn, 2019-11-05:
>  It appears that the controller is expecting the joint names to be in the `[joint_s, joint_l, ..., joint_t]` format, whereas in the `.urdf`, these joints are labeled as `[joint_1_s, joint_2_l, ..., joint_6_t]`.

I'm not sure I completely understand. If "the controller" refers to your YRC1000, then that would seem to be incorrect: the YRC1000 doesn't actually use any joint names, it only uses joint indices (at the MotoROS level, that is).

Could you please provide some more information on your system, so we can try to diagnose what problem you are encountering?

emielke12, 2019-11-05:
Sorry for the imprecise language, I'll try to clarify.

I have set up a moveit configuration for the motoman gp8 to be able to send `JointTrajectory` messages to the robot. According to the tutorials, I need to launch `robot_interface_streaming_yrc1000.launch` from the `motoman_driver` package. If I follow correctly, this launch file expects the robot joint names to be set in the `robot_description` on the rosparam server. I have a launch file that calls the streaming interface, robot state publisher, and the required moveit launch files, and also loads the joint names into the `robot_description` on the rosparam server. 

When I run this launch file, I get the following error, for all joints s,l,u,r,b,t repeated over and over:
`Joint 'joint_s' not found in model 'gp8'`. I'm assuming this is because I set the joint names to be of the format `joint_x` in the robot description, but the `.urdf` from this repository has the joints in a `joint_#_x` format.

Does that clarify my issue a little?

emielke12, 2019-11-06:
Nevermind, I see where I was getting mixed up. I thought that the controller was expecting `joint_x` but thanks to your comment, I see it's not taking the joint names from the robot description. Making the robot description names match those in the `.urdf` works for streaming a trajectory.

Sorry for the inconvenience.

closed 2019-11-06 by emielke12