ros-industrial/motoman #304: We want to know did GP8 can work like a streaming smoothly? like the questions below?

namuhevil, 2019-11-05:
[image]

- Basically, if I use this way to interact with Rviz, 
group = moveit_commander.MoveGroupCommander("manipulator")
	Only specified desired joint/pose can be accepted. For example,
pose_target.position.x = 0.1 #0.1 
pose_target.position.y = 0.2 #0.2 
pose_target.position.z = 0.6 #0.6
group.go(wait=True)
- I already tried to find existing API from moveit_commander (Python) but it seems that there is no API that can work with trajectory message.
- Link of existing API from moveit_commander:
docs.ros.org/jade/api/moveit_commander/html/classmoveit__commander_1_1move__group_1_1MoveGroupCommander.html 

 Question: Thus, I would like to ask you if there is some way to give trajectory commands (prepared as the code below here) based on messages to simulated robot in Rviz?

[image]

Can it work??

gavanderhoorn, 2019-11-05:
>  Thus, I would like to ask you if there is some way to give trajectory commands (prepared as the code below here) based on messages to simulated robot in Rviz?

please clarify whether you *only* have a robot in RViz, or actual hardware and have started the driver.

namuhevil, 2019-11-06:
I just have RViz, I don't have the hardware

gavanderhoorn, 2019-11-14:
I've just re-read your question and I'm afraid I don't understand it completely.

It's possible to send `JointTrajectory` goals to the driver, and you can create those in the way you show.

gavanderhoorn, 2019-11-14:
As in any case this doesn't look like an issue with the packages in this repository, I'm going to close this issue.

Feel free to keep commenting of course @namuhevil.

closed 2019-11-14 by gavanderhoorn