ros-industrial/motoman #306: I want to ask following question about real gp8 action

namuhevil, 2019-11-14:
The action we want to do is to give two commands for position control.
1. First, we will give position A to let the robot to reach. Before reaching the target position A, we will give another command to reach position B. 
- In the simulation part, in RVIZ environment, it will reach position B directly instead of reaching position A. Besides, the velocity will not reach 0 in those actions.
- But in actual situation, the GP series manipulator will stop and appear an error like it cannot do that action.

Does anyone know why the commands in the manipulator cannot do like what we did in RVIZ simulation?
Is there any constraint for it? or we should set up something for it?

gavanderhoorn, 2019-11-14:
I'm not exactly sure what you are asking, could you try to make it a bit clearer?

> Does anyone know why the commands in the manipulator cannot do like what we did in RVIZ simulation?

as a general comment: if by "in RViz simulation" you mean the `industrial_robot_simulator` or just a plain MoveIt `fake_controller`, then the answer would be: the real robot has inertia, mass, maximum acceleration and jerk limits, etc, etc. `industrial_robot_simulator` and the MoveIt `fake_controller` do not take any of that into account, they just show a kinematic play-out.

Real robots cannot execute instanteneous changes of direction, as that would lead to infinite jerk. Kinematic play-out / position `JointState` visualisation does not limit the robot in any way.

Finally:

> RVIZ simulation

RViz is not a simulator. It's a data visualisation tool. It only "draws" a 3D scene based on datastreams coming in over ROS topics.

If you need dynamics, you'd have to look at actual simulators such as Gazebo, V-REP and Webots.

gavanderhoorn, 2019-11-14:
As I don't believe this is an issue with the packages in this repository, I'm going to close this issue.

That does not mean we cannot continue discussing here of course, so feel free to post additional comments.

closed 2019-11-14 by gavanderhoorn