ros-industrial/motoman #307: Motoplus SDK support

impet14, 2019-11-16:
Hi guys,

Since our group would like to purchase Motoplus SDK. However, we have contact yaskawa by email and contact form from Yaskawa website. We have not got reply yet.
Could anyone suggest me how to contact Yaskawa officer for purshasing Motoplus SDK?
We are from Japan.

Thank you in advance.

gavanderhoorn, 2019-11-17:
I'm closing this, as it's not an issue with the packages in this repository.

That does not mean we're ignoring you're question @impet14.

Please feel free to keep commenting on this issue, and I've asked @EricMarcil and @ted-miller to take a look at it.

closed 2019-11-17 by gavanderhoorn

ted-miller, 2019-11-18:
Hello @impet14 

I recommend contacting the head office in Japan at +81-93-645-8801.  The product name in Japan is "MotoPlusIDE".

I hope this helps.

impet14, 2019-11-19:
Thank you very much sir