ros-industrial/motoman #308: GP8 with YRC1000micro Matlab control

Bdelspi, 2019-11-19:
Has anyone been able to/ seen modern examples of how to control their motoman via matlab? 

Currently:
- MATLAB 2019b with
      -- robotics toolbox
      --ROS toolbox
- YRC1000micro
-GP8 manipulator

Running through the workflow, I am able to make connection with the YRC1000micro and ROS via rosint (confirmed through pinging the robot and executing rosparam list).

Our obstacle is in initiating robot motion, for example when running rostopic /joint_trajectory_action, MATLAB returns  The "/joint_trajectory_action" operation is not supported.

Any suggestions? Is their Motoman support in Matlab? For example, like the Kuka sunrise toolbox for the Kuka robots...

Bdelspi, 2019-11-19:
> Has anyone been able to/ seen modern examples of how to control their motoman via matlab?
>
> Currently:
>
> * MATLAB 2019b with
>   -- robotics toolbox
>   --ROS toolbox
> * YRC1000micro
>   -GP8 manipulator [...]

Also, the following show as available in MATLAB. However, not all seem to be supported...
[image]

gavanderhoorn, 2019-11-19:
The packages in this repository support a regular ROS API for robot drivers (such as a `JointState`s topic and a `FollowJointTrajectory` action server).

If those can be used from Matlab, you should be able to control the robot.

There is no specific support for Matlab here.

gavanderhoorn, 2019-11-19:
As this is not an issue with the packages in this repository (but a configuration/integration issue on the user's side), I'm going to close this.

Feel free to keep commenting on the issue of course.

gavanderhoorn, 2019-11-19:
Some observations:

> `rostopic /joint_trajectory_action`

that is not a valid `rostopic` command line. So the problem is with how `rostopic` is invoked. Not with `motoman_driver`.

> `rostopic echo /joint_trajectory_action`

This is also not a valid `rostopic` command line, as `/joint_trajectory_action` is the base namespace of the `FollowJointTrajectory` action server. `rostopic` cannot `echo` namespaces, only topics.

Furthermore, only certain topics in that namespace would publish something.

closed 2019-11-19 by gavanderhoorn

Bdelspi, 2019-11-21:
> As this is not an issue with the packages in this repository (but a configuration/integration issue on the user's side), I'm going to close this.
> 
> Feel free to keep commenting on the issue of course.

@gavanderhoorn Thank you for all your help with this! Yeah, I have been unable to find a suggested workflow for integrating existing ROS industrial workpackages into Matlab and am working on creating one. As pointed out by [Safeea and Neto](https://ieeexplore.ieee.org/document/8542757) for their [Kuka sunrise toolbox](https://github.com/Modi1987/KST-Kuka-Sunrise-Toolbox) having a Matlab interface could enable robotic applications to those with basic Matlab skills, even if they are not programming experts...

Thank you again for all your help with this gavanderhoorn!

gavanderhoorn, 2019-11-21:
Well as I wrote, if Matlab supports `actionlib`, interfacing with the driver would be as simple as populating a `FollowJointTrajectory` goal and submitting it to the driver.

A quick Google suggests it does support `actionlib`: [Creating ROS action client](https://www.mathworks.com/help/ros/ref/simpleactionclient.html), so this should be possible. [sendGoal(..)](https://www.mathworks.com/help///ros/ref/sendgoal.html) should be used to submit the goal to the server.

Be sure to call the `enable_robot` server before sending the goal, or otherwise it'll be returned with a `FAIL` immediately.

Bdelspi, 2019-11-22:
@gavanderhoorn you are awesome! Thank you so much! Yeah, I think the obstacle is stemming from the need to create [ROS custom messages in Matlab](https://www.mathworks.com/help/ros/ug/ros-custom-message-support.html;jsessionid=f871af4b4be68d009c34bf2a49de?s_tid=answers_rc2-1_p4) (since for example FollowJointTrajectory is not supported by Matlab out-of-the-box), and have been learning about these today. My plan is to implement this tomorrow and will keep you up to date as we progress. 

Thank you again for all that you do gavanderhoorn!

Gloriabhsfer, 2019-11-29:
@gavanderhoorn 

Hi, gavanderhoorn:
Thanks for your reply to this issue. In my understanding, the motorman driver is the package that can encode and decode the control information into some data that YRC100 can receive. And we can use ROS to establish the connection between our host computer and the controller box(YRC 1000). In a Linux terminal we can establish the connection through the IP address of the robot. And we can enable or disable the controller through "rosservice call /robot_enable". As you mentioned in this terminal, we can see the rosservice list through the Matlab command line, and find /robot_enable inside it. But it will return us 

> Cannot find a MATLAB message class for type std_srvs/TriggerRequest.

 I think that's because Matlab cannot support type in C++, so we cannot enable robot and sent the message through Matlab.

Do you have any idea about that? 

Thanks,
Gloria

gavanderhoorn, 2019-11-30:
> In my understanding, the motorman driver is the package that can encode and decode the control information into some data that YRC100 can receive. 

That is roughly what it does, yes.

> And we can use ROS to establish the connection between our host computer and the controller box(YRC 1000). 

To be precise: no, ROS is not used between the YRC and the ROS PC. A different communications protocol is used there.

> In a Linux terminal we can establish the connection through the IP address of the robot. And we can enable or disable the controller through "rosservice call /robot_enable". As you mentioned in this [...]

If with "type in C++" you are referring to messages/services and/or actions, then I believe there may be a misunderstanding: ROS messages are not specific to any particular language. The `.msg`, `.srv` and `.action` format is a custom format, plain text. Only after being processed by a code generator (ie: `gencpp`, `genpy` et al) do we get their representation in a specific programming language.

> so we cannot enable robot and sent the message through Matlab.

For matlab, you'll have to figure out how to make it use certain messages or services. The specific error message you show seems to indicate that there is no support for [std_srvs/Trigger](http://docs.ros.org/api/std_srvs/html/srv/Trigger.html) (or you have not configured things correctly). This particular service is part of the `std_srvs` package, which is a core ROS package. I would be surprised if Mathworks has not built in support for those services.

As I wrote earlier: I have no experience with Matlab and ROS. I just googled for 5 minutes with some keywords.

I would recommend you ask Mathworks support to help you in this case. In the end, all you need to do is call a service and submit an action goal (with a trajectory). There is no requirement for C++ (or any other programming language outside of what Matlab can already do).

---

Note that for all of the above to work, you'll still need a Linux machine running `motoman_driver` and the rest of the required nodes.

gavanderhoorn, 2019-11-30:
And just to repeat myself as a form of expectation management:

> In any case: only trajectory execution is supported (by default). No direct streaming of jointstates/setpoints.

Don't expect real-time control of the robot (a la simulink/xpc).

That's not what the driver allows.

Bdelspi, 2019-12-03:
Thank you @gavanderhoorn, we have reached out to Mathworks and have been working with them, but have been unable to reach a resolution on this. We will keep you up to date as we progress. Thank you again for all your help!

gavanderhoorn, 2019-12-03:
I'm a bit surprised Mathworks is having such a hard time supporting you in calling a service with a type from `std_srvs` and/or submitting an action goal also from a standard ROS package.