ros-industrial/motoman #310: How did you control multiple ip address!?

abhishekpg01, 2019-12-18:
Hi, 
I'm trying to implement project in a dual arm aubo_i5 robot. But my both arms hav different ip address. How did u manage this multiple ip for each device. ??

gavanderhoorn, 2019-12-18:
This repository hosts packages for Yaskawa Motoman robots.

Please don't post off-topic issues here on the tracker.

closed 2019-12-18 by gavanderhoorn

abhishekpg01, 2019-12-18:
I just want to know if your robot's each arm have same ip adress or not. If you have different ip, how did you manage to merge into single ip??

gavanderhoorn, 2019-12-18:
How will knowing this about Yaskawa Motoman controllers help you do this with an Aubo robot?

The two brands do not share any hw, nor any software.

abhishekpg01, 2019-12-18:
As far as I understood the right arm and left are using same motoman controller. Or is the controller specially designed for dual arm robot. I need only one information, how did u manage to connect all the robots with different ip adress using a single ip address. I want to implement the same methode in my research

gavanderhoorn, 2019-12-18:
Yaskawa Motoman controllers support connecting multiple robots configured as multiple motion groups. This is in fact supported by many (if not all) industrial robot controllers.

There is only a single controller, with a single IP.

I still don't understand how this will help you with Aubo controllers, as they lack the hardware to do this, but there you go.