ros-industrial/motoman #311: Disable collision link_2_l and link_4_r in GP7 moveit_setup_assistant

SCH923, 2019-12-26:
I had a problem when I managed to controll GP7 with moveit. 
If you don't set the collision disable between  link_2_l and link_4_r in setup_assistant.launch,
there is large space robot can't approach.

closed 2019-12-26 by SCH923

gavanderhoorn, 2019-12-26:
Could you please provide a little more detail?

Did you close this issue because you resolved your issue, or was this just to let us know what you feel should be done?

SCH923, 2019-12-31:
Thank you for reply. I'm always helped by your good comments everywhere, gavanderhoorn.(respect very much!!)

When I posted this , I wondered which repository I should submit ros-industrial/motoman or moveit. This problem is about setting to controll motoman GP7 with yrc1000u, ros-industrial and moveit setup assistant. 

OS:Ubuntu 18.04
ROS: melodic

>Did you resolved your issue? 
Yes.
I resolved this issue by using soooo much time...

>was this just to let us know?
Yes.
I wanted you and users "using yasukawa GP7" know about this problem.
Not moveit users but ros-industrial users especially yasukawa GP7.
but I did not know this is issue or not. 
And problem was resolved by ourselves. (not perfect)
So I closed this issue for just telling information.

I have much information to controll yasukawa GP7.
Because I struggled for a month to connect real robot GP7, ros-i GP7  driver and moveit controller.
I wanted the tutorial for ros-industrial users not only simulated robot in rviz but real industrial robot .....

gavanderhoorn, 2020-01-02:
> I resolved this issue by using soooo much time...

It would have been OK to ask questions earlier. That could have saved you quite some time it appears.

> I wanted you and users know about this problem.
> Not moveit users but ros-industrial users especially GP7.
> but I did not know this is issue or not.

I'm somewhat confused: is the issue you solved with the GP7 specifically, or with setting up the driver with YRCu controllers?

> I have much information to controll GP7.
> Because I struggled for a month to connect real robot, ros-i driver and moveit controller.

If you have specific points that would need clarifying, perhaps we can work to address those.

> I wanted the tutorial for ros-industrial users not only simulated robot in rviz but real industrial robot

There are a few tutorials but those definitely could use some attention.

SCH923, 2020-01-03:
>It would have been OK to ask questions earlier. That could have saved you quite some time it appears.

I will do so next time. Thank you.

>is the issue you solved with the GP7 specifically, or with setting up the driver with YRCu controllers?

I have found with the GP7 only. This is not a problem with YRCu controllers.
This problem is in this step.
http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/setup_assistant/setup_assistant_tutorial.html#step-2-generate-self-collision-matrix

gavanderhoorn, 2020-01-03:
But what is the problem exactly?

SCH923, 2020-01-04:
> what is the problem exactly?

Compatibility between [Optimized Self-Collision Checking](http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/setup_assistant/setup_assistant_tutorial.html#step-2-generate-self-collision-matrix) and GP7 collision model.

>If you don't set the collision disable between link_2_l and link_4_r in setup_assistant.launch,

When I [clicked "Generate Collision Matrix"](http://docs.ros.org/melodic/api/moveit_tutorials/html/_images/setup_assistant_panda_self_collisions.png), there was no settings between link_2_l and link_4_r 
in the initial state with GP7 collision model.
(I tried sampling density both lowest and highest.)

>there is large space robot can't approach.

After all settings, when I moved GP7 to some x,y,z coordinates with [this command](http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/move_group_python_interface/move_group_python_interface_tutorial.html#planning-to-a-pose-goal), 
there were some points that robot didn't go because of that auto collision generator settings, though GP7 could move to those coordinates with Yasukawa remote pendant.

cjue, 2021-12-13:
The OPs issue was caused by the same problem mentioned here for the GP4:
https://github.com/ros-industrial/motoman/pull/433#issuecomment-988610550

The collision mesh for link2 is too large / fills in the concave geometry and restricts the range of movement for joint3 significantly. Should I check for other GP models for the same issue and create an issue just for this issue mentioning all affected robots?