ros-industrial/motoman #313: YRC1000micro with discrete input output gripper

Bdelspi, 2020-01-16:
Looking at the [simple_message folder](https://github.com/ros-industrial/motoman/tree/kinetic-devel/motoman_driver/src/simple_message) inside the motoman_driver folder, it appears we can send commands and control a discrete I/O gripper (e.g. [EGP40-N-N-B](https://schunk.com/us_en/gripping-systems/product/28088-0310940-egp-40-n-n-b/)) using these files. Can anyone confirm this posit? Is there ROS documentation that explains this further anyone can please suggest?

Bdelspi, 2020-01-16:
It seems this question was also asked [here](https://answers.ros.org/question/323058/control-gripper-with-motoman_driver/), but does not provide a solution...

EricMarcil, 2020-01-16:
Hi,
We have implemented the read/write IO on the MotoRos driver but I don't think that there has been an official release to support it on the ROS side (motoman_driver).  
I do know that some people have used it by implemented it in their own branch.
If you look at the issue #233 Write/read IO digitals dx100 there is a link to sinaiaranda-CIDESI that has implemented it.

gavanderhoorn, 2020-01-16:
I have actually implemented the ROS side of this IO support, and quite some time ago. The base branch I did that in is `io_support` in my personal fork of `motoman`, with an example service server implementation in [io_support_kinetic-devel_head_yuvraj](https://github.com/gavanderhoorn/motoman/commits/io_support_kinetic-devel_head_yuvraj).

It was not merged into mainline as it lacks support for non-multi-group setups.

@marip8: have you been able to take a look at this?

marip8, 2020-01-17:
I haven't had an opportunity to work on that feature yet. I'm tied up for the next few weeks, but I'm hoping to incorporate it in a PR in the next month or two.

Bdelspi, 2020-01-18:
@EricMarcil @gavanderhoorn @marip8 Thank you all for your help with this! I plan on spending the weekend diving into this and have been reading over the resources:

1) [Industrialsupported_hardware](http://wiki.ros.org/Industrial/supported_hardware)
2) [Create a simple EtherCAT IO network (Implementation Notes)](http://wiki.ros.org/Industrial/Tutorials/Simple_EtherCAT_Network_IO)
3) [ROS wrapper for the Simple Open EtherCAT Master SOEM](https://wiki.ros.org/soem)

Specifically, in [Industrialsupported_hardware](http://wiki.ros.org/Industrial/supported_hardware) Sect. 1.3 IO Networks, it states "In the Motoman stack, limited support for controlling basic I/O has been implemented. This can be used for general gripper control." but have been unable in locating any documentation which greater details this. Any suggestions?

Thank you all again for all your help with this! You all have been awesome and I greatly appreciate all your help!

Bdelspi, 2020-02-13:
@marshallpowell97 any thoughts?

gavanderhoorn, 2020-02-14:
@Bdelspi: I'm not entirely sure why you are @-mentioning @marshallpowel97 here. He works for Staubli.

marshallpowell97, 2020-02-14:
@gavanderhoorn Yesterday I was helping @Bdelspi with the `staubli_val3_driver` on his machine, and we were discussing the possibility of implementing some kind of IO control similar to [`motoman_read_single_io.cpp`](https://github.com/ros-industrial/motoman/blob/kinetic-devel/motoman_driver/src/simple_message/motoman_read_single_io.cpp). I can open an issue in the other tracker so we don't bring this one off-topic.

Bdelspi, 2020-02-14:
@gavanderhoorn I apologize, I should have provided more detail. @marshallpowell97 and I were working together yesterday on our lab's Staubli and looking at ways of implementing IO control for the [Schunk EGP 40-N-N-B gripper](https://schunk.com/ro_en/gripping-systems/product/28088-0310940-egp-40-n-n-b/)...

Thank you both again for all your help with this!

gavanderhoorn, 2020-10-05:
I'm going to link this issue with PR #355, as that would finally add ROS support for reading and writing individual IOs on the controller.

Future updates will add reading/writing bytes and words.

closed 2020-10-07 by gavanderhoorn