ros-industrial/motoman #314: error material 'yaskawa_blue' is not unique when using multiple robots

EvilPictureBook, 2020-02-09:
When using multiple robots each seems to add a definition for yaskawa_blue, causing this error:

`[ERROR] [1581285016.618736830]: material 'yaskawa_blue' is not unique.`

example xacro:
```
<?xml version="1.0" ?>

<robot name="my_robot_workspace" xmlns:xacro="http://ros.org/wiki/xacro">
    <xacro:include filename="$(find motoman_gp25_support)/urdf/gp25_macro.xacro" />

    <xacro:motoman_gp25 prefix="first_"/>
    <xacro:motoman_gp25 prefix="second_"/>

    <link name="world" />

    <joint name="world-first_base_link" type="fixed">
        <origin xyz="0 0 0" rpy="0 0 0"/>
        <parent link="world"/>
        <child link="first_base_link"/>
    </joint>

    <joint name="world-second_base_link" type="fixed">
        <origin xyz="0 1.5 0" rpy="0 0 0"/>
        <parent link="world"/>
        <child link="second_base_link"/>
    </joint>

</robot>

```

gavanderhoorn, 2020-02-10:
Yes, this is an unfortunate side-effect of all macros redefining `yaskawa_blue`.

We could introduce a `motoman_resources` package (similar to `fanuc_resources`, `abb_resources`, etc) which provides this definition once and then reuse it everywhere.

@EvilPictureBook: could you please update your OP with information about your ROS version? Afaik newer versions of `xacro` should namespace definitions like this with the scope they are defined in (so `yaskawa_blue` would be local to `motoman_gp25`).

EvilPictureBook, 2020-02-12:
> could you please update your OP with information about your ROS version?

Thanks, updated, and I also provided the launch and test files in case I'm doing something wrong there.

Please let me know if there's something I could test to see why the name-spacing isn't working.

Doomerdinger, 2020-12-07:
Before I go and add a resources package and change everything to use it, am I understanding correctly that @gavanderhoorn you don't think this should be an issue with newer `xacro` versions since the definitions should be namespaced?

I'm using the latest as far as I can tell, and while most things namespace properly the material definitions still seem to break.

gavanderhoorn, 2025-03-19:
There is now a `motoman_resources` (since #378), so closing this.

closed 2025-03-19 by gavanderhoorn