ros-industrial/motoman #315: [ERROR] [1582746251.37862388]: Failed to set TrajectoryMode: Not Ready (5) : Unable to enable drive power (5006)

alexander-jh, 2020-02-26:
Hello, so running motoman on a MH-5 from a FS-100 controller on Melodic 1.14.3. I've gotten roadblocked with the below error trying to enable:

rosservice call robot_enable

[ERROR] [1582746251.37862388]: Failed to set TrajectoryMode: Not Ready (5) : Unable to enable drive power (5006)
[ERROR] [1582746251.557802159]: Motoman robot was NOT enabled. Please re-examine and retry.

I've followed the remedy instruction as far as they'd take me. The robot_enable will turn the servo motors on, then it instantly fails.

Any feedback would be greatly appreciated.

gavanderhoorn, 2020-02-26:
Please provide a little more information:

 - version of system software on the FS100
 - safety equipment attached to the controller

any errors / warnings shown on the TP?

alexander-jh, 2020-02-26:
There are no errors/warnings on the TP. The only thing other than the linux machine connected direct via ethernet is the teach pendant. Not outside safety devices either.

System: FS3.300.00A
Parameter: 1.52
Language: 3.30

gavanderhoorn, 2020-02-26:
It would probably help if you could connect with Telnet to the controller, log in and then try calling the service.

That should provide us with a little more information.

EricMarcil, 2020-02-26:
The error message is saying that the servos are not turning on.  
You are saying that the servo DO turn on?  

How long do the servo take to turn on? There is a time out of 5 sec. which is normally sufficient for the servos to turn on.

Try to turn on the servos from the pendant before starting ROS.
- Switch the pendant to PLAY mode.  
- Select the INIT_ROS job
- Press the ServoOn button
- Switch the pendant to REMOTE mode.
Then try starting trajectory again.

alexander-jh, 2020-02-26:
> It would be helpful if you could connect with Telnet to the controller, log in and then try calling the service.
> 
> That should provide us with a little more information.

We’re presently waiting to hear back from Yaskawa for the default VxWorks login.

> The error message is saying that the servos are not turning on.
> You are saying that the servo DO turn on?
>
> How long do the servo take to turn on? There is a time out of 5 sec. which is normally [...]

Calling robot_enable turns the servos on, then there’s about a 5-10 second delay and the error comes. I will try with ROS_INIT tomorrow.

EricMarcil, 2020-02-26:
telnet login is:
MOTOMANrobot
password is:
MOTOMANrobot

acbuynak, 2020-02-27:
Hello! I am working with @alexander-jh on this issue.

When I login to VxWorks using info provided by @EricMarcil, I get a continuous output of..
`Failed to update controller status. Check robot parameters.`
After a few minutes the connection times out and stops.

> Try to turn on the servos from the pendant before starting ROS.
>- Switch the pendant to PLAY mode.
>- Select the INIT_ROS job
>- Press the ServoOn button
>- Switch the pendant to REMOTE mode.
>- Then try starting trajectory again.

There is no change in the error or the robot when the following trajectory command is sent.
` rosrun motoman_driver move_to_joint.py "[1, 0, 0, 0, 0, 0]" `

I did notice that `OUT#0890` came on when I send the `robot_disable` command.

EricMarcil, 2020-02-27:
It indicates that the mpReadIO is failing. That is unusual...

@acbuynak @alexander-jh:
Did you follow all the installation step of the installation tutorial?
[http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer](http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer
)

Can you e-mail me a backup of the controller parameter file (ALL.PRM) at eric.marcil71@motoman.com so that I can check parameters.

gavanderhoorn, 2020-02-27:
This starts to sound like the controller was not (correctly) configured for use with MotoROS.

acbuynak, 2020-02-27:
@EricMarcil I've had the email address you provided bounce back. As well as a few variations on arrangement I've tried. Is the `eric.marcil71@motoman.com` you gave correct?

We tried changing parameter `S2C1119` from `0` to `2` to (what we think) allows a telnet connection.

When both TP & Telnet are connected to the controller. The telnet terminal outputs:
`Failed to update controller status. Check robot parameters.`

To work around this, the TP was psychically disconnected. This allowed a telnet connection. This done, we attempted to send a trajectory command.
The telnet terminal output a stream of:
`Waiting for Robot Alarms to clear....`

EricMarcil, 2020-02-27:
@acbuynak Sorry, I mixed up my e-mail.  The correct one is: eric.marcil@motoman.com
If by the TP, you mean the Teach Pendant, you need connect it unless you have a bypass plug.  That's probably why you have alarms that can't be cleared.

acbuynak, 2020-02-27:
I've fired off our parameter file to @EricMarcil for review.

In checking our controller setup further today, I realized/discovered two things..
-> I did have an issue where I could not load the actual INIT_ROS.JBI file due to [this EOL error](http://wiki.ros.org/motoman_driver/Troubleshooting#ERROR_3200:_NOP_or_END_instruction_not_found). I tried fixing it, but ended up manually typing in the job on the TP (teach pendant). 
When I download the JBI file and compare it to the [provided job file](https://github.com/ros-industrial/motoman/tree/kinetic-devel/motoman_driver/Inform/DX200%2C%20FS100%2C%20YRC1000/single_arm), they are the same.

-> In [this same tutorial step](http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer#Installing_INFORM_Code), I never use the IONAME.DAT file beyond saving it on my flashdrive. When comparing it to the IONAME.DAT file I downloaded from my controller just now, I noticed that that lines 482 & 483 are different.

I am missing the following...
IONAME.DAT Line 482: `ROS_READY,ROS_DONE,INIT_DONE,CONNECTION_SRV`
IONAME.DAT Line 483: `MOTION_SERVER,STATE_SERVER,IO_SERVER,FAILURE`

This may be the root cause.
How would I upload a modified IONAME.DAT file or change it directly on the TP?

gavanderhoorn, 2020-02-28:
The IO file just contains the names / comments for the IO elements. It's not really important.

I would suggest to wait for @EricMarcil to verify your controller has the correct parameter values for use of MotoROS.

If it has been correctly configured, then we can take further steps to try and diagnose the problem.

impet14, 2020-03-05:
I use fs100 version 3.30 with driver version 1.90. I got error print in telnet "Failed to update controller status.  Check robot parameters." as well. But once I use driver version 1.70 it is fine.
Thank you in advance

gavanderhoorn, 2020-03-05:
Hm, thanks for the additional datapoint @impet14.

@EricMarcil: have you been able to take a look at the backup @alexander-jh provided?

gavanderhoorn, 2020-03-05:
Here is the diff between MotoROS `v1.7.0` and `v1.9.0`: https://github.com/ros-industrial/motoman/compare/4b9fd56d...f4fb6398.

Unfortunately it appears to be impossible to limit this to a certain directory only, so Github shows changes to the entire repository between those commits.

To see changes to just `motoman_driver/MotoPlus`, use the git CLI, navigate to the `motoman` clone, and run this command:

```
git diff 4b9fd56d..f4fb6398 motoman_driver/MotoPlus
```

This gives:

<details>
<summary>Diff between v1.7.0 and v1.9.0 of MotoROS (click to show)</summary>

```patch
diff --git a/motoman_driver/MotoPlus/Controller.c b/motoman_driver/MotoPlus/Controller.c
index 4956097..6c6b518 100644
--- a/motoman_driver/MotoPlus/Controller.c
+++ b/motoman_driver/MotoPlus/Controller.c
@@ -92,61 +92,6 @@ void reportVersionInfoToController()
 #endif
 }
 
-// Verify most of the setup parameters of the robot controller.
-// Please note that some parameters cannot be checked, such as
-// the parameter(s) which enable this task to run.
-// Returns FALSE if a critical parameter is not set such that it
-// will prevent intialization.
-BOOL Ros_Controller_CheckSetup()
-{
-	int parameterValidationCode;
-
-	parameterValidationCode = ValidateMotoRosSetupParameters();
-	switch (parameterValidationCode)
-	{
-	case MOTOROS_SETUP_OK: return TRUE;
-
-	case MOTOROS_SETUP_RS0: 
-		mpSetAlarm(MOTOROS_SETUPERROR_ALARMCODE, "MotoROS Cfg: Set RS000=2", parameterValidationCode);
-		return TRUE;
-
-	case MOTOROS_SETUP_S2C541:
-		mpSetAlarm(MOTOROS_SETUPERROR_ALARMCODE, "MotoROS Cfg: Set S2C541=0", parameterValidationCode);
-		return TRUE;
-
-	case MOTOROS_SETUP_S2C542:
-		mpSetAlarm(MOTOROS_SETUPERROR_ALARMCODE, "MotoROS Cfg: Set S2C542=0", parameterValidationCode);
-		return TRUE;
-
-	case MOTOROS_SETUP_S2C1100:
-		mpSetAlarm(MOTOROS_SETUPERROR_ALARMCODE, "MotoROS Cfg: Set S2C1100=1", parameterValidationCode);
-		return FALSE;
-
-	case MOTOROS_SETUP_S2C1103:
-		mpSetAlarm(MOTOROS_SETUPERROR_ALARMCODE, "MotoROS Cfg: Set S2C1103=2", parameterValidationCode);
-		return FALSE;
-
-	case MOTOROS_SETUP_S2C1117:
-		mpSetAlarm(MOTOROS_SETUPERROR_ALARMCODE, "MotoROS Cfg: Set S2C1117=1", parameterValidationCode);
-		return FALSE;
-
-	case MOTOROS_SETUP_S2C1119:
-		mpSetAlarm(MOTOROS_SETUPERROR_ALARMCODE, "MotoROS Cfg: Set S2C1119=0 or 2", parameterValidationCode);
-		return TRUE;
-
-	case MOTOROS_SETUP_NotCompatibleWithPFL:
-		mpSetAlarm(MOTOROS_SETUPERROR_ALARMCODE, "MotoROS not compatible with PFL", parameterValidationCode);
-		return FALSE;
-
[... 1500 lines cut ...]
index 0000000..810175e
Binary files /dev/null and b/motoman_driver/MotoPlus/output/DX200/MotoRosDX2_/MotoRosDX2_190.out differ
diff --git a/motoman_driver/MotoPlus/output/FS100/MotoRosFS100/MotoRosFS100.out b/motoman_driver/MotoPlus/output/FS100/MotoRosFS100/MotoRosFS100.out
deleted file mode 100644
index 27c94af..0000000
Binary files a/motoman_driver/MotoPlus/output/FS100/MotoRosFS100/MotoRosFS100.out and /dev/null differ
diff --git a/motoman_driver/MotoPlus/output/FS100/MotoRosFS_/MotoRosFS_190.out b/motoman_driver/MotoPlus/output/FS100/MotoRosFS_/MotoRosFS_190.out
new file mode 100644
index 0000000..c79f1ba
Binary files /dev/null and b/motoman_driver/MotoPlus/output/FS100/MotoRosFS_/MotoRosFS_190.out differ
diff --git a/motoman_driver/MotoPlus/output/YRC1000/MotoRosYRC1000/MotoRosYRC1000.out b/motoman_driver/MotoPlus/output/YRC1000/MotoRosYRC1000/MotoRosYRC1000.out
deleted file mode 100644
index 5507592..0000000
Binary files a/motoman_driver/MotoPlus/output/YRC1000/MotoRosYRC1000/MotoRosYRC1000.out and /dev/null differ
diff --git a/motoman_driver/MotoPlus/output/YRC1000/MotoRosYRC1_/MotoRosYRC1_190.out b/motoman_driver/MotoPlus/output/YRC1000/MotoRosYRC1_/MotoRosYRC1_190.out
new file mode 100644
index 0000000..0e56301
Binary files /dev/null and b/motoman_driver/MotoPlus/output/YRC1000/MotoRosYRC1_/MotoRosYRC1_190.out differ
diff --git a/motoman_driver/MotoPlus/output/YRC1000u/MotoRosYRC1u_/MotoRosYRC1u_190.out b/motoman_driver/MotoPlus/output/YRC1000u/MotoRosYRC1u_/MotoRosYRC1u_190.out
new file mode 100644
index 0000000..afdcf41
Binary files /dev/null and b/motoman_driver/MotoPlus/output/YRC1000u/MotoRosYRC1u_/MotoRosYRC1u_190.out differ
```

</details>

gavanderhoorn, 2020-03-05:
In `v1.7.2`, velocity feedback support was added to MotoROS for FS100 (#205).

In `v1.9.0` some changes were made to the "parameter extraction functions when using four control groups on an FS100" (#284).

gavanderhoorn, 2020-03-05:
@impet14: was `v1.7.0` the first version of MotoROS that started working? Have you tried any of the versions in between `v1.7.0` and `v1.9.0`?

impet14, 2020-03-05:
@gavanderhoorn , Thank you for your response.
I have not tried the version between v1.7.0 and v1.9.0. 
When I use v1.7.0 and telnet to fs100, there is no error. But when I install v1.9.0 to fs100 there would be the error "Failed to update controller status.  Check robot parameters." from telnet without connecting stream launch file.

EricMarcil, 2020-03-05:
Thanks for the information.  We are looking into it but I don't have access to an FS100 so I had to ask for assistance from a different office which is causing delay.  We were also focusing on the controller setup but now it seems that is would be related to some change in the code.  I'll try to follow-up on it and look at the code between 1.70 and 1.9.0 for the differences.

EricMarcil, 2020-03-05:
I've identified the source of the problem with this morning information that version 1.7.0 was working, We added new I/O check for the PFL function (used by collaborative robots) but those I/O signals don’t exist for older controller models.
I’ve put conditional compile around those modifications and recompiled it.  @acbuynak tested it and confirm that the issue was solved.
I'll create a PR for version 1.9.1 that will fix this problem.

acbuynak, 2020-03-05:
Confirming that the proposed version 1.9.1 fixed all errors identified in this thread. I've tried a variety of commands, settings, and etc. Everything seems to be working as intended.

I was able to confirm message receipt by the FS100 using output from `telnet`.

impet14, 2020-03-06:
Thanks. I look forward to test v1.9.1

gavanderhoorn, 2020-03-20:
Just merged #320 which should fix this.

alexander-jh, 2020-04-16:
Issue sorted out!

closed 2020-04-16 by alexander-jh

gavanderhoorn, 2020-04-16:
@alexander-jh: `v1.9.1` has fixed this for you as well?