ros-industrial/motoman #318: cannot locate init_ros.jbi for yaskawa STEM robot MHJF

saitejaallam, 2020-02-29:
Hey all,

I wanted to create a MotoROS interface with my STEM educart. According to the tutorial here 
`http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer (go to step 3), `
it mentions loading a INIT_ROS.JBI file. The list of INIT_ROS.JBI file is given on this link: 
`[(https://github.com/ros-industrial/motoman/tree/kinetic-devel/motoman_driver/Inform]`
I m not sure which one is for the STEM cart though. Can someone help me to figure this out ?
we have to choose between these options: 

sda_dual_arm
single_arm
single_arm_with_base_axis
single_arm_with_ext_axis
two_arms

the link for our robot (with which we want to create a MotoROS interface is given here: 
`https://www.motoman.com/en-us/products/robots/education` (Education Robot MHJF)

marip8, 2020-02-29:
Assuming the cell only has a single robot and is running on the FS100 controller per the link, you'll want to use the Inform files for the `single_arm` [here](https://github.com/ros-industrial/motoman/tree/kinetic-devel/motoman_driver/Inform/DX200%2C%20FS100%2C%20YRC1000/single_arm). The others are for dual arm robots, robots mounted to rails, or robots with external joints (i.e. part positioner)

gavanderhoorn, 2020-02-29:
Seeing as this is indeed an FS100, the correct job should be the `single_arm` one, as @marip8 writes.

Assuming there are no peculiarities of that controller we're not taking into account (ie: any "educational tweaks"), I'm going to assume this is answered and close the issue.

@EricMarcil and/or @ted-miller: if you could confirm, that would be appreciated.

@saitejaallam: feel free to keep commenting on the issue of course.

closed 2020-02-29 by gavanderhoorn

ted-miller, 2020-03-02:
@gavanderhoorn, you are correct.