ros-industrial/motoman #322: Inertial tags in xacro file for GP Series missing?

rntatsuya, 2020-04-03:
From my understanding, you need to have the `inertial` tags in your xacro files to be able to simulate your robot in Gazebo. I noticed that the GP series all do not have an `inertial` tag in its xacro file. Is there any way for me to get accurate `mass`, `inertia`, and `origin` values to fill in this information manually?

Also, if this information is not available, how would you recommend going about running the GP series robots in Gazebo? Should I use the sia10d `inertial` tag values as a substitute to get a somewhat ok simulation or should I just give up on running it in Gazebo and just aim for visualizing the movement in RVIZ?

gavanderhoorn, 2020-04-03:
I believe it's safe to say all `inertial` tags you can find in this repository will be either approximations, or outright wrong.

So I would not recommend depending on them.

> should I just give up on running it in Gazebo and just aim for visualizing the movement in RVIZ?

that's a bit fatalistic, don't you think?

> Also, if this information is not available, how would you recommend going about running the GP series robots in Gazebo?

I would suggest to do what "everyone" with the desire/need to run a dynamic simulation of an industrial robot (for which no accurate data is available, nor any system identification has been done by someone else): use approximations.

Mass is typically just guestimated.

Inertial properties can either be derived from meshes (there's a Gazebo tutorial about how to do that with Meshlab and SolidWorks can also give you a starting point).

Alternatively you could use primitives. Formulas exist for ~~calculating~~ approximating the inertia of primitive shapes. See [ur_description/urdf/ur10.urdf.xacro](https://github.com/ros-industrial/universal_robot/blob/06d8b9e2f5f86aa54f9f2845f11edbc84e2f951e/ur_description/urdf/ur10.urdf.xacro#L12-L20) for one example.

gavanderhoorn, 2020-04-03:
It all depends on a bit on what sort of simulations you need.

If you're looking to accurately simulate robots (ie: to be able to tune controllers and run experiments with effort controllers) I don't believe this is going to be feasible.

If you're only looking to to 'high-level' simulations (so almost event-based, discrete type applications, such as pick-and-place or some simple continuous processes) that would be possible.

gavanderhoorn, 2020-04-03:
As this is not really a problem with the packages in this repository, I'm going to close the issue.

Please do keep commenting here of course.

If it turns out there is something we can do -- and not just for the GP7 -- we'll re-open.

closed 2020-04-03 by gavanderhoorn

gavanderhoorn, 2020-04-03:
@rntatsuya wrote:
> Is there any way for me to get accurate `mass`, `inertia`, and `origin` values to fill in this information manually?

@EricMarcil  / @ted-miller: is there any data available that could help here?

ted-miller, 2020-04-06:
We will investigate this.

EricMarcil, 2020-05-20:
Sorry for the long delay.  I've been trying to get authorization to release this information in the public space but so far without success.  It seems that internally some people are considering this information as intellectual property and are questioning the need for end-user to have access to this information.  It is debatable... I'm still pushing for it but it is taking longer then anticipated,

If someone truly needs this information, please contact your Yaskawa local representative.  I know that we have made this information available to some customers but they had to sign a None Disclosure Agreement.  So you could use the information internally but wouldn't be able to publish or release your URDF model in the public domain.

I'll continue to push to change this policy...

gavanderhoorn, 2020-05-20:
> It seems that internally some people [..] are questioning the need for end-user to have access to this information

This is not an uncommon hesitation actually.

Without this data any sort of simulation with ROS & Gazebo doesn't really work. Using purely position control can partially work around it, but then interacting with objects in the simulated environment will be almost impossible (force <-> force simulation can be made stable, position <-> force is very, very tricky).