ros-industrial/motoman #323: YRC1000 jobs in teach pendent

arsh09, 2020-04-15:
Hello, 

Thank you for such an amazing support. 

I have a rather very basic question about YRC1000 controller. Can we run 2 jobs at the same time. One being the ROS_INIT and one from MotoCom? 

Thank you

Arshad

EricMarcil, 2020-04-15:
Hi Arshad,

On the YRC1000, you can run multiple tasks/jobs in parallel but the option is enabled.  Make sure that the Language level is set to Advance and check in the Inform List for the PSTART instruction.  If it is not there, then you'll need to contact you local Yaskawa support to have it enabled.  But I'm not sure you need that.

Motocom is our Windows' High-Speed Ethernet Client.  The High-Speed Ethernet Server is not related to job and should work even if the ROS_INIT job is running.  It is possible to use MotoCom to switch between the execution of regular jobs and the ROS_INIT job. 

Even simpler, you could add code to the ROS_INIT job to run some regular motion and then switch over to ROS and then switch back to regular motion.  You just need to do the I/O handshaking that is in the ROS_INIT to handover controller to ROS whenever you need it.

The main restriction is having a single point of control.  Which mean you can't have two jobs in parallel trying to move the same robot.  So even if you have to multiple tasks, you can't have two tasks running a job for the same robot (you'll get an alarm).

I hope this helps.

arsh09, 2020-04-15:
Hi Eric. 

Thank you for your answer. It is very helpful. 

Just one more small thing about single point of control, do the motion commands (either from MotoCom or ROS) blocking? By blocking, I meamt that If I send a joint motion command from ROS and during its execution, I send another motion command MotoCom, will the second one be blocked/queued untill the first one is completely executed?

EricMarcil, 2020-04-15:
If a robot job is executing and you send a Motion command from MotoCom, it will return an error code and will not be queue.  The MotoCom motion API is old and fairly limited.  You can't queue any motion.  If you send a 2nd motion command before the 1st one is complete, it will also return an error.

arsh09, 2020-04-15:
Thank you Eric. 

I am closing the issue.

closed 2020-04-15 by arsh09