ros-industrial/motoman #325: Gazebo Model Motomini issues

ShrutheeshIR, 2020-05-08:
Apologies if this isn't the right place to ask this question. I did not receive solutions from anywhere else. This is specific to Motomini, which I am trying to spawn in Gazebo.

I have added the gazebo tags for the visual elements and joints, along with the transmission elements. However, the joint - joint_4_r slowly slips and after a while gets detached from the main part and begins to float and move randomly in the air. Enabling selfCollision makes the entire model explode and spasm violently.

What can I do to solve this? Is it a problem with my actuator values?

Any help would be appreciated! Thanks

gavanderhoorn, 2020-05-08:
None of the models in this repository have been created with Gazebo-compatibility in mind.

I would not be surprised if issues come to light when you try to use them with Gazebo.

Without seeing what you've done, it'll be impossible to us to suggest anything other than: check the modifications you've made.

If you can point to your fork, I could take a look, but I cannot promise you any times or dates when that will happen.

gavanderhoorn, 2020-05-08:
As this is not really an issue with the packages in this repository (we do not provide explicit support for Gazebo simulations at this time), I'm going to close this.

That does not mean we cannot keep discussing and diagnosing your problem, so feel free to keep commenting on the issue.

If/when it turns out it there is something which we should fix here, we will re-open the issue.

closed 2020-05-08 by gavanderhoorn