ros-industrial/motoman #328: Motoman ROS Server Installation Issue

Pranav186, 2020-05-23:
Hello,
I am trying to install the Motoman ROS Server from the steps provided on the site - http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer

The controller I am using is FS100. I followed step 2.2(downloading the MotoRosXXXXX.out file on ext USB) followed by step 2.4.2(loading the MotoRosXXXXX.out file on the controller). But, I am getting the error message on my screen after rebooting
[image]

1.Resetting the alarm does not work because it is a major alarm.
2. The software has been updated so that minimum requirements are satisfied.
3. When I delete the .out file, there is no error, so the error only occurs after I load the .out file on the controller.

Can someone please help me solve this problem as quickly as possible.

gavanderhoorn, 2020-05-24:
Seems #273 lists the same errors.

> Can someone please help me solve this problem as quickly as possible.

there is no need to add this.

gavanderhoorn, 2020-05-24:
As a check: could you see whether [this version](https://github.com/ros-industrial/motoman/blob/f4fb63984061f8fca2a9a3f886a41137a1ee0764/motoman_driver/MotoPlus/output/FS100/MotoRosFS_/MotoRosFS_190.out) of MotoROS does not error out?

Pranav186, 2020-05-24:
Thanks for the reply.
Would also like to give some background. Initially my colleague was working on this problem for a few weeks. I have taken over and trying to solve this problem since last week. There could be places where I do not have information.

> As a check: could you see whether [this version](https://github.com/ros-industrial/motoman/blob/f4fb63984061f8fca2a9a3f886a41137a1ee0764/motoman_driver/MotoPlus/output/FS100/MotoRosFS_/MotoRosFS_190.out) of MotoROS does not error out?

I do not have access to the robot for 2 days so I will not be able to check this for now. Also I will get back to you on the version of system software after checking it on the robot.
Now to answer the remaining questions-

2. My colleague(not a technician from Yaskawa) performed the upgrade after receiving instructions via email from the Yaskawa tech team.
3. I am not sure what you mean here. The robot was set up by my colleague. It has not been checked manually by any Yaskawa technician.
4. The .out file was downloaded from http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer  step 2.2 fs100 and then placed in a usb with no other files. After that, step 2.4.2 was followed. 

I have not compiled from source since there is a prebuilt file in step 2.2

gavanderhoorn, 2020-05-24:
> Would also like to give some background. Initially my colleague was working on this problem for a few weeks. 

Which problem? Getting MotoROS installed? Why wait so long before asking for support?

> 2\. I am not sure what you mean here. The robot was set up by my colleague. It has not been checked manually by any Yaskawa technician.

There are certain settings which could benefit from a check by someone from Yaskawa. It should not be necessary any more (now that `MOTOMAN DRIVER` is part of the *OPTION FUNCTION* menu), but I've seen situations before where tons of errors were fixed by getting the system parameters sorted out by Yaskawa.

I'm sure @ted-miller and/or @EricMarcil will comment here when they have a chance.

Something you could already check: you have the `MOTOMAN DRIVER` option (in the *OPTION FUNCTION* menu) set to `USED`, correct? Refer to steps 2 to 4 of section 2.4.2 in the installation tutorial).

> 3\. The .out file was downloaded from http://wiki.ros.org/motoman_driver/Tutorials/indigo/InstallServer  step 2.2 fs100 and then placed in a usb with no other files. After that, step 2.4.2 was followed.

Could you post the output of `md5sum MotoRosFS_191.out` here? It should return `d81aaa2fb19b66891bd982bf30b9e996`.

EricMarcil, 2020-05-25:
This error would indicate that the MotoRos driver is making reference to a function that can't be found on the controller.  It might be that your controller has old software.
Please send us your controller software by looking on the pendant in the menu: SYSTEM --> VERSION

You can also get debugging information by connecting to the controller using telnet but you have to connect as the FS100 is booting up to see the error message that interest us.  (This only works on FS100.  On other controllers the telnet service is not available quickly enough, you would need to attached a monitor with a special cable).  You can refer to section 5.4 of the manual: [MotoPlus SDK for Visual Studio User](https://www.motoman.com/getmedia/0b44600a-ea48-4174-b81a-89e3cbd90dcd/190844-1CD.pdf.aspx?ext=.pdf) (You don't need Visual Studio for doing this step)
As the MotoPlus application is loading, it should indicate which function reference is missing.

gavanderhoorn, 2020-05-25:
@EricMarcil wrote:
> This error would indicate that the MotoRos driver is making reference to a function that can't be found on the controller.

So this does seem -- at least right now -- to be a similar problem as was the cause in #273, but perhaps with a different cause.

ted-miller, 2020-05-26:
@EricMarcil @gavanderhoorn 
This problem you describe is not related to this issue. 
> This error would indicate that the MotoRos driver is making reference to a function that can't be found on the controller. It might be that your controller has old software.

> So this does seem -- at least right now -- to be a similar problem as was the cause in #273, but perhaps with a different cause.

These issues will result in alarm 1020, with a subcode of [5]. However, this issue has a subcode of [6].

@Pranav186  Is this system located at Texas A&M? If so, then I believe all remote troubleshooting options have been exhausted during the previous work with Yaskawa Tech Support. In that case, an on-site technician is the only remaining option.

Either way, it would be helpful to see the information requested by @EricMarcil above. When connecting to telnet, please see 5.4.2.2 for the login information. Also, you must establish the connection immediately after turning on the FS100.

Pranav186, 2020-05-26:
Sorry for the late reply. As I said before, I did not have access to the robot for some time.
I have implemented some of the suggestions given by you.
> Something you could already check: you have the `MOTOMAN DRIVER` option (in the _OPTION FUNCTION_ menu) set to `USED`, correct? Refer to steps 2 to 4 of section 2.4.2 in the installation tutorial.

It is set to USED. So no problem with that

> Could you post the output of `md5sum MotoRosFS_191.out` here? It should return `d81aaa2fb19b66891bd982bf30b9e996`.
This is the file that was loaded on the robot controller from usb. Hope I have followed the correct procedure because the output is different than expected.
[image]

> Please send us your controller software by looking on the pendant in the menu: SYSTEM --> VERSION

[image]

> You can also get debugging information by connecting to the controller using telnet but you have to connect as the FS100 is booting up to see the error message that interest us. (This only works on [...]

I have not tried this. I saw in the manual that I need a Motoplus SDK "disc" to install the software. I am not able to find this disc. Does it come with the robot? Is there another way to go about it?

> @Pranav186 Is this system located at Texas A&M? If so, then I believe all remote troubleshooting options have been exhausted during the previous work with Yaskawa Tech Support. In that case, an on-site technician is the only remaining option.

Yes it is the one at Texas A&M. We are looking forward to call a tech onsite. But I thought of posting it here anyway to see if the Github community could help me get some insights.

EricMarcil, 2020-05-26:
For the debugging information, you don't need the MotoPlus SDK.  You just need to enable Telnet on you PC if it is not already enabled.  Then, you just type:
telnet 192.168.1.31  // assuming that you are using the default IP address of the controller
Then enter the username and password when prompt which is "MOTOMANrobot" for both.

gavanderhoorn, 2020-05-26:
@Pranav186 wrote:
> > Could you post the output of `md5sum MotoRosFS_191.out` here? It should return `d81aaa2fb19b66891bd982bf30b9e996`.
> 
> This is the file that was loaded on the robot controller from usb. Hope I have followed the correct procedure because the output is different than expected.
> [image]

That is indeed not the hash I would have expected.

Could you please redownload the file from [this link](https://github.com/ros-industrial/motoman/blob/kinetic-devel/motoman_driver/MotoPlus/output/FS100/MotoRosFS_/MotoRosFS_191.out?raw=true) and run `md5sum` again?

Pranav186, 2020-05-26:
> Could you please redownload the file from [this link](https://github.com/ros-industrial/motoman/blob/kinetic-devel/motoman_driver/MotoPlus/output/FS100/MotoRosFS_/MotoRosFS_191.out?raw=true) and run `md5sum` again?

I followed this and ran the command and now I am getting exactly what you were expecting. 

> For the debugging information, you don't need the MotoPlus SDK. You just need to enable Telnet on you PC if it is not already enabled. Then, you just type:
> telnet 192.168.1.31 // assuming that you [...]

Also I need some help with telnet since I have less experience with networking. This is the procedure followed so far
1. I have enabled telnet on the computer. 
2. Connected the ethernet cable between the controller and the computer.
3. Started the controller and quickly ran the command - telnet 192.168.1.31 on the command prompt. But got the error- could not open connection to the host, on port 23

EricMarcil, 2020-05-26:
You also need to set parameter S2C1119=2.  It's in the manual but forgot to post it in the above instructions.

Pranav186, 2020-05-26:
I am sorry, I am not able to find PARAMETER anywhere in the controller. I am in management mode.

gavanderhoorn, 2020-05-26:
> I followed this and ran the command and now I am getting exactly what you were expecting.

I'm sorry for asking you to do things in parallel, but seeing as you now have the correct file: does this still result in the errors your report in your OP if you install it on the controller?

EricMarcil, 2020-05-26:
Parameter changes are done from the standard mode, not the maintenance mode.  Once you are in management mode, you will get an extra button in the left column menu.  At the bottom of the left column, press the right arrow to go to the next set of buttons, and it should be under the EX.MEMORY button.

Pranav186, 2020-05-26:
> I'm sorry for asking you to do things in parallel, but seeing as you now have the correct file: does this still result in the errors your report in your OP if you install it on the controller?

I tried this and finally, gentlemen, I am not getting any alarm on the screen. 
I do not know what was wrong in the .out file. As far as I know, my colleague had got it from the roswiki page. Where did you get the file from @gavanderhoorn ?

Sorry to have bothered you with such a stupid problem.
Thanks a ton for helping me with this issue !! Request you to keep this thread open for a day or two. Just to make sure some other issue does not pop up.

Thanks once again,
Pranav.

gavanderhoorn, 2020-05-27:
> As far as I know, my colleague had got it from the roswiki page. Where did you get the file from @gavanderhoorn ?

It's the exact same link as is listed on the wiki (first link: wiki, second: from my https://github.com/ros-industrial/motoman/issues/328#issuecomment-634122993):

```
https://github.com/ros-industrial/motoman/blob/kinetic-devel/motoman_driver/MotoPlus/output/FS100/MotoRosFS_/MotoRosFS_191.out?raw=true
https://github.com/ros-industrial/motoman/blob/kinetic-devel/motoman_driver/MotoPlus/output/FS100/MotoRosFS_/MotoRosFS_191.out?raw=true
```

I wouldn't know what happened with your download. Github is sort of (in)famous for suffering from transient network problems. Without trying to shift the blame, it *could* be the case your download was corrupted by something on their end.

To avoid future users from running into the same issues, I'll add a section to the [Troubleshooting](http://wiki.ros.org/motoman_driver/Troubleshooting) page *and* update the tutorial to have users verify the integrity of the download using `md5sum` before continuing with the installation proper.

gavanderhoorn, 2020-05-27:
> Request you to keep this thread open for a day or two. Just to make sure some other issue does not pop up.

Actually, let's turn this around: I'm closing this with a tentative `worksforme` resolution.

If it turns out something is still not working -- and it's related to this issue -- we'll re-open it.

In any case: feel free to keep commenting on it.

gavanderhoorn, 2020-05-27:
And an (almost) off-topic request: please don't post screenshots of terminals.

It's all text, so there is no need to use graphics to show it to us. Instead, copy-paste it into a comment here, and make sure to surround it with the proper code-formatting tags (on github, use three consecutive backticks (ie: '`') to open and close a code block).

Text is preferred, as we can easily copy and paste it ourselves (for instance when trying to reproduce a problem). I cannot copy-paste text from a screenshot.

closed 2020-05-27 by gavanderhoorn

gavanderhoorn, 2020-05-27:
@EricMarcil @ted-miller: if you could take a look at the changes to the installation tutorial and the troubleshooting page (mentioned in https://github.com/ros-industrial/motoman/issues/328#issuecomment-634472739)?

ted-miller, 2020-05-27:
I edited the troublshooting page. I broke out the two different subcodes to be consistent with the other alarms listed on the page. Also, I added another possible cause for 1020[6].

gavanderhoorn, 2020-05-27:
Thanks @ted-miller.

There appears to be a change to line 330 as well:

> Additionally, if using the standard teach pendant, select [In/OUT]?[PSEUDO INPUT SIG]

That `?` was a `→` earlier.

Did you change that on purpose?

ted-miller, 2020-05-27:
No. That's an accident. Probably because I copied to notepad to make the edit in a larger window.

gavanderhoorn, 2020-05-27:
Seems only the diff is confused.

The question mark doesn't actually render in the normal page.

Sorry for the noise.

Pranav186, 2020-05-27:
> Text is preferred, as we can easily copy and paste it ourselves (for instance when trying to reproduce a problem). I cannot copy-paste text from a screenshot.

Good point

> I edited the troublshooting page. I broke out the two different subcodes to be consistent with the other alarms listed on the page. Also, I added another possible cause for 1020[6].

Yes, it is a good idea to add this to the troubleshooting page. It is rare, but people might face this same issue.

(Also, I have proceeded to the next step, that is, installing inform code. There is a small issue that I am facing. Will open a separate thread for that.)

gavanderhoorn, 2020-05-27:
We're not entirely convinced file corruption was the sole cause, but as it works for you now, we cannot diagnose any further.