ros-industrial/motoman #329: Motoman GP8 interfacing with Arduino

Bdelspi, 2020-05-24:
I realize this is not an issue with the packages in this repository, but have greatly appreciated the assistance from this community in the past, and would greatly appreciate it again...

Building from the [simple_trajectory_action_client.py](https://github.com/gavanderhoorn/motoman/blob/50b9fd428758932beb32fa60fa9e27a4ffba1214/motoman_driver/examples/simple_trajectory_action_client.py) from @gavanderhoorn (Thank you for this, you are awesome!) I was able to create a complete path for our GP8.

Building further off of this code, I have been trying for the past three days to expand these capabilities to be able to interface with an arduino running multiple stepper motors.

For context, I created an arduino program in the arduino software IDE that commands four stepper motors depending on what value it receives. To do this I created a subscriber within the program to listen for a Float64 value. I have tested this from the terminal and have been successful.

Likewise expanding @gavanderhoorn's code with Python, I have been able to achieve the desired robot path. Testing this from the terminal, this also works.

Ideally the robot would go to its first waypoint and send a signal which would then tell the arduino to excite one of the stepper motors, depending on what signal it received. Ideally the robot and stepper motors could also move in tandem (i.e while the GP8 is on its way to the second waypoint, the stepper would also be performing its operation).

Can anyone please offer a starting point or suggested workflow for this? 

I tried [Writing a Simple Service and Client (Python)](http://wiki.ros.org/ROS/Tutorials/WritingServiceClient%28python%29) and [Examining the Simple Publisher and Subscriber](http://wiki.ros.org/ROS/Tutorials/ExaminingPublisherSubscriber), but I am not sure this is the right way to go since that is generally used for constantly sending commands to the hardware.

I also tried directly making a connection (i.e not using publishers/subscribers) and adding a delay, but then the whole code gets delayed and the timing becomes obtuse.

My most recent thoughts are that using an action would be a potential option, since I am only occasionally sending commands that may take a long time (or a highly variable amount of time) and have been working through some of the examples in [Programming Robots with ROS](http://shop.oreilly.com/product/0636920024736.do), but have still not been able to achieve desired performance.

Can anyone please provide guidance on potential paths forward? 

I am still pretty new to all of this and greatly appreciate the help!

gavanderhoorn, 2020-05-24:
I understand why you posted this here, but as you already mention yourself, this issue tracker is not the place for questions about things which are not either MotoROS nor the other packages in this repository.

The ROS community has a forum dedicated to these kinds of questions called [ROS Answers](https://answers.ros.org).

I'd suggest you post your question there.

gavanderhoorn, 2020-05-24:
As this is not an issue with any of the packages in this repository, I'm going to close this here.

Feel free to keep commenting on it, but do please keep things on-topic.

closed 2020-05-24 by gavanderhoorn