ros-industrial/motoman #330: INFORM Code question

Pranav186, 2020-05-27:
This is a question that I have in the step 3 (Installing INFORM Code) of the Installation of Motoman ROS Server Tutorial.

This is my understanding of the procedure to be followed-
1. Download the INIT_ROS.JBI and IONAME.DAT files and place both these files into a usb.
2. Load the job file into the controller using the procedure mentioned, check if has been properly installed and proceed to the next step.

Please correct me if I am wrong in the above steps. If this procedure is correct, what is the use of IONAME.DAT file?

EricMarcil, 2020-05-27:
It is putting name to the Input and output used by MotoRos so there are easier to recognize.  Those output are being set by MotoPlus to report it's current state.  The name themselves are not essential and take note that is you have already name any I/O, loading the IONAME.DAT will overwrite the names you've already entered.  So you might want to manually enter the names when doing this on an existing system that already has name for other I/O.

OUT#889  ROS_READY
OUT#890  ROS_DONE
OUT#891  INIT_DONE
OUT#892  CONNECTION_SRV
OUT#893  MOTION_SERVER
OUT#894  STATE_SERVER
OUT#895  IO_SERVER,FAILURE

Pranav186, 2020-05-28:
Ok. So do we have to load this file to the controller as well? In the tutorial it only says to load the jbi file.

EricMarcil, 2020-05-28:
It's optional.  There are just names for the user, the system doesn't care.

gavanderhoorn, 2020-05-28:
I understand why this was posted here, but it's actually not an actionable issue and doesn't report anything which is wrong or doesn't work in the packages in this repository.

As such, I'm going to close it.

Feel free to keep commenting on it of course.

closed 2020-05-28 by gavanderhoorn