ros-industrial/motoman #332: Sending trajectory using ROS

Pranav186, 2020-06-01:
Hello,
I have completed the installation of Motoman ROS Server. I am trying to to test the robot by sending a trajectory. I ran the command- ```rosrun motoman_driver move_to_joint.py "[0, 0, 0, 0, 0, 0]"``` and got the following output-
```
[ERROR] [1591022893.022516]: Start and End position joint_names mismatch
[ERROR] [1591022893.023062]: Unable to move to commanded position. Aborting.
Traceback (most recent call last):
  File "/home/robotics/catkin_ws/src/motoman-kinetic-devel/motoman_driver/src/move_to_joint.py", line 133, in <module>
    main(sys.argv[1:])
  File "/home/robotics/catkin_ws/src/motoman-kinetic-devel/motoman_driver/src/move_to_joint.py", line 127, in main
    move_to_joint(end_pos, duration)
  File "/home/robotics/catkin_ws/src/motoman-kinetic-devel/motoman_driver/src/move_to_joint.py", line 69, in move_to_joint
    traj = build_traj(get_cur_pos(), end_pos, duration)
  File "/home/robotics/catkin_ws/src/motoman-kinetic-devel/motoman_driver/src/move_to_joint.py", line 25, in build_traj
    raise
TypeError: exceptions must be old-style classes or derived from BaseException, not NoneType
```
I ran the following command which is mentioned in #228 

```rostopic pub /joint_path_command trajectory_msgs/JointTrajectory \
"{joint_names: ['joint_1_s', 'joint_2_l', 'joint_3_u', 'joint_4_r', 'joint_5_b', 'joint_6_t'], \
points: [ \
{time_from_start: {secs: 0.1}, \
positions: [0.0, 0.0, -8.751316272537224e-06, 0.0, -1.917476038215682e-05, 0.0],
velocities: [0,0,0,0,0,0]},\
{time_from_start: {secs: 1}, \
positions: [0, 0, 0, 0, 1, 0],
velocities: [0,0,0,0,0,0]}\
]}" -1
```

I get the following output-
```publishing and latching message for 3.0 seconds```
but the robot does not move. It tried changing the positions and velocities as well.

In a few posts it is suggested to not use the move_to_joint.py script and use action server. 
Could you guide me how to use action server. I simply want to test that the robot is able to get trajectories from ROS and move. Right now I am unable to move the robot at all using ROS.

I also think it would be a good idea to include a few steps of action server in the tutorial.

gavanderhoorn, 2020-06-01:
Please see #229.

Do not publish messages directly to any topics, use the action server.

Also do not use "random" commands you find in comments on issues. Those are typically only used for diagnosing a specific issue. Re-using them for other purposes -- without understanding what they're doing -- can be dangerous and should not be done.

gavanderhoorn, 2020-06-01:
I'm going to close this, as I expect the example shown in #229 will work and has sufficient detail to resolve this.

If it turns out something is egregiuosly wrong somewhere we can re-open.

gavanderhoorn, 2020-06-01:
> I simply want to test that the robot is able to get trajectories from ROS and move

please be careful with sentences which include "simply" or "just".

Experience tells me they are often not as easy or straightforward as they appear.

closed 2020-06-01 by gavanderhoorn

Pranav186, 2020-06-02:
> I'm going to close this, as I expect the example shown in #229 will work and has sufficient detail to resolve this.

Sorry, missed the example file that you had mentioned in #229. I think you should attach that file somewhere in the tutorial.

Would like to point that in line 61 of your code in https://github.com/ros-industrial/motoman/compare/kinetic-devel...gavanderhoorn:actionclient_example is giving an attribute error. Changing robot_joint_states.positions to robot_joint_states.position solved the problem. You might want to make the change.

Anyway, I am able to send the trajectory and the robot is following the trajectory so everything is working fine. 

Thanks for the help @gavanderhoorn