ros-industrial/motoman #334: SIA50D robot model support

wxmerkt, 2020-06-03:
Hi,
Thank you for maintaining this high-quality package!

I was looking into an application using a [SIA50D](https://www.motoman.com/en-us/products/robots/industrial/assembly-handling/sia-series/sia50d). As part of this I am looking for the robot description but it is not yet included in this repository.

Are there plans to add support for this model (even if only description/MoveIt config) to this package?

I am happy to contribute time and effort to adding the robot model but do not have access to meshes/further specifications. If someone from Yaskawa/Motoman follows this repository, or if you have access within ROS-Industrial, I am happy to help getting it added to this package if of interest.

Thanks,
Wolfgang

EricMarcil, 2020-06-04:
Hi,
There was not plan to create a SIA50D package since it is an older model.  But if you have an application, we can help you out.  I've requested the SLT files for the links to our drawing department, so I can send them to you if you e-mail me at: eric.marcil@motoman.com

I can also help on the package but I'm not sure if I'll have time to do that before next week.

gavanderhoorn, 2020-06-12:
@wxmerkt: have you been able to put something together?

wxmerkt, 2020-06-12:
Hi @gavanderhoorn,
I received the files but not had the time yet to extract the transforms to build the URDF. My plan is to do this over the next week.

gavanderhoorn, 2020-06-12:
Take a look at [motoman_gp88_support/tools/process_meshes](https://github.com/ros-industrial/motoman/tree/2ec6a962918c0650ea3b4295e9e2ea1ab75e9bce/motoman_gp88_support/tools/process_meshes). It could help reduce the amount of work.

PR #337 (2020-10-06): Add SIA50 support
This is a WIP PR to add support for the SIA50D. This PR is based on the support for the SIA20D and the mesh file processing for the GP88. Thanks to @EricMarcil for providing the meshes, model, and instructions.

The automatically created collision mesh/convex hull for the 6th link is currently not great (used the same script as for GP88).

I am planning on adding approximate CoG, inertia, and mass information from CAD assuming uniform mass distribution in each link (as we are planning to use the model for dynamic trajectory optimisation and the correct values are not availabl [... cut]
closed 2020-10-06 by gavanderhoorn