ros-industrial/motoman #336: Unable to read joint_states

Pranav186, 2020-06-11:
Hello,
 I am using the motoman gp50 robot with yrc1000 controller. After installing motosros, I am not able to read the joint_states. The command ```rostopic echo joint_states``` give the following output-
```
header:
  seq: 522
  stamp:
    secs: 1591816796
    nsecs: 152996560
  frame_id: ''
name: []
position: []
velocity: []
effort: []
```

Also, the terminal used for launching the file is giving the following error-
```
[ERROR] [1591816782.236914786]: Failed to find topic_list parameter
[ERROR] [1591816782.242008140]: Cannot find user-specified joint names. Tried ROS parameter 'controller_joint_names' and the URDF in 'robot_description'.
[ERROR] [1591816782.268223575]: Failed to find topic_list parameter
[ERROR] [1591816782.273167157]: Cannot find user-specified joint names. Tried ROS parameter 'controller_joint_names' and the URDF in 'robot_description'.
[ERROR] [1591816782.274316488]: Cannot find user-specified joint names. Tried ROS parameter 'controller_joint_names' and the URDF in 'robot_description'.
[ERROR] [1591816782.274503627]: Failed to initialize joint_names.
```
```rosparam set controller_joint_names "[joint_s, joint_l, ..., joint_t]" ```  does not have any change.

gavanderhoorn, 2020-06-11:
So this is a different robot from your other issues then?

In any case: please tell us how you start things.

Which `.launch` files do you use? Show us the *verbatim* commands.

Pranav186, 2020-06-11:
Yes, this is a different robot.

The command used was-

```roslaunch motoman_driver robot_interface_streaming_yrc1000.launch robot_ip:=192.168.1.31```

gavanderhoorn, 2020-06-11:
You shouldn't start that `.launch` file directly.

Look at [motoman_gp25_support/launch/robot_interface_streaming_gp25.launch](https://github.com/ros-industrial/motoman/blob/2ec6a962918c0650ea3b4295e9e2ea1ab75e9bce/motoman_gp25_support/launch/robot_interface_streaming_gp25.launch) and mimic that for your GP50.

gavanderhoorn, 2020-06-11:
I'm going to close this, as it's a configuration issue, not an issue with the packages in this repository.

In the meantime: please keep commenting if you have any further questions.

If it turns out there is something which needs work related to this issue, we can re-open.

closed 2020-06-11 by gavanderhoorn

Pranav186, 2020-06-11:
> Look at [motoman_gp25_support/launch/robot_interface_streaming_gp25.launch](https://github.com/ros-industrial/motoman/blob/2ec6a962918c0650ea3b4295e9e2ea1ab75e9bce/motoman_gp25_support/launch/robot_interface_streaming_gp25.launch) and mimic that for your GP50.

That worked, I am able to read the joint_states. Also, the robot is following the trajectory. Thanks @gavanderhoorn 

> Ideally you'd also create a support package for that GP50, so you have a place to store those files.
> 
> Finally: see whether [Working With ROS-Industrial Robot Support Packages](http://wiki.ros.org/Industrial/Tutorials/WorkingWithRosIndustrialRobotSupportPackages) (linked from the main readme and the wiki pages) provides some insight into how robot support packages are structured, and why they contain the files they contain.

Yes, I will definitely look into that.