ros-industrial/motoman #340: Problem of: ERROR: Service [/robot_enable] is not available.

zhangOSK, 2020-07-01:
Dear Sir, 
  I am using FS100 controller and ROS kinetic. The problem is that Service [/robot_enable] is not available even thought I checked the pendant key switch is in Remote mode and the E-stop is released.
  I firstly run the roslaunch as follow, 
[image]
  Then I tried robot_enable service.
[image]
  I think the connection is okay because I can see the rosservice list and also 'rostopic echo joint_states'.
  The setting of pendant should be fine because I can successfully enable robot service in the other pc.
  Could you please help give me some hints to solve this problem?

gavanderhoorn, 2020-07-01:
First: please do **not** post screenshots of terminals.

It's all text, so copy-paste the terminal text into your post here.

I've removed the screenshots for now, so you can replace them with the text itself. Edit your original comment to do that.

gavanderhoorn, 2020-07-01:
> I am using FS100 controller and ROS kinetic

Please provide us more information:

 - which version of the system software is running on your FS100?
 - what robot are you trying to control?
 - how did you install MotoROS?
 - did you encounter any errors or warnings while installing MotoROS?
 - has your controller been configured correctly for use of MotoROS?
 - which version of `motoman_driver` are you running?
 - how did you install `motoman_driver` (and the other packages)?

zhangOSK, 2020-07-03:
Dear @gavanderhoorn ,
  Thank you very much for the reply! 
  I am trying to do experiment on sda5f and this 'motoman_sda5f_support' is created by myself. Following your reply, I found that I did not put motoman_driver and motoman_sda5f_support in the same workspace. After add motoman_driver, I can successfully enable the robot! 
  Sorry for this silly mistake I made. 
  I appreciate your help very much!

closed 2020-07-03 by zhangOSK