ros-industrial/motoman #342: How to move two arms of dual arm robots simultaneously?

zhangOSK, 2020-07-09:
Hello!  I want to move both of arms of SDA5F but I am not sure how to deal with the groups of the dual arm robot.
  I am using ROS kinetic and FS100 controller. The config file of **sda5f_motion_interface.yaml** is shown below:
`topic_list:
    - name: sda5f_r1_controller
      ns: sda5f
      group: 0
      joints: ['arm_left_joint_1_s','arm_left_joint_2_l','arm_left_joint_3_e','arm_left_joint_4_u','arm_left_joint_5_r','arm_left_joint_6_b','arm_left_joint_7_t']
    - name: sda5f_r2_controller
      ns: sda5f
      group: 1
      joints: ['arm_right_joint_1_s','arm_right_joint_2_l','arm_right_joint_3_e','arm_right_joint_4_u','arm_right_joint_5_r','arm_right_joint_6_b','arm_right_joint_7_t']
    - name: sda5f_b1_controller
      ns: sda5f
      group: 2
      joints: ['torso_joint_b1']
    - name: sda5f_b2_controller
      ns: sda5f
      group: 3
      joints: ['torso_joint_b2']`

I am using python to publish joints data to the robot. After checking the issues before, I use 
`client = actionlib.SimpleActionClient('joint_trajectory_action', FollowJointTrajectoryAction)` 
 to send goal of  `goal = FollowJointTrajectoryGoal()`.  
The problem is that I don't know how to send 4 groups of joints information into the goal. I know that we can add points to goal by `goal.trajectory.points.append(JointTrajectoryPoint(positions=q0,
        velocities=qdot, time_from_start=rospy.Duration( 0.0)))`. 
But what if I want to send the joints information of torso_joint_b1, torso_joint_b2 and joints of both left and right arm at the same time?
Could you please tell me how to do that?

zhangOSK, 2020-07-09:
The other question confused me is that how to separate the subscribed joint_states. 
I tried `robot_joint_states = rospy.wait_for_message('joint_states', JointState)` but it only showed me the information of torso_joint_b1. 
Then I tried 
`rostopic echo joint_states`
`header: 
  seq: 72352
  stamp: 
    secs: 1594200152
    nsecs: 988018091
  frame_id: ''
name: [torso_joint_b2]
position: [0.0]
velocity: [0.0]
effort: []`
`header: 
  seq: 72353
  stamp: 
    secs: 1594200152
    nsecs: 988159792
  frame_id: ''
name: [torso_joint_b1]
position: [0.0]
velocity: [0.0]
effort: []`
`header: 
  seq: 72354
  stamp: 
    secs: 1594200152
    nsecs: 988303657
  frame_id: ''
name: [arm_right_joint_1_s, arm_right_joint_2_l, arm_right_joint_3_e, arm_right_joint_4_u,
  arm_right_joint_5_r, arm_right_joint_6_b, arm_right_joint_7_t]
position: [-6.075171404518187e-05, -1.5707507133483887, -1.570796251296997, -1.5707658529281616, 3.0375857022590935e-05, 0.0, -2.59996741078794e-05]
velocity: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
effort: []`
`header: 
  seq: 72355
  stamp: 
    secs: 1594200152
    nsecs: 988458383
  frame_id: ''
name: [arm_left_joint_1_s, arm_left_joint_2_l, arm_left_joint_3_e, arm_left_joint_4_u, arm_left_joint_5_r,
  arm_left_joint_6_b, arm_left_joint_7_t]
position: [-7.593964255647734e-05, -1.5707658529281616, -1.5707811117172241, -1.5707658529281616, -3.0375857022590935e-05, 1.723573950584978e-05, 0.0]
velocity: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
effort: []
`

It showed four groups of joint states. How can I separate the arm's states from this topic? Or is it better to subscribe **/sda5f/sda5f_r2_controller/joint_states** ?
My questions are mainly about the 4 groups in the topic. I would appreciate it if you can give me some documentations about how to deal with the groups.

gavanderhoorn, 2020-07-16:
To receive a coalesced `JointState` for all groups in a single message, you'll probably want to use a `joint_state_publisher` with the `source_list` parameter (see [the wiki](http://wiki.ros.org/joint_state_publisher#Parameters)).

Make it subscribe to the `joint_state` topics for each of the groups. Make sure to configure it to publish on a topic *other* than `joint_states` (otherwise you'll create a loop). You can use the [remap](http://wiki.ros.org/roslaunch/XML/remap) tag for this.

> The problem is that I don't know how to send 4 groups of joints information into the goal.

On the "ROS side", you don't care about motion groups on the robot controller. You only care about joints.

If you want to submit a goal for all joints, make sure to add *all joints* to the `joint_names` member of the `trajectory` member of the goal.

Then also make sure to add a sufficient (ie: equal) number of elements to the `positions`, `velocities` and `accelerations` members of the individual [JointTrajectoryPoint](http://docs.ros.org/melodic/api/trajectory_msgs/html/msg/JointTrajectoryPoint.html)s in the trajectory.

If you then submit *that* goal to the global `joint_trajectory_action` goal (so not one namespaced by the name of the group from the `sda5f_motion_interface.yaml` file), things should start working.

zhangOSK, 2020-07-20:
Thank you for the reply! I have managed to move the robot!

closed 2020-07-20 by zhangOSK