ros-industrial/motoman #343: SDA10F (real robot) is NOT moving - Without errors -  using motoman_sda10f_moveit_config pkg

inbarbd, 2020-07-14:
hello, 
i work with sda10f robot with  fs100 controller with the motoman_sda10f_moveit_config. 
 the robot is connected and we can read information from the controller. when move the robot manually with the pendant the robot position is updating in rviz.

when i try to plan and executed , moveit plan the path but the robot is not response. there is  NO ERRORS  

the only changes we made in the pakeg is:
1. ExecuteTrajectoryAction relapse MoveGroupExecuteService in move_group.launch
2.  ADD <param name="trajectory_execution/execution_duration_monitoring" value="false" /> to jrajectory_exexution.launch.xml

if anyone have any idea what to do next it will be very helpful. thank you in advance

[image]
[image]

gavanderhoorn, 2020-07-14:
Please do not post screenshots of terminals. It's all text, so copy-paste the text from the terminal into your post.

Edit your original post using the `Edit` option from the dropdown in the top-right.

Additionally, while removing that screenshot of your terminal, add:

 - system software version of your FS100
 - information on how you installed MotoROS (built from sources, downloaded the binaries, etc)
 - which version of MotoROS you've installed
 - whether your robot has been setup by Yaskawa Motoman or not

And include:

 - *verbatim* copies of the commands (ie: `roslaunch` and/or `rosrun`) you run
 - a screenshot of `rqt_graph` after you've started everything

Without that information, we cannot help you.

gavanderhoorn, 2020-07-14:
Thanks for the update, but you still don't show *verbatim* copy-pastes of the console output, nor the exact commands you run.

gavanderhoorn, 2020-07-16:
@inbarbd: please post a comment next time you edit your post, as Github does not send out notifications for edits.

gavanderhoorn, 2020-07-16:
> when run:
> 
> ```
> moveit_planning_execution.launch robot_ip:=10.0.0.2 controller:=fs100
> ```
> 
> the robot is connected and we can read information from the controller. when move the robot manually with the pendant the robot position is updating in rviz.

Is this an exact copy-paste of the command you run?

If it is, it would appear to be missing `sim:=false`, which would mean you're running not with data from the real robot, but from a simulated one.

Try to use Wireshark to make sure there is actual network traffic between your ROS PC and the FS100.

If there is, and you're certain you've actually added `sim:=false`, then attach a short Wireshark capture here so we can investigate it.

inbarbd, 2020-07-23:
Thank you for the response.

The sim argument is config in the launch file to be equal to false.

The reason i add -- <param name="trajectory_execution/execution_duration_monitoring" value="false" /> to jrajectory_exexution.launch.xml --is due to an error in the terminal.  I try to run it on two different commuters (ROS-kineck ubuntu 16.4 and ROS-melodic Ubuntu 18.04 and get the same error that led me to fond this solution.

Attached   to this message the Wiresark file records while try to execute trajectory with moveit. 

[wireshark_records.zip](https://github.com/ros-industrial/motoman/files/4964061/wireshark_records.zip)

We also try to send trajectory message direct to the controller (from python script not with moveit) and the results are the same - the message send but the robot is not move.

Please let me know if you have  idea what should be the next step.
Thank you in advance.

EricMarcil, 2020-07-23:
I've review the first capture:

From what I can see in the first capture.

@37.985 E-Stop is removed
@44.275 Servo turned on
@44.376 Motion Possible
@57.531 Motiion Ctrl: Motion Ready (200101)
@57.669 Joint trajectory starts
@57.669 Motion Reply: Success
…
@58.213 Joint Trajectory ends: Sequence Number 3;
@58.213 Motion Reply: Success
@58.226 Status: InMotion=true
@59.951 Status: InMotion=false

I can see that the robot is not moving throughout the sequence from the Joint Sequence.

The SDA10F type robot have 4 control groups (where group 3 and 4 need the same data) and I can see that you are only trying to move one group sending the standard simple message.  I believe that you need to use the [ROS_MSG_MOTO_JOINT_TRAJ_PT_FULL_EX = 2016] command and command all 4 joints together.

I haven't used a dual robot and don't know the detail on how to make the message switch on the ROS side.  Maybe @gavanderhoorn or someone from the community with an SDA robot can confirm and help you out.

gavanderhoorn, 2020-07-24:
> The sim argument is config in the launch file to be equal to false.

Please tell us about this sort of thing in the future.

We cannot guess what you are doing, and changing argument default values makes diagnosing issues much harder if you don't tell us about them.

> The reason i add -- to jrajectory_exexution.launch.xml --is due to an error in the terminal. 

*Which error*?

> I try to run it on two different commuters (ROS-kineck ubuntu 16.4 and ROS-melodic Ubuntu 18.04 and get the same error that led me to fond this solution.

It is not needed if everything is working correctly.

The fact you need to add it tells me something is either not correctly configured, or interaction with the driver is not done correctly.

> Attached to this message the Wiresark file records while try to execute trajectory with moveit. [...]

Could you please describe the steps you took which resulted in the 4 wireshark captures you've shared?

If you used some code, please show it here.

From your minimal descriptions, it's difficult to figure out what you are doing.

inbarbd, 2020-07-26:
* The following error occurred without adding   :
<param name="trajectory_execution/execution_duration_monitoring" value="false" />

[ERROR] [1595747455.605097855]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 2.667912 seconds). Stopping trajectory.

* I try to run this on two different computers with different ROS configuration because I want to see if the problem is the ROS melodic. the package available this Git is until ROS kinetic. But the logs are the same on both computers.

* All the Wireshark records was taken while try to execute trajectory from the Rviz (moveit) after run in the terminal:
roslaunch motoman_sda10f_moveit_config moveit_planning_execution.launch  robot_ip:=10.0.0.2 controller:=fs100 sim:=false

* At this moment we dont use extra code only try to move the robot from the moveit interface in Rviz.
 
If you need more information please let me know.

gavanderhoorn, 2020-07-28:
> All the Wireshark records was taken while try to execute trajectory from the Rviz (moveit) after run in the terminal:

that's great, but please describe how you use the MoveIt RViz plugin.

So which buttons do you press, which groups do you select, how do you change the target pose, etc.

It's been a long time since I've been able to test with an SDA robot, so I can't be sure whether things are correctly configured to be able to control a single group only.

gavanderhoorn, 2020-07-28:
This could be (somehow) related to #251.

inbarbd, 2020-07-29:
Thank you for the answer.
We didn't yet get to the problem describe issue #251. Our problem is that we cant move non of the group control of the robot. For now we try to move arm_left or arm_right.

We also get help from Yaskawa professionals and we preform several changes in the robot controller:

- The robot version was FS3.27 with  MotoRosFS100.
- Delete MotoRosFS100 from the controller and install MotoRosFS_191​.
- Upgrade the controller version to FS3.3 with MotoRosFS_191

 
In all this configuration we try to preform the follow steps:

1. Run 'roslaunch motoman_sda10f_moveit_config moveit_planning_execution.launch robot_ip:=10.0.0.2 controller:=fs100 sim:=false'
2. Run rosservice call /robot_enable , in different terminal.
3.servo on light is on 
success: True
message: "Motoman robot is now enabled and will accept motion commands."
3. in the Rviz the selected group is arm_left of arm_right in etch attempt. 
4. Select from the Rviz GUI New 'Goal State' to be random valid.
5. Press the Plan button.
6. Press the Execute button. 

All results was the same as before  --> The robot receive the commend but No errors show in the terminal or in the roslogs.

Only after upgrading the controller version a new error occur :

 INFO] [1596026066.002472947]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1596026066.002679648]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1596026066.003713328]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1596026066.003975286]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1596026066.005146053]: ParallelPlan::solve(): Solution found by one or more threads in 0.002826 seconds
[ INFO] [1596026066.009451122]: SimpleSetup: Path simplification took 0.003951 seconds and changed from 3 to 2 states
[ INFO] [1596026072.361043818]: Execution request received
[ERROR] [1596026072.381973959]: 0,1
[ERROR] [1596026072.382037953]: 0,1
[ERROR] [1596026072.382072539]: 0,1
[ERROR] [1596026072.382105891]: 0,1
[ERROR] [1596026072.382133661]: 0,1
[ERROR] [1596026072.382159280]: 0,1
[ERROR] [1596026072.382192109]: 0,1
[ERROR] [1596026072.382329334]: 0,1
[ERROR] [1596026072.382367289]: 0,1
[ERROR] [1596026072.382390542]: 0,1
[ERROR] [1596026072.382415638]: 0,1

- The following is the telnet messages :

roblab4@roblab4-ThinkCentre-M920s:~$ telnet
telnet> open 10.0.0.2
Trying 10.0.0.2...
Connected to 10.0.0.2.
Escape character is '^]'.

VxWorks login: MOTOMANrobot
Password: 

-> Starting new connection to the Motion Server
Creating new task: IncMoveTask
IncMoveTask Started
Creating new task: tidAddToIncQueue (groupNo = 0)
Creating new task: tidAddToIncQueue (groupNo = 1)
Creating new task: tidAddToIncQueue (groupNo = 2)
Creating new task: tidAddToIncQueue (groupNo = 3)
Creating new task: tidMotionConnections (connectionIndex = 0)
Starting new connection to the State Server
Starting State Server Send State task
Controller number of group = 4
In StartTrajMode
Robot job is ready for ROS commands.
No more errors or messages on the telnet connection.

This is the wireshark recorded for 3 identical attempts:
[motoman_new_vertion_ubuntu.zip](https://github.com/ros-industrial/motoman/files/4994908/motoman_new_vertion_ubuntu.zip)

 
The Robot still Not move. 

Do you have any idea what is the next steps in order to control the robot throw the Ethernet connection?
Thank you in advance,
inbar

gavanderhoorn, 2020-07-29:
> We also get help from Yaskawa professionals

That's great, but it also makes me hesitant to continue diagnosing here.

Trying to figure out a problem like this from two sides without those sides communicating or keeping each other up-to-date is not a good way to approach this.

@EricMarcil @ted-miller: are any of you involved here?

gavanderhoorn, 2020-07-29:
> ```
> [ERROR] [1596026072.381973959]: 0,1
> ```

I don't recognise this `ERROR` log line.

Please configure your rosconsole configuration to include the node name, so we can see which node prints this error.

Add this line to your `.bashrc`:

```
export ROSCONSOLE_FORMAT='[${severity}] [${time}] [${node}] [${file}:${line}] [${logger}]: ${message}'
```

and try running things again. The log output should now include the name of the node.

ted-miller, 2020-07-29:
@gavanderhoorn, I have instructed @inbarbd to use my Windows emulator application to test the robot and MotoROS application without using ROS-I. He is experiencing the same issue (no motion) using the emulator app. I can't find any obvious issues in the Wireshark captures.

Given that the emulator application is experiencing the same symptoms, I'm fairly confident that there is a problem with the configuration of the FS100. (Note: There might also be a problem with the ROS setup... but one issue at a time.) This controller has been used for other purposes in the past. It is likely that there is some safety or interlock signal that is left over from a previous application.

I have requested that they attempt to program a standard job on the teach pendant to see if motion is being blocked by something else in the sysytem. I have also asked the local Yaskawa rep to look into sending a technician to re-initialize the controller.

gavanderhoorn, 2020-07-29:
Thanks for the update @ted-miller. In the future: please post about any interaction like this with users which are not the result of something discussed here, for obvious reasons.

Given what you write, I'm going to put my efforts on hold until we get clarity about the status of the controller @inbarbd is trying to use.

ted-miller, 2020-08-03:
Today, it was discovered that the robot job (INIT_ROS) is setup incorrectly. Their robot job is a non-group job. The motion API will only work if the commanded control groups are sitting on a WAIT command in Inform. 

Tomorrow they are going to replace the INIT_ROS job and try again.

ted-miller, 2020-08-04:
After correcting the INIT_ROS job, the robot is behaving as expected.

closed 2020-08-04 by ted-miller

gavanderhoorn, 2020-08-04:
For future cases: do we know how the `INIT_ROS` job ended up groupless?

Was this a copy-pasta error, a corrupted file or something else?

ted-miller, 2020-08-04:
There was an issue with the configuration of the group combinations, so the file from Github wouldn't load. So, the job was manually typed on the pendant as a non-group job.

gavanderhoorn, 2020-08-05:
Ah, ok.

So the root cause was a problem with the group configuration on the controller then.

Thanks for updating the issue.