ros-industrial/motoman #345: Trajectory execution seems slow

lkeselman, 2020-07-21:
Hello,

Thanks to all in the community for maintaining this driver. I have it running on a GP7 and it works basically as expected. However, the trajectory execution speed appears to be be much lower than the robot's capability.

I programmed a similar path to the one being done through ROS in INFORM and the execution is way faster (like 2x+). Is this expected? Are there parameters I should take a look at? I'm using the time optimal parametrization.

Thanks!

EricMarcil, 2020-07-22:
What happens if you try to go faster?  You get an alarm?

lkeselman, 2020-07-22:
I guess I am missing something. How can I ask it to go faster? I tried the /move_group/trajectory_execution/velocity_scaling parameter but it seemed to have no effect. I assumed that MoveIt parametrizes the path to be as fast as possible based on robot limits. Is this not correct?

gavanderhoorn, 2020-07-22:
We're going to need more information.

The driver should be able to run the robot at (almost) 100% (I've easily made robots which were not correctly bolted to the floor "jump").

Please tell us:

 - how you are generating your trajectories
 - which support package you are using (if one provided here: link to it, if you've created one yourself: link us to it)
 - which MoveIt configuation you are using (if one provided here: link to it, if you've created one yourself: link us to it)
 - how you have you controller configured (no maximum/demo/safety speed limit active?)
 - how you are executing your trajectories

In general: please also include an example trajectory (you can use `rostopic echo /topic/on/which/your/action_server/listens/for/goals`, show relevant snippets of (rosconsole) logs (ie: copy-paste console output you see verbatim into a comment here) *and* tell us how you are starting things (again: verbatim copy-pastes).

lkeselman, 2020-07-22:
Thanks for the help! Here is the requested information

> how you are generating your trajectories

MoveGroup CPP Interface. I use the `plan` and `execute` functions separately rather than calling `move`.

> which support package you are using (if one provided here: link to it, if you've created one yourself: link us to it)

The one here: <https://github.com/ros-industrial/motoman/tree/kinetic-devel/motoman_gp7_support> branch:kinetic-devel

> which MoveIt configuation you are using (if one provided here: link to it, if you've created one yourself: link us to it)

Here is my MoveIt config: <https://github.com/lkeselman/motoman_moveit_debug>

> how you have you controller configured (no maximum/demo/safety speed limit active?)

I believe I've only followed the instructions here but in terms of speed limit, I believe. All the settings under UTILITY->SETUP SPECIAL RUN say `INVALID`. The fact that it runs much faster if executing an INFORM program under the same settings would also indicate that no clamping is in play I would think.

> how you are executing your trajectories

I'm using the MoveGroup CPP interface `execute` API. Everything is running on my laptop.

>please also include an example trajectory

[example_trajectory.txt](https://github.com/ros-industrial/motoman/files/4962236/example_trajectory.txt)

> tell us how you are starting things

roslaunch motoman_driver robot_interface_streaming_yrc1000.launch robot_ip:=10.40.0.121
(Starting MoveIt is a bit complicated, I can share more info later if needed)

Console log from starting MoveIt (relevant bits only)
```
PARAMETERS
 * /controller_joint_names: ['joint_1_s', 'jo...
 * /hardware_control_loop/loop_hz: 500
 * /hardware_interface/joints: ['joint_1_s', 'jo...
 * /joint_state_controller/publish_rate: 500
 * /joint_state_controller/type: joint_state_contr...
 * /joint_state_publisher/use_gui: False
 * /move_group/allow_trajectory_execution: True
 * /move_group/capabilities: 
 * /move_group/controller_list: [{'default': True...
 * /move_group/disable_capabilities: 
 * /move_group/endeffector/default_planner_config: 
 * /move_group/endeffector/planner_configs: ['SBL', 'EST', 'L...
 * /move_group/generic_hw_control_loop/cycle_time_error_threshold: 0.01
 * /move_group/generic_hw_control_loop/loop_hz: 300
 * /move_group/hardware_interface/joints: ['joint_1_s', 'jo...
 * /move_group/hardware_interface/sim_control_mode: 1
 * /move_group/jiggle_fraction: 0.05
 * /move_group/joint_state_controller/publish_rate: 50
 * /move_group/joint_state_controller/type: joint_state_contr...
 * /move_group/manipulator/default_planner_config: PersistentLazyPRM...
 * /move_group/manipulator/enforce_joint_model_state_space: True
 * /move_group/manipulator/longest_valid_segment_fraction: 0.005
 * /move_group/manipulator/planner_configs: ['SBL', 'EST', 'L...
 * /move_group/manipulator/projection_evaluator: joints(joint_1_s,...
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: False
 * /move_group/moveit_sim_hw_interface/joint_model_group: todo_group_name
 * /move_group/moveit_sim_hw_interface/joint_model_group_pose: todo_state_name
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BFMT/balanced: 0
 * /move_group/planner_configs/BFMT/cache_cc: 1
 * /move_group/planner_configs/BFMT/extended_fmt: 1
 * /move_group/planner_configs/BFMT/heuristics: 1
 * /move_group/planner_configs/BFMT/nearest_k: 1
[... 282 lines cut ...]
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1595446879.934795652]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1595446879.934813977]: MoveGroup context initialization complete

You can start planning now!

QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
[ INFO] [1595446880.912735341]: Ready to take commands for planning group manipulator.
[ INFO] [1595446880.912826769]: Looking around: no
[ INFO] [1595446880.912868544]: Replanning: no
[ INFO] [1595446886.348567545]: Stopping planning scene monitor
```

I'm including joint_states bags for both the INFORM and ROS case. To me it seems that the acceleration rate commanded by MoveIt is very low but it's not clear to me how it decides that if the robot has no acceleration constraints such as this case.

[joint_state_bags.zip](https://github.com/ros-industrial/motoman/files/4962203/joint_state_bags.zip)

Thanks again for the help.

lkeselman, 2020-07-22:
So I think I figured out what's going on here. In the case that acceleration limits for each joint are not specified, MoveIt uses the value 1.0rad/s^2. This is well below the limit of the robot. Based on the rosbag of the INFORM path from the previous comment, I determined the J1 accel limit to be at least 13rad/s^2.

For an experiment, I updated joint_limits.yaml for J1 to have accel limits and for that to be equal to 12. The robot moves much faster, but now a different axis is limited by the 1 rad/s^2 limitation.

Can someone point me to where the actual acceleration limits for the robot come from?

gavanderhoorn, 2020-07-24:
> So I think I figured out what's going on here. In the case that acceleration limits for each joint are not specified, MoveIt uses the value 1.0rad/s^2. This is well below the limit of the robot. [...]

Seeing your MoveIt configuration, I was indeed going to tell you that not having any acceleration limits configured will result in MoveIt using a default of `1.0 rad/s^2`.

I also see quite a few 'strange things' in your MoveIt configuration and console copy-paste: why is there a `joint_state_publisher` started? Why the mention of `ros_control` parameters and nodes? `motoman_driver` is not `ros_control` compatible.

You probably have other hardware which is, and does use `ros_control`, correct?

> Can someone point me to where the actual acceleration limits for the robot come from?

Afaik we never release maximum joint acceleration information. @EricMarcil: would this fall in the same category as what was discussed in #322?

gavanderhoorn, 2020-07-24:
And some other comments:

> > tell us how you are starting things
> 
> ```
> roslaunch motoman_driver robot_interface_streaming_yrc1000.launch robot_ip:=10.40.0.121
> ```

Please try to avoid starting `robot_interface_streaming_yrc1000.launch` and similar `.launch` files directly.

You're already using `motoman_gp7_support` which comes with the necessary convenience `.launch` files in the `launch` directory, such as `robot_interface_streaming_gp7.launch`. That `.launch` file should be used to start the driver for your robot.

> Starting MoveIt is a bit complicated, I can share more info later if needed)

I'd be interested, as I don't see any of the "normal" `.launch` files for it.

EricMarcil, 2020-07-24:
Yes, it would.

I used to have a value of the time to reach maximum speed that I could use to estimate the acceleration by taking the maximum speed divided by that time.
But that value is now obsolete on newer robots because the acceleration is modulated based on the robot current inertia.  So I can't rely on it.

From past experience, that value would be between 100 to 650 ms.
I've never seen anything below 100 ms.
Generally, the L and U axis have the longest time between 400 and 650 ms.
Smaller axis RBT would be somewhere between 100 and 300 ms.

So you can probably start with a value around 500 ms for SLU axis and 300 ms for RBT to estimate the acceleration.
If you get any alarms it should indicate which motor, then you will want to decrease the acceleration (increase acc time).
If your process is time critical and your not getting alarms, you could gradually increase the acceleration.

You'll want to evaluate with the arm in a configuration that maximize the inertia of the arm to get the worst case scenario.

You basically want to monitor the motor torque.  You can display the torque information on the pendant by setting the security level to management, then under the menu ROBOT --> SERVO MONITOR you can display information.  You can change the displayed information pressing the DISPLAY button on the top.

There is also a Pendant Oscilloscope function that is available on some system.  You can lookup the manual on our website: 
[https://www.motoman.com/en-us/service-training/product-documentation](https://www.motoman.com/en-us/service-training/product-documentation)

lkeselman, 2020-07-24:
Thanks for the help @gavanderhoorn and @EricMarcil . I figured this would be a guess-and-check process. Besides the intellectual property concerns, I think the issue is just that MoveIt time parameterization is too simplistic and doesn't work at all like industrial robot motion control does. Anyway, it's good to have a procedure to improve the performance. I'll close the issue. Thanks again!

closed 2020-07-24 by lkeselman

gavanderhoorn, 2020-07-25:
@lkeselman: I'd still be interested in some more information on how you're using the driver, especially icw MoveIt. From your description, the console output and your MoveIt configuration, it doesn't appear you're applying best-practices.

lkeselman, 2020-07-27:
@gavanderhoorn there is a lot going on there and I can't share everything here due to IP concerns. My launch file launches different robots and other devices based on parameters. You're right in that it hasn't been productized yet and there is quite a bit of unnecessary launching going on. I will take the advice you listed earlier and apply it to improve the launching. Thanks again for your help and advice.