ros-industrial/motoman #347: GP25 robot simulation and frame assignments

arsh09, 2020-07-29:
Hello, 

Intead of using GP25 meshes provided in the support package, I downloaded the CAD model, use Solidworks to URDF package to convert the model to URDF. As I am creating Gazebo simulation (and support package does not have inertia tags in robot description). 

Adding tool0 frame is a bit difficult for me. Can anyone tell me where on each link is the frame origin, in the support package? Is there any Yaksawa standard that is being followed as I used ros indstrial frame standard ( Z up, X forward). 

When I looked at gp25 support package, the frames on each mesh seemed to be place differenty than how I placed them (starting of each link's sub assembly as origin of frame). That is why I can not figure out the exact transformation from link_6_t to tool0. 

I figured if I know how the support package meshes were imported and what frame assignment standard they followed, I can import the SW CAD to URDF using the same standard and then put tool0 in urdf manually. 

Thank you.

gavanderhoorn, 2020-07-29:
Your post reads a bit like an [xy-problem](http://xyproblem.info).

> Intead of using GP25 meshes provided in the support package, I downloaded the CAD model, use Solidworks to URDF package to convert the model to URDF. As I am creating Gazebo simulation (and support package does not have inertia tags in robot description).

Could you clarify *why* you redo the meshes?

Is it to use SolidWorks mass and inertia estimation?

If so, you may want to take a look at this article: [Inertial parameters of triangle meshes](http://gazebosim.org/tutorials?tut=inertia&cat=build_robot).

It would allow you to use the provided support package and extend it to add the necessary elements without needing to redo any of the meshes.

arsh09, 2020-07-30:
@gavanderhoorn Yes, I wanted to use SW estimation. I did try to use the meshlab and get  inertial parameters. However the support package meshes have very less number of faces and that in turn does not look so good in visual (I am aware that visual and collision tags use different STL meshes). 

If that is the only solution. I will try it. Nonetheless I would like to know how the frames were assigned for any future references. 

Thank you for your response.

closed 2020-07-30 by arsh09

gavanderhoorn, 2020-07-30:
> If that is the only solution. I will try it. 

No, it's not the *only solution*.

> Nonetheless I would like to know how the frames were assigned for any future references.

Nothing special really:

 - put the robot in its zero-pose
 - place frames at the "logical joint positions" (ie: where they are defined by the mfg)
 - for everything: follow [REP-103](https://www.ros.org/reps/rep-0103.html#coordinate-frame-conventions)

As for which frames to include: [Coordinate Frames for Serial (Industrial) Manipulators](http://gavanderhoorn.github.io/rep/rep-0199.html) (a WIP).