ros-industrial/motoman #351: Validation failed: Trajectory doesn't start at current position.  (Using GUI)

Pranav186, 2020-08-12:
Hello,
The goal of my project is to execute trajectories on the Motoman MHJ using a GUI panel on rviz. I will have to explain the details of this GUI first to get to the problem.
In the GUI panel, I have put 6 QDoubleSpinbox buttons for 6 joint angles and a ‘move’ button. The user inputs the 6 joint angles and presses the move button on the GUI to command the robot to go to the desired joint angles. In the back end, I am actually running a modified version of  simple_trajectory_action_client.py file that subscribes to a topic named 'hmi_joint_angles' that the GUI publishes. 

Observations-
The trajectory gets executed the first few times (6-8 times approx). After that, I get the error when I try to execute the next trajectory. I am not sure if it is a glitch in the robot or a glitch in my code. 
Following is the node that is run everytime 'move' is pressed on the GUI. It is a modified version of simple_trajectory_action_client.py-

```
#!/usr/bin/env python

# Note: joint names must match those used in the urdf, or the driver will
#       refuse to accept the goal.

import rospy
import actionlib
import time
import math
import sys

from control_msgs.msg import FollowJointTrajectoryAction, FollowJointTrajectoryGoal
from sensor_msgs.msg import JointState
from trajectory_msgs.msg import JointTrajectory, JointTrajectoryPoint

flag = 0
q1=[0,0,0,0,0,0]
def exeFct():
	global flag, q1
	while not rospy.is_shutdown():
		if flag:
			client = actionlib.SimpleActionClient('joint_trajectory_action', FollowJointTrajectoryAction)
			client.wait_for_server()
			rospy.loginfo('Connected to trajectory action server')

			# setup simple goal
			goal = FollowJointTrajectoryGoal()
			goal.trajectory = JointTrajectory()

			goal.trajectory.joint_names =  ['joint_1_s', 'joint_2_l', 'joint_3_u', 'joint_4_r', 'joint_5_b', 'joint_6_t'] #check that these match with the ones in the robot_joint_states

			# motoman_driver only accepts goals with trajectories that start at the
			# current robot state, so retrieve that and use it as the first point in
			# the trajectory
			robot_joint_states = rospy.wait_for_message('joint_states', JointState)

			# make sure the state we get contains the same joints (both amount and names)
			if set(goal.trajectory.joint_names) != set(robot_joint_states.name):
				rospy.logfatal("Mismatch between joints specified and seen in current "
					"JointState. Expected: '{}', got: '{}'. Cannot continue.".format(
						', '.join(robot_joint_states.name),
						', '.join(goal.trajectory.joint_names)))
				sys.exit(1)
			
			q0 = robot_joint_states.position
			
			# Make the robot come to a complete stop at each trajectory point (ie:
			# zero target velocity).
			qdot = [0.0] * len(goal.trajectory.joint_names)
		
			# add points

			goal.trajectory.points.append(JointTrajectoryPoint(positions=q0,
				velocities=qdot, time_from_start=rospy.Duration(0)))
		
			print("goal is set to:")
			print(q1)
			goal.trajectory.points.append(JointTrajectoryPoint(positions=q1,
				velocities=qdot, time_from_start=rospy.Duration(3.0)))

			# goal constructed, submit it for execution
			rospy.loginfo("Submitting goal ..")
			client.send_goal(goal)
			rospy.loginfo("Waiting for completion ..")
			client.wait_for_result()
			rospy.loginfo('Done.')
			flag=0

def callback(msg):
	global flag,q1 
	flag = 1
	q1=msg.position
	print(q1)

def main():
    rospy.Subscriber('hmi_joint_angles', JointState, callback)
    exeFct()

if __name__ == '__main__':
    rospy.init_node('hmi_node')
    
    # create client and make sure it's available
    
    rospy.loginfo('Waiting for driver\'s action server to become available ..')
    main()
```

gavanderhoorn, 2020-10-05:
I'm not sure this is still an issue, but as a general comment: you're re-creating the `SimpleActionClient` and other related infrastructure every time you run through the loop. Additionally, there is no rate limiting anywhere.

Those are at least two things I'd change.

gavanderhoorn, 2020-10-05:
As this does not appear to be issue with the driver immediately (you report it works "the first few times"), I'm going to close this.

If you feel otherwise, please update and re-open.

closed 2020-10-05 by gavanderhoorn

Pranav186, 2020-10-06:
Thanks for the reply @gavanderhoorn .

> Additionally, there is no rate limiting anywhere.

 I think you are right that the absence of Rate is causing the problem and it is not a glitch in the driver. I will check it and comment back if there is still an issue. Hopefully it should be solved.