ros-industrial/motoman #352: motoman_driver MotoPlus copyright

lights0123, 2020-08-15:
The ROS packages in this repository are licensed under the 3-BSD license, [all having a proper BSD header](https://github.com/ros-industrial/motoman/blob/7fdb8b8a0e77166c420634ee42827361201c029c/motoman_driver/src/motion_ctrl.cpp#L1-L30) in every source file. However, the source files for the MotoPlus code, well, don't. Although [they say](https://github.com/ros-industrial/motoman/blob/7fdb8b8a0e77166c420634ee42827361201c029c/motoman_driver/MotoPlus/mpMain.c#L14) "Software License Agreement (BSD License)", the actual license text is curiously different:

https://github.com/ros-industrial/motoman/blob/7fdb8b8a0e77166c420634ee42827361201c029c/motoman_driver/MotoPlus/mpMain.c#L14-L39

Specifically, the correct 3-BSD header, compared to the one in the source files, is

> ```diff
> - Redistribution and use in source and binary forms, with or without
> - modification, are permitted provided that the following conditions are met:
> + Redistribution and use in binary form, with or without modification,
> + is permitted provided that the following conditions are met:

It is pretty clear that someone intentionally removed the "source and" mention. [Googling that string](https://www.google.com/search?hl=en&q=%22Redistribution%20and%20use%20in%20binary%20form%2C%20with%20or%20without%20modification%2C%20is%20permitted%20provided%20that%20the%20following%20conditions%20are%20met%22) leads to many instances of proprietary licenses, but nothing looking like 3-BSD. Additionally, the following clause was removed:
> ```diff
> - * Redistributions of source code must retain the above copyright
> - notice, this list of conditions and the following disclaimer.

What is the proper license for these files? I have some modifications planned for the MotoPlus code, but am hesitant to do so if a dubious license is present. I'd like one of these to happen:
- the "(BSD License)" is removed from the header of these files as this isn't BSD
- each file is updated to include the actual BSD license. It looks like all substantial changes [have been made by people who are obviously Yaskawa employees](https://github.com/ros-industrial/motoman/commits/kinetic-devel/motoman_driver/MotoPlus) (@EricMarcil and @ted-miller) except possibly [this one](https://github.com/ros-industrial/motoman/commit/138effbdcbb3b080b3599641c0e64e882e379113#diff-208056fb6d5c96d4960b9ad847511ed3), so a relicense should be simple.

I am aware that the actual MotoPlus SDK is proprietary—I'm just wondering about the source files present in this repository.

gavanderhoorn, 2020-08-16:
A nuance/clarification here: this is not about MotoPlus itself (a Yaskawa product), but about MotoROS. The latter has the former as a dependency, but they are not the same thing. Nor do they fall under the same licenses.

lights0123, 2020-08-16:
That's true—I was referring to the MotoROS code as the code used *with* MotoPlus, not that it *is* MotoPlus.

ted-miller, 2020-08-17:
It took some digging, but I found where this originated. It comes from a discussion between myself and shaun-edwards back in 2012. 

The MotoROS project includes some compiled libraries (ParameterExtraction.*Lib) to hide proprietary code. These libraries will not be released as open source. Thus, the "source" text was removed.

Any additional restrictions are laid out in the MotoPlus SDK license agreement. Beyond that, you're free to do whatever you want with the code that is publicly posted. This code is frequently used by other parties as a starting template for their own proprietary implementations.

lights0123, 2020-08-17:
Ok, so that license applies mainly to the final binaries, as we can't redistribute the source of the proprietary libraries.

closed 2020-08-17 by lights0123