ros-industrial/motoman #354: Time Optimal Trajectory Generation (TOTG) - Jerk Limited

garyedwards, 2020-08-27:
I am considering updating my MoveIt config for our GP8 to use TOTG for time parameterization.  I understand that this algorithm does not obey jerk limits of robots.  As this driver uses trajectory playout, does Yaskawa's black boxes handle the jerk limits for the hardware and mitigate this issue or is this something to be concerned about?

Thanks for any insights.

gavanderhoorn, 2020-08-27:
None of MoveIt's time-parameterisation algorithms are jerk-limited nor aware.

So if what you currently use works, TOTG would also work.

gavanderhoorn, 2020-08-27:
As this is not an issue with the packages in this repository, I'm going to close it.

That does not mean you can't continue the discussion of course.

Please feel free to keep commenting.

closed 2020-08-27 by gavanderhoorn