ros-industrial/motoman #356: How to implement impedance control in motoROS

zhangOSK, 2020-09-30:
Hello! I want to implement impedance control in SDA5F with FS100.  I am confused by two questions:
1. How to get force information from force sensor through ROS? 
I am using a DynPick force sensor and it can be monitored by MotoFit FSE-Tool in windows. But I have no idea how to read force  from ROS.
2. Does the current motoRos support impedance control? Do I have to create a new .out file to implement force control?

Could you please help to give me some suggestions?
Thanks in advance for your help!

gavanderhoorn, 2020-09-30:
> 1\. How to get force information from force sensor through ROS?
>        I am using a DynPick force sensor and it can be monitored by MotoFit FSE-Tool in windows. But I have no idea how to read force  from ROS.

Getting these values out of MotoROS is not supported by default.

It could be possible to add this, but it would have to be a custom addition.

Reading/writing variables (or similar concepts) through MotoROS would make a nice addition, so if you're willing to work on it we could see whether we can support it.

> 2\. Does the current motoRos support impedance control? Do I have to create a new .out file to implement force control?

The direct answer here would be: no, there is no support. MotoROS currently only supports trajectory execution, with trajectories specified as motions between positions with desired velocities and accelerations.

Closing the loop on the host machine would be an alternative, but control delays will make this difficult. See #219 and related discussions.

EricMarcil, 2020-09-30:
If you want to explore the use of MotoFit sensor, you will need to purchase (if you haven't already have it) the MotoPlus SDK to be able to modify and recompile the MotoRos application.

For the Motofit force sensor, there are MotoPlus APIs specifically for it.  You can refer to the MotoPlus Programming manual section 12 Servo Control and the API manual section 15 Servo Monitor/Control API section.  

The MotoFix application was developed in Japan. I haven't used the MotoFit option or tested out the related API.  So you'll have to review the documentation and test the function out and see what you can get out of it. Depending on where you are in the world, your local Yaskawa representative should be able to put you in contact with someone that can answer MotoPlus development related questions.

gavanderhoorn, 2020-10-05:
I'm going to close this, as right now it's not something we can really address here in this repository.

@zhangOSK: if you make any progress, do please update us.

You can still comment on a closed issue, so feel free to do so.

closed 2020-10-05 by gavanderhoorn