ros-industrial/motoman #359: ROS 2 driver version

gavanderhoorn, 2020-10-08:
This issue would be the place to discuss development approaches for a ROS 2 version of `motoman_driver`.

There is no requirement to keep the current architecture, nor to host any eventual results in this repository necessarily. Different architectures could be considered and new repositories can be created.

gavanderhoorn, 2020-10-15:
@tingelst: you appear to have a fork with an `eloquent` branch.

Could you comment here on the status of that work?

tingelst, 2020-10-15:
@gavanderhoorn 
We received three new robot cells with gp25 manipulators some months ago, and we are now working on setting these up. I have also just received a license for MotoPlus and have gotten that up and running with the current ROS1 driver.

I started some initial experiments with ROS2 in the `eloquent` branch around summer time this year, as we want to move all our robots towards ROS2. However, I didnt get very far at that time. 

I would very much like to be a part of this discussion, and will also be able to contribute with development work as getting a ROS2 driver for out motoman robots is a part of our long term strategy.

tingelst, 2021-01-07:
@gavanderhoorn 
To start up this conversation again.

I have over the last months developed a pure Python 3 library for controlling MOTOMAN robots which you can find here: https://github.com/tingelst/moto. 

The `moto` library interfaces with the current ros industrial driver using the SimpleMessage structures. This library might be a starting point for a new ros2 driver.

Moreover, I have extended the robot driver for real-time control which is synced with the RT clock on the controller. The code can be found [here](https://github.com/tingelst/motoman/blob/realtime-motion-server/motoman_driver/MotoPlus/RealTimeMotionServer.c). An accompanying ros2 control hardware interface can be found [here](https://github.com/tingelst/motoman_hardware).

Has there been any ROS2 development efforts on your side?

gavanderhoorn, 2021-01-18:
Hi @tingelst.

Interesting work. I'll take a look at the packages later.

re: `ros2_control` hw iface: how do you deal with the latency in the control signals? Afaiu there is quite some latency in the incremental motion API (#219) and you seem to be using that as well.

tingelst, 2021-01-18:
Hi, @gavanderhoorn!

Due to the Christmas holidays and reduced lab work due to Covid-19 restrictions, we have not yet performed experiments regarding latency. This is something that is on the agenda this spring.