ros-industrial/motoman #360: Catch writes to 'unwritable' addresses on the ROS side and print informative error

gavanderhoorn, 2020-10-12:
With #355 basic read/write of individual IOs was added to `motoman_driver`.

As mentioned in the `.srv`, writing is only supported to certain ranges of addresses:

https://github.com/ros-industrial/motoman/blob/591a09c5cb95378aafd02e77e45514cfac3a009d/motoman_msgs/srv/WriteSingleIO.srv#L9-L12

Instead of letting writes requests to other addresses through, the driver should perhaps check which address is targeted and implement a fail-early policy (don't even forward the request to MotoROS).

It could then also print a better error message than just the current:

https://github.com/ros-industrial/motoman/blob/591a09c5cb95378aafd02e77e45514cfac3a009d/motoman_driver/src/joint_trajectory_streamer.cpp#L600

a more informative error would include why the write failed.

gavanderhoorn, 2020-10-12:
@ted-miller @EricMarcil: would you happen to know the ranges which are writable, or should we look it up in the manual(s)?

gavanderhoorn, 2020-10-12:
Related ROS Answers Q&A: [motoman_driver external IO writing](https://answers.ros.org/question/363259).

gavanderhoorn, 2020-10-12:
From the DX200 manual:

[image]

So that would be `[27010; 29567]` for Network Inputs and `[10010; 15127]` for General Outputs.

ted-miller, 2020-10-12:
DX100:
10010 - 12567
25010 - 27567

FS100: 
10010 - 11287
25010 - 26287

DX200: listed in your image above
YRC1000 and micro: Same as DX200.

ted-miller, 2020-10-12:
Also, keep in mind that the last digit will never exceed 7.

10010 - 10017 = Valid
10018 - 10019 = NOT VALID
10020 - 10027 = Valid

gavanderhoorn, 2020-10-12:
@ted-miller: thanks.

@marip8 @steviedale: if/when we introduce the stand-alone IO relay we can provide it with the information on the connected controller, so as to filter for the correct ranges.

gavanderhoorn, 2020-11-03:
@ted-miller: the DX200 *Reference Manual For New Language Environment MotoPlus* (`166580-1CD`) suggests the *interface panel* IOs are also writable:

[image]

I haven't checked for the other controllers, but are they similar?

steviedale, 2020-11-05:
@gavanderhoorn these address values are all in decimal right (not hex)?

steviedale, 2020-11-05:
@gavanderhoorn as far as identifying which controller is being used, one possible solution would be to create a separate `io_relay.launch` file for each type, then have each provide a ROS param that contains the name of the controller. What do you think?

gavanderhoorn, 2020-11-05:
Please check the discussion in #367. It may be more efficient to extend MotoROS to return appropriate error codes, and then add generic error handling code to the `io_relay`.

ted-miller, 2020-11-10:
> Please check the discussion in #367. It may be more efficient to extend MotoROS to return appropriate error codes, and then add generic error handling code to the io_relay.

See PR #369

gavanderhoorn, 2020-12-17:
This is addressed (in a slightly different way) by #371.

closed 2020-12-17 by gavanderhoorn