ros-industrial/motoman #361: I/O Interface Development

marip8, 2020-10-13:
Opening this thread as a discussion point for the development of I/O interface capabilities from the ROS driver

## Milestones
The tasks below are the currently planned high-level milestones for the I/O interface

- [x] Add I/O communication into joint trajectory streamer node
- [ ] Port I/O communication into stand-alone node
- [ ] Incorporate better error feedback for incorrect I/O addresses per #360 
- [ ] Add support for multi-group I/O reading/writing

## Future Development
- [ ] Implement I/O solution for ROS 2 driver
- [ ] Leverage high-speed ethernet server for I/O communication and other teach pendant functionality

gavanderhoorn, 2020-10-13:
@marip8 wrote:
> Add support for multi-group I/O reading/writing

what is this exactly @marip8 ?

Reading/writing bytes and words, or getting everything to work in both the single and multi-motion group code paths?

marip8, 2020-10-13:
I meant I/O support for bytes/strings/etc. I just changed the wording above

gavanderhoorn, 2020-10-13:
Ok.

Suggestion: split this into (at least) two items, as reading/writing bytes is already supported by/through MotoROS (see also [REP-I4: Assigned Message Identifiers for the Simple Message Protocol - Assigned Message Identifiers - Vendor Specific Messages - Motoman](https://github.com/ros-industrial/rep/blob/bbd88b9b1dc9b61be5178c8b3a93b9bec488702a/rep-I0004.rst#motoman)), while variables, strings and others like that would probably need the ethernet server.

cjue, 2020-10-15:
> Suggestion: split this into (at least) two items, as reading/writing bytes is already supported by/through MotoROS

Yes, just adding read/write byte/word support would be helpful. IMHO having read/write_byte/word_io services would already save time and complexity for simple applications.

gavanderhoorn, 2020-10-15:
While convenience is why we're adding them here, I'm going to repeat my advice (which I always give when talking about ROS, robot drivers and IO): consider using a proper fieldbus for industrial deployments. Having to add IO support to each and every robot driver doesn't scale very well, and will most likely use custom implementations of protocols which do not have the same level of support, integration nor testing as any of the available (software defined) fieldbuses in existence.

Even a simple wrapper around a modbus/tcp client would be a good option.

Having written that: we will add services to read & write bytes, as that is already supported by MotoROS.

Reading & writing single elements (ie: single bits) is already supported now that #355 is merged.

gavanderhoorn, 2020-10-15:
@steviedale: #362 would also be good to consider here.

gavanderhoorn, 2020-12-07:
I've marked "provide better error messages" as completed, as we now have input verification on the MotoROS side and error reporting on the `motoman_driver`.

gavanderhoorn, 2021-12-12:
WIP PR for "Group IO" support: #435.

gavanderhoorn, 2021-12-12:
WIP PR for "M Register" support: #439.

gavanderhoorn, 2021-12-14:
#438 (and the related PRs #435, #436 and #439) made it clear that including the MotoROS result (sub) code as part of the service response would actually be really nice. Right now we only include a human-readable error message, which is good, but doesn't work as well when trying to implement some automated recovery behaviour.

Adding a field to a message cannot be done in a bw-compatible way though (as in: the MD5 will change, even if the field is not a required field, or could be ignored), so changing the message definitions *will* require users to rebuild the driver.

(there would be no changes required to call sites though, as the new field(s) are additions)

gavanderhoorn, 2022-03-06:
Just merged #435 and updated the OP.

Keeping this open to track status of "add MotoROS result (sub) code to service response".