ros-industrial/motoman #363: MS210 Joint Effort Limits - Missing Data

acbuynak, 2020-10-16:
Working to build a ros-industrial robot support package for the Motoman MS210.

**Need help identifying MS210 effort limits for each URDF joint.**

The [motoman website](https://www.motoman.com/en-us/products/robots/industrial/welding-cutting/ms-series/ms165) provides values for each joint's max range, speed, allowable moment (joints rbt), allowable moment of intertia (joints rbt). From this I defined, joint limits and velocity limits.

My draft code is stored here: [OSU-AIMS/motoman/tree/kinetic-devel-add-ms210](https://github.com/OSU-AIMS/motoman/tree/kinetic-devel-add-ms210)
Hoping to submit a PR once this is resolved.

This is a cross-post to seek help from the Motoman team. Original Question from ROS Answers: https://answers.ros.org/question/363519/calculating-effort-limit-for-motoman-ms210/

EricMarcil, 2020-10-16:
Below is a csv file with the MS210 values. (also attached but renamed with .txt extension)
[Robot_Specs_MS210-A0.txt](https://github.com/ros-industrial/motoman/files/5392473/Robot_Specs_MS210-A0.txt)
 
I believe that the "Max Acc/Dec Torque[Nm]" corresponds to the Effort value that you are looking for.

[image]

gavanderhoorn, 2020-10-16:
With @EricMarcil's answer, I'm going to close this.

Feel free to keep commenting @acbuynak.

closed 2020-10-16 by gavanderhoorn

acbuynak, 2020-10-16:
Thanks for the additional information.

Seeing the Theta, D, A values.. I rechecked my URDF joint origins which were originally based on the MS210 "Instructions" Manual ([motoman part # 165835-1CD-rev5](https://www.motoman.com/en-us/service-training/product-documentation?d=A0B2D511-4B8D-41D8-A6A7-76E671203766/165835-1CD)) and diagram on the [motoman website](https://www.motoman.com/en-us/products/robots/industrial/welding-cutting/ms-series/ms165).

I was missing one dimensions for each of the S, L, R, B, T joint origins. I used the SW2URDF converter w/ my STL files to back out these numbers. Comparing the result to the robot's official max dimensions, two of my values are off by ~3 mm each. Ideally, I would like to make these accurate as possible.

@EricMarcil Can you share the absolute coordinate position for the physical joints SLRBT?
Not sure if this info is readily available to you on the backend as I've been pulling from multiple diagrams.

gavanderhoorn, 2020-10-17:
Looks to me like all required dimensions are listed on page 28 of the manual you link.

Could you be more specific about what it is you're actually missing?

acbuynak, 2020-10-17:
Yup. I've sketched the location of the missing dimensions onto the mentioned Page 28 Diagram.
[image]

gavanderhoorn, 2020-10-18:
Ah, now I see your confusion.

You've got the `origin`s of the joints wrong, hence you don't see the match with the D-H parameters.

`joint_s` does not have its origin where you place it here. It's in the X-Y plane which goes through `joint_l`. So in this case, its Z would be at `0.650 m`. That's also where the robot's `base` frame is (and is coincident with the 'start' of its kinematic tree).

I'd suggest you compare with other support packages we have here in this repository, such as the GP180:

https://github.com/ros-industrial/motoman/blob/591a09c5cb95378aafd02e77e45514cfac3a009d/motoman_gp180_support/urdf/gp180_120_macro.xacro#L105-L153

Notice:

 - never translate in two dimensions in a single `joint`
 - all meshes have their `origin`s set to `0 0 0`
 - all transforms always relative to the previous `joint` (so no shared `origin` at the base)

acbuynak, 2020-10-19:
Gotcha, Thanks.

I had been following the structure of the MH5 and MH50 and had misinterpreted the meaning of the [ROS-I URDF Tutorial's](http://wiki.ros.org/Industrial/Tutorials/Create%20a%20URDF%20for%20an%20Industrial%20Robot) note that the joint shall be `coincident with the joint nearest to the base of the robot.` I had understood this to mean the physical joint location, but without the priority of a single joint-translation.

Having this completed, I will move this into a PR.