ros-industrial/motoman #367: Filter illegal values passed to IO write service(s)

gavanderhoorn, 2020-10-30:
Similar to #360, the `io_relay` could check for invalid values being passed with write requests.

For single digital IOs, anything but `0` or `1` is illegal.

For group IO, there is a specific maximum value that could be written.

We could consider checking for illegal values in the `io_relay` and not pass those requests on to MotoROS. This would also allow us to return meaningful error messages to users.

gavanderhoorn, 2020-10-30:
Instead of checking on the ROS side, it would perhaps also be an idea to check on the MotoROS side.

There is some checking right now, but MotoROS does not return a very informative status code. It basically just says "didn't work".

@ted-miller: would it perhaps be possible / desirable to make MotoROS return specific causes for different failure modes?

That would remove the need to extend `io_relay` with this, and -- more importantly -- would allow the `io_relay` to setup generic error handling code with MotoROS being responsible for checking controller-specific cases. That would also help with #360: instead of having to tell `io_relay` which ranges are acceptable, MotoROS would handle this and return something like "invalid address: out of range".

I would imagine perhaps MotoPlus return codes could be mapped?

gavanderhoorn, 2020-10-30:
@marip8 @steviedale: thoughts?

ted-miller, 2020-11-03:
I'm fine with adding additional return codes. I'm assuming that you want this in the 'resultCode' field, and not in the header info.
https://github.com/ros-industrial/motoman/blob/4b0a030de3917aa13ab1407f2dcf131d4f0ffd27/motoman_driver/MotoPlus/SimpleMessage.h#L285-L290

gavanderhoorn, 2020-11-03:
Would you already have some result codes in mind? We could start with failure reported by MotoPlus' read/write functions.

gavanderhoorn, 2020-11-03:
Possible failures:

 - illegal value for target adress (ie: `123` for digital out)
 - illegal address (out of writable/readable range): should this be split into read and write errors respectively? Are the ranges identical?
 - "motoplus reported `-1`" (despite having checked the previous two bullets)

anything else? 

@ted-miller @EricMarcil @marip8 @steviedale: feel free to edit this comment (drop down in the top right of this comment).

gavanderhoorn, 2020-11-09:
@ted-miller: would you see an opportunity to add checking on the MotoROS side? Would you want someone else to contribute a PR to add this?

ted-miller, 2020-11-09:
Sorry for the delay. I'll try to get to it this afternoon. If not, then I'll do it first in the morning.

ted-miller, 2020-11-09:
Do you want the change in this repo, the new one, or both?

gavanderhoorn, 2020-11-09:
This one. I'll migrate the complete history to the other one -- it's a script.

gavanderhoorn, 2020-11-09:
@ted-miller @EricMarcil @steviedale @marip8: could you all take a look at https://github.com/ros-industrial/motoman/issues/367#issuecomment-721162583 and see whether there is a failure mode missing? Would be good to cover the expected failure modes with result codes in MotoROS.

gavanderhoorn, 2020-11-09:
Seeing as the discussion has shifted from filtering in `io_relay` to extending MotoROS with checks, I've updated the title.

steviedale, 2020-11-09:
@gavanderhoorn those failure modes look good to me. The only addition I can think of is having a different error code for addresses that don't exist and addresses that do exist but are not readable/writable. Not sure if this is worth adding another error code for or not, just a thought.

gavanderhoorn, 2020-11-09:
Ok. So `10001` would be an illegal address in all cases but `70010` is valid, but just not writable (from https://github.com/ros-industrial/motoman/issues/360#issuecomment-707246441, DX200).

Is that what you're suggesting?

steviedale, 2020-11-09:
Exactly. It depends on whether you guys think this is overkill. I can see both an argument for and against.

steviedale, 2020-11-09:
Is there an easily accessible lookup table for valid addresses and read/write availability of these? If it is easy to find, then maybe this is in fact overkill, as they can simply lookup the address to see why it's invalid.

gavanderhoorn, 2020-11-09:
The documentation is pretty clear imo. See https://github.com/ros-industrial/motoman/issues/360#issuecomment-707246441 for a screenshot of the table from the DX200 manual.

steviedale, 2020-11-09:
Gotcha. In that case, I think the `[READ/WRITE]_INVALID_ADDRESS` error codes suffice.

ted-miller, 2020-11-10:
PR #369

PR #371 (2020-12-04): Forward MotoROS (1.9.2+) IO service results to ROS clients
This PR uses the updated `result_code` values added in #369 (which hasn't been merged yet) to provide ROS service clients with more descriptive error messages in case something goes wrong.

Examples of messages returned:

```shell
$ rosservice call /read_single_io "address: 10018"
message: "Read failed (address: 10018): Illegal address for read: outside permitted range on this controller, see documentation (1001)"
success: False
value: 0

$ rosservice call /write_single_io "{ address: 0, value: 0 }"
message: "Write failed (address: 0): [... cut]
closed 2020-12-04 by gavanderhoorn