ros-industrial/motoman #368: Inconsistent end effector speed

wlawler45, 2020-10-30:
Hello I'm having an issue moving the motoman in a consistent manner, the RViz motion view looks fine when I plan with compute cartesian move, and I've tried a number of methods to try and replan and execute the trajectory correctly and it keeps but the real robot consistently has these jerky motions where it slows almost to a stop in the middle of the trajectory, the application we are working on requires very smooth trajectories, I believe it has something to do with the actual motoman ros code but I'm not sure. Has anyone else experienced this issue? If so what should I do? Any feedback is appreciated.

EricMarcil, 2020-10-30:
It sounds as if the packet transmission is slow.  Do you have a direct connection to the controller or are you going through an large network with losts of traffic?

wlawler45, 2020-10-30:
It should just be me and the controller on the ethernet switch, but I am running a virtualbox, would that make the difference?

EricMarcil, 2020-10-30:
Maybe the virtualbox is causing delay?  You might want to do a Wireshark capture of the communication and where the delay is.  How often is Ros sending motion message to the controller?  How fast is the reply from the controller? Is the controller replying with busy signal?

wlawler45, 2020-10-30:
could I check the busy signal with ros? I'll try that and let you know.

EricMarcil, 2020-10-30:
The MotoRos has an internal queue that can accumulate up to 800 ms of trajectory (it usually helps with communication variation).  When this is full and it gets more information, it will send back a busy signal.  The ROS side should automatically be resending the message until the queue cleared up and the next message is accepted. I'm not sure how to expose that on the ROS side.

wlawler45, 2020-10-30:
Another thing is its not consistently pausing, it pauses at some specific points along the curve, there shouldn't be any singularities or anything there either so I don't know why it's doing that.

wlawler45, 2020-10-30:
Here is a Rosbag with the trajectory in it, anything else I can provide just let me know. Thanks again for all the help.
[2020-10-30-13-01-13.bag.tar.gz](https://github.com/ros-industrial/motoman/files/5466835/2020-10-30-13-01-13.bag.tar.gz)

wlawler45, 2020-10-30:
If it is a moveit time parameterization issue, what would you recommend I use? What planner would give the smoothest time parameterization?

EricMarcil, 2020-10-30:
The data transmission looks good.  I can see two trajectory being send.
The transmission time is shorter than the commanded trajectory time.
I can see the initial command being accepted right-away and then once the motion buffer is full reply with busy response.
The motion point is repeatedly sent until space frees up and it is accepted.

I don't have a good tool for visualizing the motion but it does look in the 2nd trajectory that speeds are fluctuating up and down. So I would tend to say the problem is on the path generation side.

wlawler45, 2020-11-02:
I've tried another few planners in OMPL but I'm still having the same issues @gavanderhoorn do you have any suggestions to help get a better timed trajectory so I have smoother cartesian motions?

gavanderhoorn, 2020-11-03:
Without knowing exactly what you are doing, I can only provide suggestions, but perhaps using one of the alternative [time-parameterisation algorithms](https://ros-planning.github.io/moveit_tutorials/doc/time_parameterization/time_parameterization_tutorial.html). The spline based one works OK-ish in most cases. But perhaps TOTG is better.

I haven't had time to look at the trajectory yet. ROS messages easily serialise to YAML, so next time please just `rostopic echo -n1 /topic/with/the/goal > trajectory.yaml` instead of creating a `.bag`. 

I'll extract the message and plot it.

wlawler45, 2020-11-03:
Okay, I am using ComputeCartesianPath if that helps, I'll look into the other time-parameterisation algorithms there and see if I can get the real robot moving in a more smooth fashion. I'll keep that in mind in the future, thank you for letting me know.

gavanderhoorn, 2020-11-03:
Here is the trajectory from the bag, which is sent to the trajectory relay:

<details>

<summary>trajectory</summary>

```yaml
header: 
  seq: 3
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: "/base_link"
joint_names: [joint_1_s, joint_2_l, joint_3_u, joint_4_r, joint_5_b, joint_6_t]
points: 
  - 
    positions: [0.5133419632911682, 0.8950279355049133, -0.10985757410526276, -3.139145612716675, 0.5655623078346252, -3.6498637199401855]
    velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
    accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
    effort: []
    time_from_start: 
      secs: 0
      nsecs:         0
  - 
    positions: [0.5172545410769429, 0.8850690967591631, -0.1291465265560356, -3.1393325465443014, 0.5562684494494176, -3.653663713123703]
    velocities: [0.038794768788364714, -0.09700459010378445, -0.18788049446231297, -0.0018774922633456692, -0.09052512091335832, -0.0376401739084548]
    accelerations: [0.20947986037779265, -0.501944784903248, -0.9721188262927587, -0.010438655312763521, -0.4683891216147727, -0.2027651253439279]
    effort: []
    time_from_start: 
      secs: 0
      nsecs: 144251009
  - 
    positions: [0.5212218245884644, 0.8752447850411885, -0.1481742015505568, -3.139525862596129, 0.5471004801586205, -3.6575108202757]
    velocities: [0.058400074950467104, -0.14241174764715686, -0.2758155544067473, -0.002877692816461701, -0.13289414442212585, -0.05658082160210583]
    accelerations: [0.22788735539880955, -0.5009548343101077, -0.9700523480812194, -0.0120235221968328, -0.4673904879554801, -0.2195421244160267]
    effort: []
    time_from_start: 
      secs: 0
      nsecs: 222863791
  - 
    positions: [0.5252448431569021, 0.8655500619320319, -0.16695006938475973, -3.1397257758942314, 0.5380538497130061, -3.661405884268146]
    velocities: [0.07266116676413648, -0.17256516927194512, -0.33419978372464004, -0.003649458798820532, -0.16102465804309118, -0.0702903721177231]
    accelerations: [0.22536892393072974, -0.4528138941838881, -0.8766516211278947, -0.012579496103542549, -0.42238219069795296, -0.21607225890740844]
    effort: []
    time_from_start: 
      secs: 0
      nsecs: 283511692
  - 
    positions: [0.5293246465584839, 0.8559803327159026, -0.18548293918291334, -3.139932514281873, 0.5291243244548177, -3.665349753556429]
    velocities: [0.08475480171409511, -0.19599890031651462, -0.3795630399389395, -0.004340033856126949, -0.18288107042913654, -0.08186233522045858]
    accelerations: [0.23691380649094723, -0.4404741462314851, -0.8525707162987768, -0.013862372806202231, -0.41077162691066543, -0.22620422273167204]
    effort: []
    time_from_start: 
      secs: 0
      nsecs: 335162408
  - 
    positions: [0.5334623068606154, 0.846531313933709, -0.20378102126939904, -3.140146316755093, 0.5203079578166235, -3.669343286286238]
    velocities: [0.09460271426189551, -0.21308551516433782, -0.41262831236521136, -0.004938511268648259, -0.19881173974058958, -0.09123256854730596]
    accelerations: [0.18502564763408902, -0.2885694290981018, -0.5582235748999362, -0.01183509969173677, -0.26894063442847377, -0.17518791197028505]
    effort: []
    time_from_start: 
      secs: 0
      nsecs: 380871663
  - 
    positions: [0.5376589189045335, 0.8371990076719236, -0.22185197655087432, -3.1403674324567614, 0.5116010668432215, -3.673387352119391]
    velocities: [0.09566716897181779, -0.21005865204531937, -0.4067417157078908, -0.005088627198819227, -0.19597448579589943, -0.09211920379674385]
    accelerations: [-0.13640791802544874, 0.42465997895075325, 0.8228982727810994, 0.0050164952793372835, 0.39651029267094545, 0.13463606815956142]
    effort: []
    time_from_start: 
      secs: 0
[... 852 lines cut ...]
    accelerations: [0.694459854724131, -0.4700972855248915, -0.908184310253764, -8.575516357695151e-08, -0.4380871906197677, -0.7559435989337914]
    effort: []
    time_from_start: 
      secs: 5
      nsecs: 207835277
  - 
    positions: [-0.3866807247127202, 0.757033158073786, -0.376963859129301, -3.1415931941080464, 0.4367990700307816, -2.5419881475588]
    velocities: [-0.08672686724143215, 0.06683570661720233, 0.12918882101918894, 1.876712252297399e-08, 0.062353136091081146, 0.09478865195968372]
    accelerations: [0.6723609604728191, -0.5046515468248342, -0.9753560815411844, -1.32717177290853e-07, -0.4707047013050805, -0.7342282866424148]
    effort: []
    time_from_start: 
      secs: 5
      nsecs: 271722623
  - 
    positions: [-0.3945443885437166, 0.7632146730598881, -0.3650145111392737, -3.1415931922914924, 0.44256690501530954, -2.53338781902601]
    velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
    accelerations: [0.611344865782534, -0.4805695577442382, -0.928978234593024, -1.4122437026599838e-07, -0.44840883082679234, -0.6686153942440518]
    effort: []
    time_from_start: 
      secs: 5
      nsecs: 432115225
---
```

</details>

gavanderhoorn, 2020-11-03:
And here is a plot of the commanded positions, velocities and accelerations per joint:

[image]

No derivatives, all values straight from the trajectory.

At this point I'm inclined to believe the driver is doing what is asked of it, and the cause of what you observe would be in the trajectory.

Notice especially the acceleration plots. That's not a very nice trajectory.

gavanderhoorn, 2020-11-03:
And seeing those accelerations almost bounded by `+- 1 rad/sec^2`: do you have correct/appropriate/reasonable acceleration values configured in your MoveIt configuration?

If you don't (and by default there will be no acceleration limits configured), MoveIt's time-parameterisation algorithms will use `1 rad/sec^2` for all joints.

gavanderhoorn, 2020-11-03:
@gavanderhoorn wrote:
> At this point I'm inclined to believe the driver is doing what is asked of it, and the cause of what you observe would be in the trajectory.

As such, I'm going to close this issue.

If it turns out this is something we need to address here (in `motoman_driver` or MotoROS), we can update and re-open.

Feel free to keep commenting on the issue of course.

closed 2020-11-03 by gavanderhoorn

EricMarcil, 2020-11-03:
@gavanderhoorn Thanks for plotting that out and confirming what I was seeing.

@wlawler45 What controller and robot are you using?  Are you using a URDF from the repo or one that you created?

wlawler45, 2020-11-10:
@EricMarcil Hello again, so I'm using the YRC1000 and the GP12, and I should be using a URDF from their repo with just an added simple end effector, question as well, if I'm going to use FTP, with the Robodk post processors I need a remote file location on the motoman to save the programs to do you know what that would be? Let me know. Thanks.

gavanderhoorn, 2020-11-10:
@wlawler45: could you please not post off-topic comments here?

This issue tracker is for issues related to ROS, specifically the `motoman` packages.

Please use appropriate support channels for other questions.

EricMarcil, 2020-11-10:
@wlawler45 You can reach at my Motoman address: 
eric.marcil@motoman.com