ros-industrial/motoman #370: Xacro mh12 correction

leitoxve, 2020-11-16:
in 
https://github.com/ros-industrial/motoman/blob/kinetic-devel/motoman_mh12_support/urdf/mh12_macro.xacro

inverse the flench translate values in xacro 
line 144 with 149
**ORIGINAL**
144-origin xyz="0 0 0" rpy="0 0 0" 
149-origin xyz="0.100 0 0" rpy="3.1415926 -1.570796 0"
to 
**MODIFICATION**
144-origin xyz=**"0.10 0 0**" rpy="0 0 0" 
149-origin xyz=**"0.000 0 0"** rpy="3.1415926 -1.570796 0"

gavanderhoorn, 2020-11-16:
It would be good if you could provide a little more context and explanation here.

Do you have a particular reason you want this change applied?

EricMarcil, 2020-11-16:
The standard is that Tool0 represent the flange and it should not be edited.
If you are adding a tool to the robot then you should create a new joint with Tool0 as the parent.
Technically, the change that you are proposing shouldn't change the result.  
The way it is currently done, matches the way Yaskawa does its modeling in our simulator (MotoSim) except that we call Flange what Ros calls Tool0.

gavanderhoorn, 2020-11-16:
I'll try not to confuse everyone by mentioning the `flange` frame (I'll mark my comment as off-topic), but we do have a `flange` frame. It just wouldn't be aligned with any TCP, but with REP-103 standard orientation of frames. This would be for connection of EEF geometry to xacro macros from robot support packages.

See [REP-199](http://gavanderhoorn.github.io/rep/rep-0199.html) for a WIP proposal (I really need to finish that). Ignore the *Vendor Nomenclature Mapping*, that will be removed).

For additional tool frames, they should not be made children of `tool0`, but siblings, and then have `flange` as their parent.

`tool0` must not be changed, and the difference between `tool0` and `flange` would be the orientation only on most robots.

gavanderhoorn, 2020-11-20:
I'm going to close this, as I believe @EricMarcil's response should clear this up.

@leitoxve: please feel free to keep commenting on the issue.

If it turns out we should actually change or fix something here, we'll re-open.

leitoxve, 2020-11-27:
what I mean is that in real life the robot does not have a washer in the last joint  in the middle of nowhere
[image]

EricMarcil, 2020-11-27:
Oops, you are correct that shouldn't be there. 
Sorry, for misunderstanding your initial message.
Now that the issue is clear. 
You suggestion would fix it but to remain consistent with other models, it's probably the origin of the model that needs to be fixed.  I'll double check and make a PR to correct the model.

EricMarcil, 2020-11-30:
Submitted PR #372 to fix this issue.

gavanderhoorn, 2020-12-04:
Thanks for reporting @leitoxve, this was indeed a mistake in the model.

Should be fixed now with the merge of #372.

closed 2020-12-04 by gavanderhoorn