ros-industrial/motoman #373: Update README.md

Gaspard-Bourgeois, 2020-12-01:
Does anyone can add in the Readme.md the Yaskawa commercial names compatible for each package. For instance the MA1440 match the package of "motomam_m12_support". Does a list exist somewhere ?

Thanks a lot for your help.

gavanderhoorn, 2020-12-01:
No, we don't have such a list.

The general convention in ROS-Industrial is to create so caled *robot support packages*, which would contain all the variants of a specific robot series or model.

For Motoman robots however, robot support packages almost always support only a single robot variant. So `motoman_mh12_support` supports just the MH12 and `motoman_gp12_support` supports just the GP12.

The only exception right now (in this repository) are `motoman_hc10_support`, which supports both the base model and the `DT` variant and `motoman_hc20_support`, which supports the base model and the `DT` and `XP` variants.

So to figure out which package supports which robot:

 1. look at the name of the package and determine which robot variant it supports
 1. take a look at the `package.xml` of that package, specifically the `description`

If specific support packages could also be used for other series or variants, that could be documented in the package manifest.

gavanderhoorn, 2020-12-01:
Could I ask you to change the title of your issue? Right now it's quite unclear how it relates to the actual question.

EricMarcil, 2020-12-01:
Your suggesting is noted.  
It is true that in recent year, some models have multiple names for marketing reasons.  For example, the MH12 is the same as the MA1440.  MH12 refers to Material Handling with a payload of 12 kg, whereas MA1440 refers to Material ArcWelding with a reach of 1440 mm.  But it is the same arm.
We usually use the General or Handling name of the model on the ROS site because we haven't seen much ArcWelding application though we are starting to see some.
Normally both appelation of the robot would be present in the identification plate on the robot. 
If you don't have access to the physical robot, then look at the motoman.com website.  When a robot has multiple names, it usually has only one manual.  So if you look for the manual of the robot you are interested in, on the cover you will also see reference to the other model names.

EricMarcil, 2020-12-17:
I updated my previous comment because even if models are physically the same, there may be variations in parameters, so we have to be careful before saying to models are the same.  We might be able to put them in the same support package but you might need different URDF to account for the specification variations.

Gaspard-Bourgeois, 2020-12-24:
Thanks a lot for those precisions and have a great end of year.

closed 2020-12-24 by Gaspard-Bourgeois