ros-industrial/motoman #377: MH5 Support Update

acbuynak, 2020-12-06:
I am working on an update to the MH5 support package.
Primarily changing things to follow ros-i conventions, ie [REP-0199](http://gavanderhoorn.github.io/rep/rep-0199.html).

I have access to a [MH5LSII](https://www.motoman.com/en-us/products/robots/industrial/assembly-handling/mh-series/mh5ls-ii) variant.

The current MH5 support package matches the MH5F variant from what I can tell (c72c98c)(it's not notated specifically in package.xml). AFAIK, the MH5F variant is now [archived](https://www.motoman.com/en-us/products/product-archive) and not sold new.

There are physical differences in link (L&R) shape and length between the "F" and "LSII/SII" variants.

I think it would be helpful to new users to change the default support model to match what Motoman sells, the MH5LSII or MH5SII (differs by high or low speed). If not I'm happy to push to motoman_experimental or otherwise.

Looking for guidance. Thanks!

EricMarcil, 2020-12-10:
Hi,

The MH5F (FS100) and MH5SII (DX200) and MH5S(DX100) are the same (physically and parameters)
The MH5LF (FS100) and MH5SLII (DX200) and MH5SL(DX100) are the same (physically and parameters)
For the long model, it has a longer L-axis (different part) and an extension to the U-axis (front of U-axis the arm is the same).

I took a look at the current URDF on github, and I can't the reference to the current values.  The physical model matches the later version with the slim (tapered) R-axis.  But the parameters are probable from the initial model (none tapered R-axis) for which there were a lot of variants.

The values that I currently have for the short: 
[image]

and for the long model:
[image]

Below are the spec files for each models:
[MH5-series.zip](https://github.com/ros-industrial/motoman/files/5673410/MH5-series.zip)

So I would tend to agree that the model could be updated and we could make a variant for the long model.

acbuynak, 2021-01-05:
Thanks for the info! I'm pulling together an updated short "MH5" and a new long "MH5L" support package. I'll get those into a PR soon.

> I can't the reference to the current values. The physical model matches the later version with the slim (tapered) R-axis. But the parameters are probably from the initial model (none tapered R-axis) for which there were a lot of variants.

I'm seeing a disagreement for the U-axis motion limits between the above shared CSV files and the [MH5LSII webpage](https://www.motoman.com/en-us/products/robots/industrial/assembly-handling/mh-series/mh5ls-ii). Not sure if this is a discrepancy error or different model variants. Which should take precedence?

gavanderhoorn, 2021-01-05:
>  I'm pulling together an updated short "MH5" and a new long "MH5L" support package.

If those are actually variants, please put them in the same package, so we avoid creating more packages than absolutely necessary.

EricMarcil, 2021-01-05:
@acbuynak Use the ones from the CSV which are angle relative to the previous axis.  The values on the spec sheet you are referring to are angle relative to the ground.  So it is actually the compound angle of L and U axis that are written for the U-axis.

gavanderhoorn, 2025-03-19:
The work of @acbuynak and @cjue has been combined in #620.

I'll link that PR with this issue.

closed 2025-03-26 by gavanderhoorn