ros-industrial/motoman #379: Two robots, two platforms on rails, and a positioner

Doomerdinger, 2020-12-23:
I am currently trying to run a setup that involves two robots, each on top of a movable platform, and then a positioner.

The platforms are set up as base axis, and the positioner as an external.

I can communicate with and control the robots and platforms just fine, but I cannot get any feedback from the positioner.

I've seen a couple things around (such as [here](https://github.com/ros-industrial/motoman/pull/179#issuecomment-363800068) mentioning the driver will have issues with more than four groups. Has this capability been added and I am just setting up something incorrectly? If not, does anyone have some idea of where I can look to start implementing this functionality (if it is possible)?

ted-miller, 2020-12-23:
Unfortunately, the motion API is only capable of controlling 4 groups. So, your controller configuration will not work.

You'll need to split the two controllers and configure them as:
R1 + B1
R1 + B1 + S1

I'm not sure if this split can be done without a technician. I recommend contacting techsupport@motoman.com or 937-847-3200 for details on how to reconfigure the controllers.

closed 2020-12-23 by ted-miller

gavanderhoorn, 2020-12-24:
And just to clarify, with:

> the motion API is only capable of controlling 4 groups

@ted-miller refers to the MotoPlus API that is being used.

`motoman_driver` currently also doesn't support more than 4 groups in a single instance, but that's tightly coupled to the MotoPlus API used.

"ROS" doesn't really care.

kleinma, 2020-12-24:
@gavanderhoorn and @ted-miller, thank you so much for the help.

As, @Doomerdinger mentioned, we are interested in coordinated motion between an AR2010 and a GP180, both both mounted on mobile platforms, and a rotating positioner, for a total of 15 degrees of freedom, all running to a YRC1000. (Eventually, we'd like to add another positioner as well, making it 16.)

Is it possible to get tightly synchronized motion between all 16 servos using two separate controllers? I imagine we would have to have two Motoman ROS drivers running, communicating over two separate `robot_ip`s, and having to publish two separate `DynamicJointTrajectory` messages. Unless the synchronization is happening between the two robot controllers, I would think that it would be very hard to synchronize the publishing of the messages on the ROS side of things to get the desired coordinated motion that we want.

Also, would it be possible to modify the MotoPlus API to accept more than 4 groups in a single instance? If this is the only option to get tightly coordinated motion between more than four groups, we may be able to devote resources to making this happen.

gavanderhoorn, 2020-12-26:
Just to clarify: MotoPlus is the name of the SDK and runtime environment on the Yaskawa controller. It is not code under our control (our == here in his repository).

kleinma, 2020-12-27:
@gavanderhoorn - thank you for the clarification. I see the MotoPlus SDK is what was used to build the driver in this repo.

With that being said, is there any way to achieve coordinated motion between more than four groups using ROS? As I mentioned in my previous question, I don't see a clear path to make that happen using multiple controllers.

EricMarcil, 2020-12-28:
Currently, the only option is to split the system in two controllers.  Then have a node for each controller.  ROS doesn't use the controller build in coordinated motion function.  The trajectories are time stamped, so if you start at the same time by sendind the information to both controllers, you should be able to synchronize within one or two interpolation cycle (4 ms).  I'm not sure how fast your moving and what would represent a 4 ms variations in the motion execution.
We don't have any means of synchronizing interpolation clocks between controllers.