ros-industrial/motoman #383: start position tolerance value

Doomerdinger, 2021-01-07:
The tolerance value for the joint trajectory streamer is specified [here](https://github.com/ros-industrial/motoman/blob/a17705d6525d3de7413e924b1f0f3dd407300d30/motoman_driver/src/joint_trajectory_streamer.cpp#L63).

I am hoping to have some insight on why it is this value in particular.  I am running into issues with the states differing, and increasing this value fixes it. I just want to make sure there isn't some hardware limitation that this value was chosen.

@JeremyZoss @thiagodefreitas, you seem to be the ones that created/modified this value. Any information you could provide would be greatly appriciated.

JeremyZoss, 2021-01-07:
The MotoROS Position Control interface uses _incremental_ position commands at the lowest level (see [MotionServer.c:1324](https://github.com/ros-industrial/motoman/blob/a17705d6525d3de7413e924b1f0f3dd407300d30/motoman_driver/MotoPlus/MotionServer.c#L1324)).  The incremental commands are generated using only the joint trajectory data (positions, velocities, etc.) - no planning is done to move from the current robot position to the commanded trajectory.  In order to ensure that the robot moves as expected, the driver requires that the first trajectory point matches the current robot position.

This constraint is checked by the MotoROS driver (see [MotionServer.c:1185](https://github.com/ros-industrial/motoman/blob/a17705d6525d3de7413e924b1f0f3dd407300d30/motoman_driver/MotoPlus/MotionServer.c#L1185)) using a tolerance of `START_MAX_PULSE_DEVIATION`, which is currently set to 30 encoder pulses.  The actual conversion of encoder-pulses to ROS joint units (radians or meters) is robot-specific, and read from the controller at run-time by the driver.

If the robot is too far from the starting trajectory position, the MotoROS driver will reject the trajectory and ROS will print a somewhat arbitrary error message.  To provide redundant protection and a more helpful (?) error message, we duplicate this trajectory-start check on the ROS-side driver, at the location you've identified.  This tolerance is only an approximation of the actual tolerance applied by the MotoROS driver.  A quick google search suggests that 2500 pulse/deg may be typical, which gives ~2e-4 radians for the 30-pulse tolerance.  So the current ROS-side tolerance of 1e-4 radians is reasonable.  It may be off by orders of magnitude for linear axes or specific robot models, though.

You can safely widen the tolerance on the ROS-side driver, with the only consequence that your joint trajectory may be rejected by the robot-side driver rather than the ROS-side driver.

This is all from memory (7+ years ago!) and a quick code review, so you should exercise caution as you open up the tolerance.

Hope this helps.  Let me know if you have more questions.  I recommend this issue can be closed as-is, but will look for input from @Doomerdinger or others before closing it.

Doomerdinger, 2021-01-08:
Thank you so much for the excellent explanation! This clears everything up, and gives me a few places to look if I have any more questions.

closed 2021-01-08 by Doomerdinger