ros-industrial/motoman #386: Choppy motion of ROS controlled yaskawa GP8 robot

iom12, 2021-02-03:
I'm trying to command the GP8 robot via ROS / motoplus, sending it a sequence of trajectories (teleoperation). Each trajectory is made up of 2 points (The first point is the robot's current joint configuration. The second is the target joint configuration.). I get a non-continuous movement of the robot. I have a doubt that the frequency of sending messages from the controller to the actuators is the cause (if it is low). Is there any way to find out this frequency? Can we modify it? I thank you for your feedback.

gavanderhoorn, 2021-02-03:
I believe this is related to / a follow-up of [Choppy motion of ROS controlled yaskawa robot](https://answers.ros.org/question/370444) on ROS Answers.

> Then I run a ROS node which publishes my two-point trajectory into "/joint_trajectory_action topic".

`motoman_driver` is not really suited for what you're trying to do. It's possible to use trajectory replacement for these kinds of things, but that's not supported by `motoman_driver` right now. `motoman_driver` is essentially *trajectory based*, ie: it consumes trajectories, not individual position setpoints.

I'm pretty sure this is the cause of your problem.

You could look at #215, but note that it's not integrated with the current state of the driver.

For ROS 2, you could potentially look at what's discussed in #359. There's nothing really ROS 2 specific about that though, so with some work it could be used in ROS 1 as well.

iom12, 2021-02-04:
@gavanderhoorn Thank you for your answer. 
This subject is a follow-up  of [Choppy motion of ROS controlled yaskawa robot ](https://answers.ros.org/question/370444/choppy-motion-of-ros-controlled-yaskawa-robot/)on ROS Answers. 

> motoman_driver is essentially trajectory based, ie: it consumes trajectories, not individual position setpoints.

I am sending trajectories to my robot. This is done by sending goals to  "/joint_trajectory_action". 

> I'm pretty sure this is the cause of your problem. Setting velocities to zero will lead to stop-and-go behaviour.

I replaced the zero velocities by non-zero velocities corresponding to the diplacement divided by trajectory duration, and I am still having the same problem...

gavanderhoorn, 2021-02-04:
I may not have been clear enough: trajectory based in this case means: from 0-state to 0-state. So from a stopped state (0 velocity and acceleration) to another stopped state (0 velocity and acceleration).

To avoid me misunderstanding you, perhaps you could show us the following:

 - the code you use to generate and send the trajectories
 - an example trajectory (ie: captured using `rostopic echo` on the appropriate `goal` topic of the action server)
 - a wireshark capture of the traffic between the ROS PC and the Yaskawa controller

gavanderhoorn, 2021-02-16:
@iom12: could you comment to let us know whether this is still a problem for you?

if you'd like to continue diagnosing the issue, I'd repeat my request for the additional information I described in my previous comment.

gavanderhoorn, 2021-02-22:
@iom12: I hope you've been able to figure this out. If not, please comment and we can re-open.

You indicated you'd been in touch with Yaskawa France, so perhaps they've been able to help you.

For now, I'll close this.

closed 2021-02-22 by gavanderhoorn