ros-industrial/motoman #394: Smaller collision models

Doomerdinger, 2021-02-26:
Is there any community desire for optional smaller (as in number of vertices/faces, not psychical volume) collision meshes?

Some of the use cases I encounter rely heavily on running collision checking a _lot_.
I've seen significant timing improvements by going in and manually tweaking the collision meshes to use fewer vertices and faces.

The drawback being that the collision meshes don't reflect the bounds of visual robot quite as nicely (it always covers everything the more true to life mesh does for safety, but it also encloses additional volume potentially hindering some more close call checks.)

If done intelligently, very little is lost. I have an example of where I have done this for the Universal Robot UR10e [here](https://github.com/path-robotics/universal_robot/tree/melodic-devel-staging/ur_description/meshes/ur10e/collision). I created minimalized collision meshes, as well as alternative minimalized collision meshes which are also convex hulls (if the first iteration was already a convex hull, only the convex hull file is listed).

It is likely that I end up making something similar for Yaskawa arms we use. If there is no desire for these files, I will keep them out of any PRs I make.

cjue, 2021-12-14:
Smaller collision models are definitely interesting. I just want to point out that the GP4, GP7 and GP8 have their link2 "hollowed out" in a way that allows for more movement of joint3. A complex hull of link2 will therefore significantly restrict the range of motion of that joint, as shown here: https://github.com/ros-industrial/motoman/pull/433#issuecomment-988610550

gavanderhoorn, 2021-12-14:
> A complex hull [..]

I believe you mean *convex* hull.

See related discussion here: https://github.com/ros-industrial/fanuc/issues/65.

Creating lower-poly geometry without sacrificing quality is rather difficult, especially with curvy meshes, like most (newer) industrial robots include.

Doomerdinger, 2022-01-06:
@gavanderhoorn Yes, when I have done it I have done it manually and it is time consuming.
Given that I may be doing this process for a select number of robots anyway, I was mainly curious if there was any desire for me to put the lower poly models in a PR in some capacity.