ros-industrial/motoman #398: SDA10F doesn't move when executing trajectories

pablomalvido, 2021-03-26:
Hello,

I am working with a SDA10F robot with a FS100 controller and I am having problems for moving the real robot through ROS. This is the information about my system:

- OS version: Ubuntu 18.04.5
- ROS distribution: ROS Melodic
- Controller version: FS3.30.00A(US/FI)-00

I have followed the tutorials in http://wiki.ros.org/motoman_driver/Tutorials for installing the ROS server in the controller (first) and for commanding the robot using ROS (second tutorial). I have installed MotoROS using binaries (MotoRosFS_v192.out) and the MD5 hash is exactly the same as the one in the tutorial.
I have noticed that my problem is very similar to the one in #343 , where the problem was in the INIT_ROS job. In my case I directly copy-pasted the job in https://github.com/ros-industrial/motoman/blob/kinetic-devel/motoman_driver/Inform/DX200%2C%20FS100%2C%20YRC1000/sda_dual_arm/INIT_ROS.JBI in the controller using a USB, following the tutorial steps, however as you can see in the picture below, when I open the JOB in the pendant I cannot see the first lines, it starts from the NOP line. I don't know if this could be a problem, so I just let you know.

[image]

I downloaded the Motoman package from https://github.com/ros-industrial/motoman and followed all the instructions in the README file for the installation.

Now I will describe the steps followed for moving the robot and the obtained error:

- Robot controller is ON, Safety is OK, and the key is in remote mode.
- roscore in a terminal (Terminal 1)
- In a new terminal (Terminal 2), I execute catkin_make and source devel/setup.bash
- In the same terminal (Terminal 2) I run the command: `roslaunch motoman_sda10f_moveit_config moveit_planning_execution.launch sim:=false robot_ip:=10.0.0.2 controller:=fs100`
- I receive the following output message in Terminal 2 after running this command:

```
... logging to /home/remodel/.ros/log/722e710c-8e25-11eb-92b3-98fa9b9c7ffc/roslaunch-remodel-TAU-15030.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://remodel-TAU:42329/

SUMMARY
========

PARAMETERS
 * /controller_joint_names: ['arm_left_joint_...
 * /mongo_wrapper_ros_remodel_TAU_15030_8803950497284364922/database_path: /home/remodel/cat...
 * /mongo_wrapper_ros_remodel_TAU_15030_8803950497284364922/overwrite: False
 * /move_group/allow_trajectory_execution: True
 * /move_group/allowed_execution_duration_scaling: 1.2
 * /move_group/allowed_goal_duration_margin: 0.5
 * /move_group/arm_left/longest_valid_segment_fraction: 0.05
 * /move_group/arm_left/planner_configs: ['SBLkConfigDefau...
 * /move_group/arm_left/projection_evaluator: joints(arm_left_j...
 * /move_group/arm_right/longest_valid_segment_fraction: 0.05
 * /move_group/arm_right/planner_configs: ['SBLkConfigDefau...
 * /move_group/arm_right/projection_evaluator: joints(arm_right_...
 * /move_group/arms/longest_valid_segment_fraction: 0.05
 * /move_group/arms/planner_configs: ['SBLkConfigDefau...
 * /move_group/arms/projection_evaluator: joints(arm_left_j...
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'action_ns': 'j...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECEkConfigDefault/type: geometric::LBKPIECE
 * /move_group/planner_configs/PRMkConfigDefault/type: geometric::PRM
 * /move_group/planner_configs/PRMstarkConfigDefault/type: geometric::PRMstar
 * /move_group/planner_configs/RRTConnectkConfigDefault/type: geometric::RRTCon...
 * /move_group/planner_configs/RRTkConfigDefault/type: geometric::RRT
 * /move_group/planner_configs/RRTstarkConfigDefault/type: geometric::RRTstar
 * /move_group/planner_configs/SBLkConfigDefault/type: geometric::SBL
 * /move_group/planner_configs/TRRTkConfigDefault/type: geometric::TRRT
[... 277 lines cut ...]
[ INFO] [1616759321.006047014]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1616759321.006192131]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1616759321.006273648]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1616759321.006863092]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1616759321.007034552]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1616759321.007270599]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1616759321.007366514]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1616759321.007611529]: ParallelPlan::solve(): Solution found by one or more threads in 0.001773 seconds
[ INFO] [1616759321.007924906]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1616759321.007995030]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1616759321.008898829]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1616759321.009103118]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1616759321.009442701]: ParallelPlan::solve(): Solution found by one or more threads in 0.001579 seconds
[ INFO] [1616759321.015630345]: SimpleSetup: Path simplification took 0.005917 seconds and changed from 4 to 2 states
[ INFO] [1616759321.017627230]: Planning adapters have added states at index positions: [ 0 ]
[ INFO] [1616759340.518005105]: Execution request received
[ERROR] [1616759340.518161232]: Unable to identify any set of controllers that can actuate the specified joints: [ arm_left_joint_1_s arm_left_joint_2_l arm_left_joint_3_e arm_left_joint_4_u arm_left_joint_5_r arm_left_joint_6_b arm_left_joint_7_t ]
[ERROR] [1616759340.518215238]: Known controllers and their joints:

[ INFO] [1616759340.523446912]: ABORTED: Solution found but controller failed during execution
```

I hope you can help me to solve this problem. Please, let me know if you need more information and thank you very much in advance.

Pablo.

marip8, 2021-03-26:
> [ERROR] [1616759340.518215238]: Known controllers and their joints:

This appears to be a MoveIt configuration issue. The `move_group` node does not know about any controllers that are able to execute trajectories for the particular set of joints that you planned for. These controllers are configured in the `controllers.yaml` and `..._moveit_controller_manager.launch.xml` files of the MoveIt configuration package. This is what they should look like for the SDA10F (pulled from the SDA10F MoveIt config package in this repository): 

https://github.com/ros-industrial/motoman/blob/cd971e23993e781ee10080b17d4ab1508bffb8f2/motoman_sda10f_moveit_config/config/controllers.yaml#L1-L24

https://github.com/ros-industrial/motoman/blob/cd971e23993e781ee10080b17d4ab1508bffb8f2/motoman_sda10f_moveit_config/launch/motoman_sda10f_moveit_controller_manager.launch.xml#L1-L8

Are you using the SDA10F MoveIt config package from the `kinetic-devel` branch of this repository, or a custom version? In either case can you confirm that these two files are configured correctly on your computer? As far as I know, this repository's MoveIt config package for the SDA10F is set up correctly

pablomalvido, 2021-03-26:
Thank you for your response,

Yes, both these files (`controllers.yaml` and `motoman_sda10f_moveit_controller_manager.launch.xml`) are the ones from the motoman_sda10f_moveit_config package from the `kinetic-devel` branch of this repository and I didn't modify them, so they are exactly the same as the ones you attached.

Thank you.

pablomalvido, 2021-03-29:
Hello,

The problem was finally solved. For those experiencing a similar problem, the error was related to some missing files in MoveIt, not with the robot config files. After reinstalling MoveIt the robot started working.

closed 2021-03-29 by pablomalvido

gavanderhoorn, 2021-03-29:
Thanks for reporting back.

> the error was related to some missing files in MoveIt

it would be great if you could provide a bit more detail than "some missing files".

*Which* files specifically?

What should future readers of this issue install in order to avoid the problems you encountered?

pablomalvido, 2021-03-29:
The problem was solved after running: 

`sudo apt-get install ros-melodic-moveit`

The packages of this Motoman repository are working fine, the problem was that MoveIt was not correctly installed in the computer I was using.

> How did you build the packages? Did you only clone the git repository into your workspace?

I followed all the steps in the README of the kinetic-devel of the repository.

Thanks

gavanderhoorn, 2021-03-29:
Hm.

The `rosdep install --from-paths src/ --ignore-src --rosdistro melodic` step should've installed all required dependencies.

if that's not the case, there would certainly be "something wrong" with the packages.

pablomalvido, 2021-03-29:
I ran that command (`rosdep install --from-paths src/ --ignore-src --rosdistro melodic`) when installing the motoman packages, but the problem was not solved until running `sudo apt-get install ros-melodic-moveit`.