ros-industrial/motoman #400: TF and RobotDK frame transformation differences

arsh09, 2021-04-09:
Hello, 

I am using the GP25 description support package from this repository and I seem to be getting different rotation matrix when looked at the tool0 frame in base frame using TF. Please see the attached file. The frame transformation reported from RoboDK  and the one with the TF has consideration differences between them. 

I also tried to create my own URDF from CAD model of GP25 arm. However, as there is very less information where the actual frames are defined by Motoman, it is difficult to assign them (talking about the /base frame especially). 

[image]

[image]

Thank you

EricMarcil, 2021-04-09:
The URDF model is correct.  So are the values in RoboDK.  The Quaternion values displayed are also correct, so that would confirm that the model is correct.  I've tried various positions in RViz and in my MotoSim environment and the quaternion values are matching. Your issue seems to be in the conversion from the rotation matrix or quaternion to RPY (Euler ZYX).  There are various online tools to check rotation conversion, you can confirm that your issue is with the RPY calculations.  I'm not sure where that conversion is coded for what you are displaying on the screen.

arsh09, 2021-04-12:
Hello, 

As you can see in the tf_echo, the RPY are different than the ones in RoboDK. 

But you are correct,  Quaternions seem to match always. I will use them.

closed 2021-04-12 by arsh09