ros-industrial/motoman #401: Running Single Group on Dual-Arm Setup (Point Streaming)

RoboRiz, 2021-04-13:
Hello everyone.

I am running a dual-arm CSDA10F setup with the FS100 controllers and the configurations of #259.
Currently, I am planning to run some jogging applications on that robot using #215. I am already aware, that this is only applicable on the single robot arms, since for a dual-arm setup, this would need to be implemented for `motoman_msgs/DynamicJointTrajectory` message. 

Since a single arm jogging movement would be fine for me now, **I wonder if it is possible to run a dual-arm setup with only one specific group using the usual `trajectory_msgs/JointTrajectory`**?

This way, I hope to bypass the implementation of #215 for the `motoman_msgs/DynamicJointTrajectory`, since I don't really know if this is a matter of "copy and pasting" more or less.

Already tried:

- Modifying the `csda10f_motion_interface.yaml` and `joint_names_csda10f.yaml` in order to run a single-group robot streaming.

RoboRiz, 2021-04-23:
No need to only run one arm of a dual-arm setup. I have successfully edited the motoman driver such that a PointStreaming for a dual-arm setup is possible.

closed 2021-04-23 by RoboRiz

gavanderhoorn, 2021-04-23:
@RoboRiz: apologies for our lack of response.

For some reason this issue had completely escaped my attention.

> I have successfully edited the motoman driver such that a PointStreaming for a dual-arm setup is possible.

interesting.

Would you have those changes available somewhere?

What were the main changes you had to make?

tdl-ua, 2021-04-23:
Hi, would be interested in those edits as well. Would they also work on a manipulator-positioner multigroup setup?

RoboRiz, 2021-04-24:
I have copied the edits into this rep https://github.com/RoboRiz/PS_Dual. 
Please keep in mind, that my "solution" is not the generic one you probably looking for. There are 1,2 parts at least, which should be changed.

The changes:
Everything builds upon the work from #215. The only difference I made, was the additional Callback `JointTrajectoryStreamer::DualArmjointCommandCB(const motoman_msgs::DynamicJointTrajectoryConstPtr &msg)
`  here [DualArmjointCommandCB](https://github.com/RoboRiz/PS_Dual/blob/c9ca585e61c2496c60817d05e969ce35dd5aae69/motoman_driver/src/industrial_robot_client/joint_trajectory_streamer.cpp).

In Line [372](https://github.com/RoboRiz/PS_Dual/blob/c9ca585e61c2496c60817d05e969ce35dd5aae69/motoman_driver/src/industrial_robot_client/joint_trajectory_streamer.cpp#L372) , I assume for my use case, that  the given `DynamicJointTrajectory` msg includes the information for all joint groups (Group 0 - 3). At the start, I check each of those groups for zero Velocities. One might think about a scenario, where only specific joints are given in the `DynamicJointTrajectory` msg and those values are extracted via the `select` function.

In Line [499](https://github.com/RoboRiz/PS_Dual/blob/c9ca585e61c2496c60817d05e969ce35dd5aae69/motoman_driver/src/industrial_robot_client/joint_trajectory_streamer.cpp#L499), I just create the Simple Message. Again, i assume that information about all groups is given, so there is no need for the select function. Groups/joints that are not supposed to move, are described with Zero Velocities and same Position. I guess, you can also keep track of this in the driver. One would need to keep track of all joint group values. In each callback only the desired jointPositions and velocties would be passed further.

RoboRiz, 2021-04-24:
> Hi, would be interested in those edits as well. Would they also work on a manipulator-positioner multigroup setup?

As far as i understood the driver, i guess this should work.  It all comes down to create the simpleMessage properly. Please correct me, if I am wrong.