ros-industrial/motoman #403: SDA10F Fails when trying to execute movements with the 'arms' and 'sda10f' groups for moving the real robot

pablomalvido, 2021-05-04:
Hello,

I am working with a SDA10F robot with a FS100 controller and I am having problems for moving the ‘arms’ and ‘sda10f’ groups using MoveIt. This is the information about my system:

•	OS version: Ubuntu 18.04.5
•	ROS distribution: ROS Melodic
•	Controller version: FS3.30.00A(US/FI)-00

I have followed the tutorials in http://wiki.ros.org/motoman_driver/Tutorials for installing the ROS server in the controller (first) and for commanding the robot using ROS (second tutorial). I have installed MotoROS using binaries (MotoRosFS_v192.out) and the MD5 hash is exactly the same as the one in the tutorial. 

I run the command: `roslaunch motoman_sda10f_moveit_config moveit_planning_execution.launch sim:=false robot_ip:=10.0.0.2 controller:=fs100` to control real the robot with MoveIt, and I get the following output:

```
... logging to /home/remodel/.ros/log/c7435418-acc1-11eb-9e95-dc7196973bd7/roslaunch-remodel-TAU-4445.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
Deprecated: xacro tag 'insert_block' w/o 'xacro:' xml namespace prefix (will be forbidden in Noetic)
when processing file: /home/remodel/catkin_ws/src/motoman/motoman_sda10f_support/urdf/sda10f.xacro
Use the following command to fix incorrect tag usage:
find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'

started roslaunch server http://remodel-TAU:46867/

SUMMARY
========

PARAMETERS
 * /controller_joint_names: ['arm_left_joint_...
 * /mongo_wrapper_ros_remodel_TAU_4445_8992849688145291343/database_path: /home/remodel/cat...
 * /mongo_wrapper_ros_remodel_TAU_4445_8992849688145291343/overwrite: False
 * /move_group/allow_trajectory_execution: True
 * /move_group/allowed_execution_duration_scaling: 1.2
 * /move_group/allowed_goal_duration_margin: 0.5
 * /move_group/arm_left/longest_valid_segment_fraction: 0.05
 * /move_group/arm_left/planner_configs: ['SBLkConfigDefau...
 * /move_group/arm_left/projection_evaluator: joints(arm_left_j...
 * /move_group/arm_right/longest_valid_segment_fraction: 0.05
 * /move_group/arm_right/planner_configs: ['SBLkConfigDefau...
 * /move_group/arm_right/projection_evaluator: joints(arm_right_...
 * /move_group/arms/longest_valid_segment_fraction: 0.05
 * /move_group/arms/planner_configs: ['SBLkConfigDefau...
 * /move_group/arms/projection_evaluator: joints(arm_left_j...
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'action_ns': 'j...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECEkConfigDefault/type: geometric::LBKPIECE
 * /move_group/planner_configs/PRMkConfigDefault/type: geometric::PRM
 * /move_group/planner_configs/PRMstarkConfigDefault/type: geometric::PRMstar
 * /move_group/planner_configs/RRTConnectkConfigDefault/type: geometric::RRTCon...
 * /move_group/planner_configs/RRTkConfigDefault/type: geometric::RRT
 * /move_group/planner_configs/RRTstarkConfigDefault/type: geometric::RRTstar
 * /move_group/planner_configs/SBLkConfigDefault/type: geometric::SBL
 * /move_group/planner_configs/TRRTkConfigDefault/type: geometric::TRRT
 * /move_group/planning_plugin: ompl_interface/OM...
 * /move_group/planning_scene_monitor/publish_geometry_updates: True
 * /move_group/planning_scene_monitor/publish_planning_scene: True
 * /move_group/planning_scene_monitor/publish_state_updates: True
 * /move_group/planning_scene_monitor/publish_transforms_updates: True
 * /move_group/request_adapters: default_planner_r...
[... 287 lines cut ...]
[ INFO] [1620133750.305990510]: RRTConnect: Created 11 states (5 start + 6 goal)
[ INFO] [1620133750.306199635]: ParallelPlan::solve(): Solution found by one or more threads in 0.094113 seconds
[ INFO] [1620133750.306513173]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1620133750.306602727]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1620133750.311435840]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1620133750.329402311]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1620133750.329593453]: ParallelPlan::solve(): Solution found by one or more threads in 0.023200 seconds
[ INFO] [1620133750.343977298]: SimpleSetup: Path simplification took 0.014204 seconds and changed from 3 to 2 states
[ INFO] [1620133750.345698323]: Disabling trajectory recording
[ERROR] [1620133750.429225068]: Aborting Trajectory.  Failed to send point (#0): Invalid message (3) : Trajectory start position doesn't match current robot position (3011)
[ERROR] [1620133751.561740416]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 1.203715 seconds). Stopping trajectory.
[ INFO] [1620133751.561923600]: Cancelling execution for 
[ INFO] [1620133751.562141028]: Completed trajectory execution with status TIMED_OUT ...
[ INFO] [1620133751.600360881]: ABORTED: Timeout reached
[ WARN] [1620133756.899362002]: Transitioning goal to LOST
[ WARN] [1620133756.899762672]: Controller  done, no result returned
```

The following was reported in http://www.smartroboticsys.eu/?p=675&lang=en in 2015 about the SDA10F ROS package: _“What is still missing though, is MoveIT support for synchronized movement of arms/arms+torso respectively. It works fine in simulation, but due to some kind of MoveIT related error, the application fails when running on real robot”_. In my case, I can also execute movements with the ‘arms’ group with the MoveIt demo (`roslaunch motoman_sda10f_moveit_config demo.launch`), but as was previously reported, it fails when running on real robot.

I would like to know which is the situation about this, and if a solution has been found for the execution of synchronized arms movements with the ‘arms’ or ‘sda10f’ groups.

Thank you.

Pablo.

gavanderhoorn, 2021-05-11:
Apologies for the delayed response.

I believe #259 works towards fixing this. It's not complete, but it should at least get you past the problem you report.

Could you try that PR and let us know whether it fixes the "goal lost" problem?

pablomalvido, 2021-05-14:
Thank you very much for the support. After making the modifications in the commits of #259 the simultaneous dual arm movement started working.

I have two additional questions, one is related with the synchronized dual arm movements in general (1) and the other just with the robot communication (2), but maybe you can help me also with them. I can open a new issue with these questions if you prefer:

1. I want to do dual arm simultaneous cartesian movements. The only way I have found to this is by generating a cartesian plan for each arm individually and then merging them by code, adjusting the final duration to the duration of the plan that takes more time. I have tried the code with the real robot and it works fine, but there are some potential problems. As each individual plan is generated individually (using the compute_cartesian_path function), it doesn't consider potential self collisions between arms during the movements calculated in each plan. Another problem of this is that if I, for example, want to move both arms to the right 20cm but the distance between them is just 10cm, the left arm won't find a plan because at the moment of planning, the right arm is in the middle of the path. Is there any better solution for implementing this kind of movements?

2. I would like to access to the torque value of the joints of the robot in ROS. Is this information published in any topic or accessible by calling any service?

Thank you very much.

Best regards,
Pablo.

gavanderhoorn, 2021-05-14:
> I want to do dual arm simultaneous cartesian movements. The only way I have found to this is by generating a cartesian plan for each arm individually and then merging them by code, adjusting the [...]

tbh, this is more a motion planning issue, instead of an issue with the driver.

You mention `computeCartesianPath(..)`, which would make this a MoveIt question. I'd suggest discussing this with the MoveIt developers.

> I would like to access to the torque value of the joints of the robot in ROS. Is this information published in any topic or accessible by calling any service?

while there is support for this in MotoROS, the messages used between MotoROS and `motoman_driver` do not allow us to send this data to the ROS side.

See #174 for an old discussion about this (as part of the work to add velocity feedback to the `JointState`s published by the driver).

In order to add this, we'd have to add additional SimpleMessage messages, which is a non-trivial amount of work.

gavanderhoorn, 2021-05-14:
@pablomalvido: to test a number of changes to `motoman_driver`, we're looking for users with SDA type robots willing to spend some time running some tests.

Would that be something you'd be interested in?

The tests are pretty low-level, with which I mean I expect they require very little time: the changes either work or they don't.

In case you'd be willing to help, it'll take me some time to make the test version available, so I'll get in touch with you somewhere next week most likely.

gavanderhoorn, 2021-05-14:
As it appears at least the initial problem has been addressed, and it looks like we're tracking the issues with multi-group `motoman_driver` setups in other issues already, I'm going to close the issue.

Feel free to keep commenting on it @pablomalvido.

closed 2021-05-14 by gavanderhoorn

pablomalvido, 2021-05-14:
Thank you again for the support @gavanderhoorn and yes, I would be happy to help to keep developing the repository. So just contact me whenever you need to run the tests.

Have a nice weekend,
Pablo.