ros-industrial/motoman #404: YRC1000 GP7 can not get the joint states

yuetin, 2021-05-12:
Hi, we want to use motoman_driver and moveit to control the real robot.
After connected the robot, ROS PC couldn't get the robot feedback and control it when it servo on.

before i enable the robot,
the TP will send the packets.
[image]

- when I enable the robot
```bash
rosservice call /robot_enable "{}"

success: True
message: "Motoman robot is now enabled and will accept motion commands."
```

After enable robot and servo on. the TP didn't send the packets out.
[image]

- robot_interface_streaming can not get the new joint states
```bash
[ WARN] [1620814245.051917540]: Motoman robot is now enabled and will accept motion commands.
[ WARN] [1620814245.555061797]: Trajectory state not received for 1.000000 seconds

# few time later
[ INFO] [1620814777.525582011]: Connection failed, attempting reconnect
[ERROR] [1620814777.525703617]: Failed to connect to server, rc: -1. Error: 'Transport endpoint is already connected' (errno: 106)
[ WARN] [1620814777.525775871]: Not connected, bytes not sent
[ERROR] [1620814777.525839343]: Failed to receive message length
[ERROR] [1620814777.525899950]: Failed to receive incoming message
```

```bash
rostopic echo /joint_states
# stopping refresh when rosservice call /robot_enable "{}"
```

ROS distro: Noetic
MotoPlus: MotoRosYRC1_v194.out
Robot: Yaskawa GP7 with YRC1000

gavanderhoorn, 2021-05-12:
Could you try with [v1.9.3](https://github.com/ros-industrial/motoman/blob/1fd459a362210f84253fd9f388b32198cd8165e6/motoman_driver/MotoPlus/output/YRC1000/MotoRosYRC1_/MotoRosYRC1_v193.out)?

yuetin, 2021-05-13:
> Could you try with [v1.9.3](https://github.com/ros-industrial/motoman/raw/1fd459a362210f84253fd9f388b32198cd8165e6/motoman_driver/MotoPlus/output/YRC1000/MotoRosYRC1_/MotoRosYRC1_v193.out)?
> 
> Please make sure the `md5sum` of `MotoRosYRC1_v193.out` is `650c2e974b632b355f665cb9c856aab1` after you've downloaded it.

It seems same situation in 1.9.3 and 1.9.1

ted-miller, 2021-05-13:
According to the [Requirements wiki page](http://wiki.ros.org/motoman_driver#Requirements), our initial testing was done on YAS1.11.00-00. So, I expect that YAS1.34 should be ok. I can't think of any updates that would require newer system software. (Generally, if such a requirement exists, you would get an alarm 1020[5] on bootup.)

@yuetin My first guess is that there is an IP address conflict. If there is another device on the network with 192.168.1.155 address, then it can cause the robot's network adapter to stop working. Please try changing the IP of the robot controller to another address that is not used.

If the IP address does not solve the issue, please get a Wireshark capture of the communication between the PC and robot controller. If you are not familiar with Wireshark, there is a tutorial here: https://itsfoss.com/install-wireshark-ubuntu/
Please recreate the error with Wireshark running and save a copy of the recording.

yuetin, 2021-05-14:
> According to the [Requirements wiki page](http://wiki.ros.org/motoman_driver#Requirements), our initial testing was done on YAS1.11.00-00. So, I expect that YAS1.34 should be ok. I can't think of [...]

@ted-miller 
after i send enable to robot 
it start at line 7515

https://drive.google.com/file/d/1Ys-nY8VHWGH0igvwSo9TBLb0zFh-_UTb/view?usp=sharing

gavanderhoorn, 2021-05-14:
Not really sure what to make of that capture.

Controller seems to respond to the `Motoman Motion Ctrl` request message in the expected way (takes about 2.5 seconds, but it needs to enable the servos, so that can take a while).

But between receiving the `Motoman Motion Ctrl` request and sending the reply, TCP packets start to get lost. I can't tell why.

@yuetin: are you using a Virtual Machine to run ROS Noetic, or is this a native Ubuntu install?

Are there any other hosts on the same network? Is there heavy other traffic on the same network segment?

---

What I do find strange though is that the `Joint Feedback` messages between the `Motoman Motion Ctrl` request and reply seem to suggest the robot is *moving*, while the `Status` messages still show `Motion Possible: False (0)`, and there can't be any motion since `Motoman Motion Ctrl` hasn't completed yet. @ted-miller: could the robot be drooping a little after disengaging the breaks? Joints involved are `J1` and `J2`.

yuetin, 2021-05-14:
> Not really sure what to make of that capture.
> 
> Controller seems to respond to the `Motoman Motion Ctrl` request message in the expected way (takes about 2.5 seconds, but it needs to enable the servos, so that can take a while).
> 
> But between receiving the `Motoman Motion Ctrl` request and sending the reply, TCP packets start to get lost. I can't tell why.
> 
> @yuetin: are you using a Virtual Machine to run ROS Noetic, or is this a native Ubuntu install?
> 
> Are there any other hosts on the same network? Is there heavy other traffic on the same network segment?
> 
> (from the network traffic statistics screenshot of the GP7 controller I don't expect other hosts to be the problem, unless there's an IP conflict as @ted-miller mentions. @yuetin: have you check whether there are other hosts with the same IP?)
> 
> What I do find strange though is that the `Joint Feedback` messages between the `Motoman Motion Ctrl` request and reply seem to suggest the robot is _moving_, while the `Status` messages still show `Motion Possible: False (0)`, and there can't be any motion since `Motoman Motion Ctrl` hasn't completed yet. @ted-miller: could the robot be drooping a little after disengaging the breaks? Joints involved are `J1` and `J2`.

Noetic is base on docker
but we also use Melodic with native ubuntu to test it
and we get the same result
We are sure that there is nothing else occupying the network channel

yuetin, 2021-05-17:
> There's definitely something wrong with the network connection. Since yuetin reports that the GP12 works ok, then I assume the problem is in the robot controller.
> 
> @yuetin, Some things to try:
> 
> * Connect ethernet cable directly from YRC1000 to the PC. Do not go through any network switch.
> * Make sure that there are no extra INFORM jobs running on the teach pendant. INIT_ROS should be the only job.
> * Make sure there isn't a SYSTEM JOB running in the background. In TEACH mode, touch [Job] > [Ctrl System Job]. If the [Ctrl System Job] button doesn't exist, then it is OK.
> * Contact local Yaskawa office and request an update for your controller system software. I'm not aware of any compatibility issues, but your GP7 software is about 2 years older than your GP12.
> * You could try swapping the ACP01 board from the GP12 into the GP7. (Make sure to also swap the SD card in CN104.) If that works, then there could be an issue with the network adapter on the board.

 

- Connect ethernet cable directly from YRC1000 to the PC. Do not go through any network switch.
- Make sure that there are no extra INFORM jobs running on the teach pendant. INIT_ROS should be the only job.
- Make sure there isn't a SYSTEM JOB running in the background. In TEACH mode, touch [Job] > [Ctrl System Job]. If the [Ctrl System Job] button doesn't exist, then it is OK.

We have tried the above three methods
But there will still be situations where there is no return
But the situation is better than the original

We have contacted the local service office to upgrade the version and are waiting for a reply

gavanderhoorn, 2021-05-23:
@yuetin: could you update us as to the status of your support request?

yuetin, 2021-05-24:
> @yuetin: could you update us as to the status of your support request?

@gavanderhoorn 
The quotation given by the local Yaskawa is too high for us to accept
Maybe we will use other currently available robotic arms
Maybe I will try gp7 again when I have time in the future

ted-miller, 2021-05-24:
@yuetin 
One other thing to check is to make sure there are not any other MotoPlus applications installed on this controller. Perhaps another app is interfering with MotoROS.

- Turn on the robot controller while holding the {Main Menu} key on the keypad to enter Maintenance Mode. You may release the key when you see the logo appear on the screen.
- Touch [MotoPlus APL] > [File List] and verify that MotoROS is the only application which is loaded.

If that is not the issue, then please save a CMOS.BIN backup of your robot controller and post here. Be sure to zip the file; it compresses extremely well.

yuetin, 2021-05-26:
@ted-miller 

[CMOS.zip](https://github.com/ros-industrial/motoman/files/6543265/CMOS.zip)

Here is the CMOS.BIN
We sure there is only MotoROS application is loaded

ted-miller, 2021-05-26:
@yuetin, Please try posting the CMOS file again. The CMOS.zip above only contains an empty folder. There is no .bin file.

The yrcVerUp.exe and Version.txt are not required. These are used to restore the controller system software.

yuetin, 2021-05-27:
@ted-miller 

[CMOS.zip](https://github.com/ros-industrial/motoman/files/6550373/CMOS.zip)

If i could get newer yrcVerUp could it update the controller version?

ted-miller, 2021-05-27:
@yuetin, I was not able to reproduce the issue using your CMOS backup and controller software YAS1.34.00-00. So, this version of controller software is OK. You do not need to update the controller version.

I suspect that there may be a hardware fault. Try swapping the ACP01 board from the GP12 into the GP7. (Make sure to also swap the SD card in slot CN104.) If that works, then there could be an issue with the network adapter on the board.

I am not sure how to troubleshoot further.

closed 2021-08-11 by gavanderhoorn