ros-industrial/motoman #406: about support for SIA30D

xuhuairuogu, 2021-05-17:
Thanks for providing such a good ROS package for motoman control. However, I do not find motoman_sia30d_support directory. I wanna know if motoman_sia30d can be normally driven by the driver within this package. If yes, are you planning to write support files for motoman_sia30d? if no, what modifications I can make on this package?

gavanderhoorn, 2021-05-17:
> I wanna know if motoman_sia30d can be normally driven by the driver within this package

the driver -- as all drivers in `ros-industrial` -- is largely manipulator agnostic, meaning it should be usable with all robots which are supported by the controllers which this driver supports.

Summarising: yes, you should be able to use it with an SIA30.

>  If yes, are you planning to write support files for motoman_sia30d? 

I don't believe we have plans specific to the SIA30, but if there is demand we can see whether that could be added.

Creating a support package isn't *that* difficult btw. It would come down to creating files similar to `motoman_sia50d_support` (for instance), updating names, replacing the meshes and updating the `xacro:macro` (correct `link`s, `joint`s, etc).

gavanderhoorn, 2021-05-23:
@xuhuairuogu: could you indicate whether you're still interest in adding support for the SIA30?

xuhuairuogu, 2021-05-24:
> @xuhuairuogu: could you indicate whether you're still interest in adding support for the SIA30?

yes. Since I am not very good at handling 3D models, it would be a great help for me if you could add support for the SIA30.

gavanderhoorn, 2021-05-24:
@EricMarcil wrote:
>  I can try to get the model but it may take a couple of weeks.

@EricMarcil: could you request the meshes please?

EricMarcil, 2021-05-24:
Yes, I've put in a request for the model.  I'll try to put the model together once I have the mesh.

EricMarcil, 2021-05-27:
I've create PR #411 to add the SIA30D model.
@xuhuairuogu : Please review the model and let me know if there are any problems.

closed 2022-02-02 by gavanderhoorn